evcc-io/evcc #18855: Kia notworking

Tuerlich086, 2025-02-16:
### Describe the bug

Kia Niro EV wird nicht erkannt, über IOBroker funktioniert alles ohne Probleme mit den gleichen LogIn Daten sowie in der Kia Connect APP

Hab schon probiert den Kia über experimentell anzulegen, aber da kommt der gleiche Fehler. Bin gerade dabei EVCC zutesten, also ist eine komplett neue Config. 

### Steps to reproduce

1.    create template
2.    start EVCC
3.    error
...

### Configuration details

```yaml
vehicles:                                                                                                                                                                                   
- type: template
  template: kia
  user: xxxxx
  password: xxxxxx
  title: Niro
  capacity: 64
  language: de                                                                                                                                                                              
  name: Niro
```

### Log details

```text
[main ] INFO 2025/02/16 10:30:19 listening at :7070
[main ] ERROR 2025/02/16 10:30:25 creating vehicle Niro failed: cannot create vehicle type 'template': cannot create vehicle type 'kia': login failed: Get "https://prd.eu-ccapi.kia.com:8080/web/v1/user/authorize?cache=reset": dial tcp: lookup prd.eu-ccapi.kia.com: i/o timeout
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.0

StefanSchoof, 2025-02-16:
> lookup prd.eu-ccapi.kia.com: i/o timeout

Klingt nach einem DNS Problem. Was sagt `nslookup prd.eu-ccapi.kia.com` auf dem Rechner?

StefanSchoof, 2025-02-16:
Bei Netzwerk Problemen immer einmal ohne Docker probieren. Also einfach auf Windows, Mac oder einem Linux mit der config.

Tuerlich086, 2025-02-16:
Ok, du hast recht mit Windows komm ich rauf... mir ist aber nicht klar wieso?? Hab alle Ports entsprechend freigegeben

closed 2025-02-16 by andig