evcc-io/evcc #18857: Forecast.solar: add multiple roof settings

q2doc, 2025-02-16:
Hi,

As follow up: https://github.com/evcc-io/evcc/pull/18709#issuecomment-2660883059

Die Anregung/Idee/Anfrage ist ob eine Abfrage und Darstellung von mehr als eine Dachausrichtung integriert werden kann. Forecast.solar kann dies (realisiert in Solectrus) und es wäre toll es auch in EVCC zu haben. Anbei wie es in Solectrus konfiguriert wird:

[image]

Danke & Grüße

q2doc

andig, 2025-02-16:
Das kannst du aktuell schon als custom tariff. Ist aber im Backlog.

fletch247, 2025-02-16:
hi,danke für das update mit

fletch247, 2025-02-16:
hi,
danke für das update mit pv forecast und für evcc ganz allgemein!
ich hätte den gleichen wunsch das relativ einfach in die config eintragen zu können da ich das dach inkl. aller gauben und garage in allen erdenklichen richtungen und neigungen voll gemacht habe (reihenhaus, da muss man auf der kleinen fläche alles nutzen was man hat). im iobroker hab ich das auch mit forecast.solar ähnlich integriert wie q2doc oben zeigt.

falls jemand ein beispiel für den workaround über custom tariff hat wäre das sehr hilfreich :)

danke!

andig, 2025-02-16:
@fletch247 das funktioniert genau so wie das Template das umsetzt:

```yaml
type: custom
tariff: solar
forecast:
  source: http
  uri: https://api.forecast.solar/...
  jq: |
    [ .result.watts | to_entries.[] | {
      "start": (.key | strptime("%FT%T%z") | strftime("%FT%TZ")),
      "end":   (.key | strptime("%FT%T%z") | mktime+3600 | strftime("%FT%TZ")),
      "price": .value
    } ] | tostring
```

die URL baust Du Dir einfach gem. FS Doku zusammen.

polojoe, 2025-02-16:
Gibt es für Solcast auch eine Lösung für mehrere Ausrichtungen?

premultiply, 2025-02-16:
@andig Das funktioniert so leider nicht universell.
Es braucht hier eigentlich für jede Ausrichtung einen sepraten Request und dann lokal einen Merge (Addition) des einzelnen FC.
`tariff: solar` sollte eine Liste zulassen und die erhaltenen FC dann stunden- und tageweise addieren.
Dass das dann im Zweifelsfall die tägliche Quota durch die mehrfachen Requests deutlich schnell senkt sollte klar sein.

andig, 2025-02-16:
> Es braucht hier eigentlich für jede Ausrichtung einen sepraten Request und dann lokal einen Merge (Addition) des einzelnen FC.

Ich hab die FC Doku anders verstanden?

premultiply, 2025-02-16:
> Ich hab die FC Doku anders verstanden?

Das geht imho nur eingeschränkt wenn die Anlagen bzw. Ausrichtungen bestimmte Gemeinsamkeiten haben.
U.a. dürfen sie keine unterschiedlichen Verschattungshorizonte haben.

andig, 2025-02-16:
Danke, aber das ist alles längst klar! Wir hatten bereits diskutiert, nicht die Beschränkungen der Anbieter zu umgehen.

SolarPower2024, 2025-02-16:
Nur mal so als Idee in den Raum geworfen: die Liste zum addieren zulassen, allerdings nur 1x per Anbieter. Dann könnte man zb mit Solcast und Forecast.solar schon mal etwas kombinieren ohne die Beschränkung der Anbieter zu umgehen?

andig, 2025-02-16:
Gerne. PR welcome.

fletch247, 2025-02-17:
> [@fletch247](https://github.com/fletch247) das funktioniert genau so wie das Template das umsetzt:
> 
> type: custom
> tariff: solar
> forecast:
>   source: http
>   uri: https://api.forecast.solar/...
>   jq: |
>     [ .result.watts | to_entries.[] | {
>       "start": (.key | strptime("%FT%T%z") | strftime("%FT%TZ")),
>       "end":   (.key | strptime("%FT%T%z") | mktime+3600 | strftime("%FT%TZ")),
>       "price": .value
>     } ] | tostring
> die URL baust Du Dir einfach gem. FS Doku zusammen.

vielen dank!
ich bin kein programmierer, versuch mir aber zu helfen (chatgpt und testen, dauert halt ewig....).
wie das im iobroker adapter gemacht wird kann ich nicht sagen (bzgl. jede ausrichtung eine eigene anfrage und dadurch tägliche quota schnell erschöpft) aber es ist eh auch immer falsch was da kommt weil hier zusätzlich zu meinen ca. 3000 verschiedenen ausrichtungen auch noch viel schatten ist.

nun habe ich in meinem victron system auch einen forecast. dieser ist "intelligent" und kennt mein system. dessen werte kann ich per mqtt bekommen und das sieht so aus:

{"success":true,"records":{"solar_yield_forecast":[[1739746800000,0],[1739750400000,0],[1739754000000,0],[1739757600000,0],[1739761200000,0],[1739764800000,0],[1739768400000,0],[1739772000000,24.774508580936654],[1739775600000,231.3302736580364],[1739779200000,595.8467250053252],[1739782800000,1284.9659761270182],[1739786400000,1870.8647525406734],[1739790000000,2142.1032451454953],[1739793600000,2046.5886287830062],[1739797200000,1709.7444892312578],[1739800800000,1246.5814579655641],[1739804400000,593.2500037191538],[1739808000000,72.94314385301772],[1739811600000,0],[1739815200000,0],[1739818800000,0],[1739822400000,0],[1739826000000,0],[1739829600000,0],[1739833200000,0],[1739836800000,0]]},"totals":{"solar_yield_forecast":11818.993204609484},"options":{}}

in die evcc.yaml habe ich mit hilfe von chatgpt folgendes rein:

tariffs:
  solar:
    type: custom
    forecast:
      source: mqtt
      topic: Victron_VRM/PVForecast
      jq: |
        [ .records.solar_yield_forecast[] | 
          {
          "start": (.[0] / 1000 | strftime("%FT%TZ")), 
          "end": ((.[0] + 3600000) / 1000 | strftime("%FT%TZ")), 
          "price": (.[1] | tonumber | round)
          } 
        ] | tostring

das hat ewig gedauert (weil ich ein noob bin), funktioniert nun aber anscheinend.
danke für den anstoß!

ich denke das ist auch eine gute hilfe für andere die ihren pv forecast aus dem PV system in evcc bringen wollen :)

StefanSchoof, 2025-02-17:
Man kann auch das js plugin nutzen. Hier mal ein kleiner PoC:

```yaml
  solar:
    type: custom
    forecast:
      source: js
      script: |
        var combined = [];
        var sued = JSON.parse(sued);
        var west = JSON.parse(west);

        for (var i = 0; i < sued.length; i++) {
            var element = sued[i];
            for (var j = 0; j < west.length; j++) {
                if (west[j].start == element.start) {
                    combined.push({
                        start: element.start,
                        end: element.end,
                        price: west[j].price + element.price
                    });
                    break;
                }
            }
        }

        JSON.stringify(combined);
      in:
        - name: sued
          type: string
          config:
            source: http
            uri: https://api.forecast.solar/estimate/<parms>?time=utc&no_sun=1
            jq: |
              [ .result.watt_hours_period | to_entries.[] | {
                "start": (.key | strptime("%FT%T%z") | strftime("%FT%TZ")),
                "end":   (.key | strptime("%FT%T%z") | mktime+3600 | strftime("%FT%TZ")),
                "price": .value
              } ] | tostring
        - name: west
          type: string
          config:
            source: http
            uri: https://api.forecast.solar/estimate/<parms>?time=utc&no_sun=1
            jq: |
              [ .result.watt_hours_period | to_entries.[] | {
                "start": (.key | strptime("%FT%T%z") | strftime("%FT%TZ")),
                "end":   (.key | strptime("%FT%T%z") | mktime+3600 | strftime("%FT%TZ")),
                "price": .value
              } ] | tostring
```

thecem, 2025-02-18:
Wer den sesnor aus Home Assistant nutzen möchte: 

[image]

<details>

<summary>Daten über Rest aus dem Sensor </summary>

`{"entity_id":"sensor.combined_power_production","state":"0.0","attributes":{"state_class":"measurement","watts_combined":[0,3344,6884,9556,11115,11319,10391,8526,6005,3375,1619,0,0,3353,6784,9376,11042,11222,9751,7222,4483,2262,1065,0],"watts_timestamps":["2025-02-18T06:28:31+00:00","2025-02-18T07:00:00+00:00","2025-02-18T08:00:00+00:00","2025-02-18T09:00:00+00:00","2025-02-18T10:00:00+00:00","2025-02-18T11:00:00+00:00","2025-02-18T12:00:00+00:00","2025-02-18T13:00:00+00:00","2025-02-18T14:00:00+00:00","2025-02-18T15:00:00+00:00","2025-02-18T16:00:00+00:00","2025-02-18T16:47:20+00:00","2025-02-19T06:26:37+00:00","2025-02-19T07:00:00+00:00","2025-02-19T08:00:00+00:00","2025-02-19T09:00:00+00:00","2025-02-19T10:00:00+00:00","2025-02-19T11:00:00+00:00","2025-02-19T12:00:00+00:00","2025-02-19T13:00:00+00:00","2025-02-19T14:00:00+00:00","2025-02-19T15:00:00+00:00","2025-02-19T16:00:00+00:00","2025-02-19T16:49:03+00:00"],"wh_days_combined":[71193,65745],"wh_days_timestamps":["2025-02-18T00:00:00","2025-02-19T00:00:00"],"wh_period_combined":[0,877,5117,8223,10339,11219,10858,9461,7267,4693,2500,639,0,930,5073,8083,10213,11135,10489,8489,5855,3376,1667,435],"wh_period_timestamps":["2025-02-18T06:28:31+00:00","2025-02-18T07:00:00+00:00","2025-02-18T08:00:00+00:00","2025-02-18T09:00:00+00:00","2025-02-18T10:00:00+00:00","2025-02-18T11:00:00+00:00","2025-02-18T12:00:00+00:00","2025-02-18T13:00:00+00:00","2025-02-18T14:00:00+00:00","2025-02-18T15:00:00+00:00","2025-02-18T16:00:00+00:00","2025-02-18T16:47:20+00:00","2025-02-19T06:26:37+00:00","2025-02-19T07:00:00+00:00","2025-02-19T08:00:00+00:00","2025-02-19T09:00:00+00:00","2025-02-19T10:00:00+00:00","2025-02-19T11:00:00+00:00","2025-02-19T12:00:00+00:00","2025-02-19T13:00:00+00:00","2025-02-19T14:00:00+00:00","2025-02-19T15:00:00+00:00","2025-02-19T16:00:00+00:00","2025-02-19T16:49:03+00:00"],"watts_combined_with_timestamps":["2025-02-18T06:28:31+00:00:0","2025-02-18T07:00:00+00:00:3344","2025-02-18T08:00:00+00:00:6884","2025-02-18T09:00:00+00:00:9556","2025-02-18T10:00:00+00:00:11115","2025-02-18T11:00:00+00:00:11319","2025-02-18T12:00:00+00:00:10391","2025-02-18T13:00:00+00:00:8526","2025-02-18T14:00:00+00:00:6005","2025-02-18T15:00:00+00:00:3375","2025-02-18T16:00:00+00:00:1619","2025-02-18T16:47:20+00:00:0","2025-02-19T06:26:37+00:00:0","2025-02-19T07:00:00+00:00:3353","2025-02-19T08:00:00+00:00:6784","2025-02-19T09:00:00+00:00:9376","2025-02-19T10:00:00+00:00:11042","2025-02-19T11:00:00+00:00:11222","2025-02-19T12:00:00+00:00:9751","2025-02-19T13:00:00+00:00:7222","2025-02-19T14:00:00+00:00:4483","2025-02-19T15:00:00+00:00:2262","2025-02-19T16:00:00+00:00:1065","2025-02-19T16:49:03+00:00:0"],"unit_of_measurement":"W","device_class":"power","friendly_name":"Combined Power Production"},"last_changed":"2025-02-17T20:08:21.207965+00:00","last_reported":"2025-02-18T06:07:52.663029+00:00","last_updated":"2025-02-18T06:07:52.663029+00:00","context":{"id":"01JMBTQ29Q954JG8YTZ6NF1B32","parent_id":null,"user_id":null}}`

</details>

<details>

<summary>Config in evcc </summary>

```
tariffs:
  solar:
    type: custom
    forecast:
      source: http
      uri: http://homeassitant:8123/api/states/sensor.combined_power_production
      method: GET
      headers:
      - content-type: application/json
      - authorization: Bearer >dein Langlebiger Zugriffstoken<
      insecure: false # set to true to trust self-signed certificates
      jq: |
        .attributes.watts_combined as $combined |
        .attributes.watts_timestamps as $timestamps |
        [
          range(0; $combined | length) |
          {
            "start": ($timestamps[.] | strptime("%FT%T%z") | strftime("%FT%TZ")),
            "end": ($timestamps[.] | strptime("%FT%T%z") | mktime + 3600 | strftime("%FT%TZ")),
            "price": ($combined[.] | tonumber | round)
          }
        ] | tostring
```

</details>

flole123, 2025-02-18:
> Wer den sesnor aus Home Assistant nutzen möchte:
> 
> 

Genau das war auch mein Gedanke. Super, vielen Dank!

Ich habe in homeassistant zwei forecast.solar Konfigurationen. Sollte ich den Sensor, den du beschreibst schon haben, oder muss ich mir den noch generieren? Ich kann ihn so bei mir nicht finden.

thecem, 2025-02-18:
Ich habe kurzer hand die integration Forecast Solar zur einer Custom umgebaut, da die Forecast Werte Liste nicht als attribute über geben werden. Zwar werden diese abgefragt aber eben nicht als Sensor.
Die Macher der integration haben den change abgelehnt, als nicht geplant. Ich stelle heute ein custom repo ein, der das beschreibt was gemacht werden muss. (Den Linke ich dann hier). Zusätzlich brauchst du noch ein template Sensor der die Werte zusammen fast, der ist dann auch im Repo, kann dann auch angepasst werden.
Bei mir sind es halt zwölf (12) Flächen, nur so bekomme ich den Forecast sauber hin.

andig, 2025-02-18:
@thecem falls sich das für ein Template (mit Standard-HA) eignet gerne PR.

thecem, 2025-02-18:
https://github.com/thecem/forecast_solar/blob/main/README.md

thecem, 2025-02-18:
@andig leider nicht mit standard ha machbar, aber im repo beschrieben. Somit custom, sonst hätte ich nen PR gemacht.

closed 2025-02-18 by andig

thlink68, 2025-02-25:
Ich bevorzuge [www.solarprognose.de](url) als Lieferant for PV-Prognose, liefert in meinem Fall gefühlt bessere Vorhersagen als Forecast.Solar. Außerdem kann mehrere Module (mit unterschiedlichen Ausrichtungen) konfigurieren, aber die Prognose als Summe der Module abfragen. Hier mal eine funktionierende Konfiguration in evcc. 
Die URL fragt eine Mehr-Tages-Vorschau (inklusive des aktuellen Tages) unter Nutzung des mosmix-Algorithmus (alternativ kann man auch own-v1 nutzen) ab.

```
solar:
   - type: custom
     forecast:
       source: http
       uri: https://www.solarprognose.de/web/solarprediction/api/v1?_format=json&access-token=<access-token>&item=inverter&token=<token>&algorithm=mosmix&type=hourly&START_DAY=0&END_DAY=2
       jq: |
         [ .data | to_entries.[] | {
           "start": (.key | tonumber | strftime("%FT%TZ") ),
           "end": (.key | tonumber + 3600 | strftime("%FT%TZ")  ) ,
           "price": (.value.[0]*1000)
         } ] | tostring`
```

[image]

andig, 2025-02-25:
@thlink68 gerne PR um das ganze in ein Template zu verpacken. Danke!

thlink68, 2025-02-27:
@andig Mache ich gerne. Ist aber das erste mal für mich und weiß nicht leider was ich da genau machen muss.

thlink68, 2025-03-03:
Habe eine passende yaml-Datei erstellt. Ich habe allerdings keinen Weg gefunden die Datei als PR zur Verfügung zu stellen, daher hänge ich sie hier mal als Datei im Kommentar rein. 

[solarprognose.yaml.txt](https://github.com/user-attachments/files/19051731/solarprognose.yaml.txt)

StefanSchoof, 2025-03-03:
Am einfachsten kannst du die Datei unter https://github.com/evcc-io/evcc/new/master/templates/definition/tariff einfügen und benennen. Wenn du speicherst solltest du die Möglichkeit bekommen einen PR zu erstellen.

Spartacus68, 2025-05-22:
Moin zusammen,

gibt es zum Thema solarprognose.de und Integration in evcc etwas Neues oder ist das Thema gestorben? Ich bin gerade dabei evcc kennenzulernen und bin in diesem Zusammenhang auf solarprognose.de gestoßen, da man dort mehrere Ausrichtungen konfigurieren kann. Wäre klasse, wenn es da weitegeht!

thlink68, 2025-05-25:
> gibt es zum Thema solarprognose.de und Integration in evcc etwas Neues oder ist das Thema gestorben? Ich bin gerade dabei evcc kennenzulernen und bin in diesem Zusammenhang auf solarprognose.de gestoßen, da man dort mehrere Ausrichtungen konfigurieren kann. Wäre klasse, wenn es da weitegeht!

PR [19364](https://github.com/evcc-io/evcc/pull/19364) hatte ich gemacht, ging aber irgendwann nicht mehr weiter. Unklar warum, für mich war es eigentlich fertig, aber ich habe da auch wenig Wissen / Erfahrung.

Spartacus68, 2025-05-26:
ok, danke!