evcc-io/evcc #18863: Mercedes - Refresh token is getting after some hours

ReneNulschDE, 2025-02-16:
### Describe the bug

See [here](https://github.com/evcc-io/evcc/discussions/12581#discussioncomment-12215098)

For some reason, the RT is missing in the settings store and therefor a token refresh is not possible.

I will work on it. 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
See [here](https://github.com/evcc-io/evcc/discussions/12581#discussioncomment-12215098)
```

### Log details

```text
See [here](https://github.com/evcc-io/evcc/discussions/12581#discussioncomment-12215098)
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

Latest dev

RandomRam2023, 2025-02-16:
Rene, ich finde Dein Engagement für diese Sache großartig (Handclap/confetti).. und völlig ok, mit neuen major-SW-Versionen gibt es mitunter Probleme, weil ihr als Entwickler niemals alle Testumgebungen absichern könnt (und bei dem mir vor Wochen passierten Token-Sync bei hartem Raspi-OFF auch Fehler auf OSI_Level8 passieren)
.... ich bleib mal entspannt auf der 0.133, und folge diesem Issue.
Gruß
Ralf

etecprojekt, 2025-02-18:
Hallo,

wann wird der PR gemerged? @andig 
Haben das Problem ein neues Fahrzeug hinzuzufügen 😐

ReneNulschDE, 2025-02-18:
@etecprojekt : Der PR ist noch nicht fertig. Ich habe noch ein Problem bei meinen Tests gefunden und der Retest läuft noch.

etecprojekt, 2025-02-18:
> [@etecprojekt](https://github.com/etecprojekt) : Der PR ist noch nicht fertig. Ich habe noch ein Problem bei meinen Tests gefunden und der Retest läuft noch.

Ah ok, dachte er wäre nur nicht freigegeben, dank dir 😉

closed 2025-02-18 by andig