evcc-io/evcc #18864: Fehler beim Starten. Überprüfe deine Konfiguration und starte neu. [mygrid] cannot create meter 'mygrid': cannot create meter type 'template': cannot create meter type 'custom': power: sunspec model not found: [{203 0 W}]

ae798, 2025-02-16:
### Describe the bug

Hallo, ich bekomme seit einigen Monaten immer wieder die Fehelermeldung "Fehler beim Starten. Überprüfe deine Konfiguration und starte neu. [mygrid] cannot create meter 'mygrid': cannot create meter type 'template': cannot create meter type 'custom': power: sunspec model not found: [{203 0 W}]". Dann geht es wieder 1-2 Tage, dann fällt es wieder aus. (Nur ein Zugriff, keine weiteren Anfragen)

### Steps to reproduce

1.siehe Bug
2.
3.
...

### Configuration details

```yaml
meters:
- name: mygrid
  type: template
  template: solaredge-hybrid
  usage: grid
  modbus: tcpip
  host: 192.168.2.144
  port: 502
  timeout: 30s
```

### Log details

```text
Fehler beim Starten. Überprüfe deine Konfiguration und starte neu. [mygrid] cannot create meter 'mygrid': cannot create meter type 'template': cannot create meter type 'custom': power: sunspec model not found: [{203 0 W}]
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.0

andig, 2025-02-16:
Gehts nach Neustart? Anscheinend hat das Gerät ein Problem wenn eine neue Netzwerkverbindung aufgemacht wird. Da können wir leider wenig tun.

closed 2025-02-16 by andig

ae798, 2025-02-16:
Sporadisch Von meinem iPhone gesendet

Closed #18864 as completed.