evcc-io/evcc #18877: Fahrzeug lädt nach Update nicht mehr

Steffen0707, 2025-02-16:
### Describe the bug

Seit dem Update auf 0.200.0 lädt mein Fahrzeug nicht mehr, ohne evcc lädt dass Fahrzeug.
ich hab nichts geändert nach dem Update und kurz davor auch nicht
kommt immer der Fehler switch phases: modbus: exception '3' (illegal data value), function '6'
wenn dass Fahrzeug abgesteckt ist kommt der Fehler nicht mehr

Update auf 0.200.1 hat auch nicht geholfen

### Steps to reproduce

1. Fahrzeug anstecken
2. Fahrzeug ausstecken Fehler weg
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

database:
  type: sqlite
  dsn: /var/lib/evcc/evcc.db

log: debug
levels:
  cache: error

# unique installation id
plant: xxxxxxxxxxxxxxxxxxxxxxx

interval: 10s # control cycle interval

sponsortoken: xxxxxxxxxxxxxxxxxxxxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: victron-energy 
  usage: grid  
  host: 192.168.2.60  
  port: 502  
  name: grid1
- type: template
  template: victron-energy 
  usage: pv  
  host: 192.168.2.60  
  port: 502  
  name: pv2
- type: template
  template: victron-energy 
  usage: battery  
  host: 192.168.2.60  
  port: 502  
  name: battery3

chargers:
- type: template
  template: alfen 
  id: 1  
  host: 192.168.2.79  
  port: 502  
[... 687 lines cut ...]
[site  ] DEBUG 2025/02/16 20:25:05 battery 1 power: -0W
[site  ] DEBUG 2025/02/16 20:25:05 battery 1 soc: 47%
[site  ] DEBUG 2025/02/16 20:25:05 grid power: 3063W
[site  ] DEBUG 2025/02/16 20:25:05 pv 1 power: 7W
[site  ] DEBUG 2025/02/16 20:25:05 site power: 3163W
[lp-1  ] DEBUG 2025/02/16 20:25:05 charge total import: 302.734kWh
[lp-1  ] DEBUG 2025/02/16 20:25:05 charger status: B
[lp-1  ] ERROR 2025/02/16 20:25:06 switch phases: modbus: exception '3' (illegal data value), function '6'
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.1

andig, 2025-02-16:
Could you share a trace log for Alfen on old and new version to compare?

Steffen0707, 2025-02-16:
was Frei geben?
bin kein ITler

Steffen0707, 2025-02-16:
[alfen ] TRACE 2025/02/16 20:43:16 modbus: send 00 01 00 00 00 06 01 03 01 32 00 06
[alfen ] TRACE 2025/02/16 20:43:16 modbus: recv 00 01 00 00 00 0f 01 03 0c 43 66 28 f5 43 65 99 99 43 67 5c 29
[alfen ] DEBUG 2025/02/16 20:43:16 detected 3p alfen
[alfen ] TRACE 2025/02/16 20:43:16 modbus: send 00 02 00 00 00 06 01 03 01 76 00 04
[alfen ] TRACE 2025/02/16 20:43:16 modbus: recv 00 02 00 00 00 0b 01 03 08 41 12 7b 58 00 00 00 00
[alfen ] TRACE 2025/02/16 20:43:16 modbus: send 00 03 00 00 00 06 01 03 04 ba 00 02
[alfen ] TRACE 2025/02/16 20:43:16 modbus: recv 00 03 00 00 00 07 01 03 04 41 80 00 00
[alfen ] TRACE 2025/02/16 20:43:16 modbus: send 00 04 00 00 00 0b 01 10 04 ba 00 02 04 40 e0 00 00
[alfen ] TRACE 2025/02/16 20:43:16 modbus: recv 00 04 00 00 00 06 01 10 04 ba 00 02
[alfen ] TRACE 2025/02/16 20:43:16 modbus: send 00 05 00 00 00 06 01 03 01 58 00 02
[alfen ] TRACE 2025/02/16 20:43:16 modbus: recv 00 05 00 00 00 07 01 03 04 44 42 93 33
[alfen ] TRACE 2025/02/16 20:43:16 modbus: send 00 06 00 00 00 06 01 03 01 40 00 06
[alfen ] TRACE 2025/02/16 20:43:16 modbus: recv 00 06 00 00 00 0f 01 03 0c 40 d8 08 32 40 d9 26 ea 40 d8 08 32
[alfen ] TRACE 2025/02/16 20:43:16 modbus: send 00 07 00 00 00 06 01 03 01 76 00 04
[alfen ] TRACE 2025/02/16 20:43:17 modbus: recv 00 07 00 00 00 0b 01 03 08 41 12 7b 58 00 00 00 00
[alfen ] TRACE 2025/02/16 20:43:17 modbus: send 00 08 00 00 00 06 01 03 04 b1 00 05
[alfen ] TRACE 2025/02/16 20:43:17 modbus: recv 00 08 00 00 00 0d 01 03 0a 43 32 00 00 00 00 00 00 00 00
[alfen ] TRACE 2025/02/16 20:43:17 modbus: send 00 09 00 00 00 06 01 03 01 76 00 04
[alfen ] TRACE 2025/02/16 20:43:17 modbus: recv 00 09 00 00 00 0b 01 03 08 41 12 7b 58 00 00 00 00
[alfen ] TRACE 2025/02/16 20:43:17 modbus: send 00 0a 00 00 00 06 01 03 01 76 00 04
[alfen ] TRACE 2025/02/16 20:43:17 modbus: recv 00 0a 00 00 00 0b 01 03 08 41 12 7b 58 00 00 00 00
[alfen ] TRACE 2025/02/16 20:43:17 modbus: send 00 0b 00 00 00 06 01 03 01 76 00 04
[alfen ] TRACE 2025/02/16 20:43:17 modbus: recv 00 0b 00 00 00 0b 01 03 08 41 12 7b 58 00 00 00 00
[alfen ] TRACE 2025/02/16 20:43:17 modbus: send 00 0c 00 00 00 06 01 03 04 ba 00 02
[alfen ] TRACE 2025/02/16 20:43:17 modbus: recv 00 0c 00 00 00 07 01 03 04 40 e0 00 00
[alfen ] TRACE 2025/02/16 20:43:17 modbus: send 00 0d 00 00 00 06 01 06 04 bf 00 01
[alfen ] TRACE 2025/02/16 20:43:18 modbus: recv 00 0d 00 00 00 03 01 86 03
[alfen ] TRACE 2025/02/16 21:01:22 modbus: send 00 0e 00 00 00 06 01 03 01 58 00 02
[alfen ] TRACE 2025/02/16 21:01:22 modbus: recv 00 0e 00 00 00 07 01 03 04 00 00 00 00
[alfen ] TRACE 2025/02/16 21:01:22 modbus: send 00 0f 00 00 00 06 01 03 01 40 00 06
[alfen ] TRACE 2025/02/16 21:01:22 modbus: recv 00 0f 00 00 00 0f 01 03 0c 00 00 00 00 00 00 00 00 00 00 00 00
[alfen ] TRACE 2025/02/16 21:01:22 modbus: send 00 10 00 00 00 06 01 03 01 76 00 04
[alfen ] TRACE 2025/02/16 21:01:22 modbus: recv 00 10 00 00 00 0b 01 03 08 41 12 7b 58 00 00 00 00
[alfen ] TRACE 2025/02/16 21:01:22 modbus: send 00 11 00 00 00 06 01 03 01 76 00 04
[alfen ] TRACE 2025/02/16 21:01:22 modbus: recv 00 11 00 00 00 0b 01 03 08 41 12 7b 58 00 00 00 00
[alfen ] TRACE 2025/02/16 21:01:22 modbus: send 00 12 00 00 00 06 01 03 04 b1 00 05
[alfen ] TRACE 2025/02/16 21:01:22 modbus: recv 00 12 00 00 00 0d 01 03 0a 42 32 00 00 00 00 00 00 00 00
[alfen ] TRACE 2025/02/16 21:01:22 modbus: send 00 13 00 00 00 06 01 03 01 76 00 04
[alfen ] TRACE 2025/02/16 21:01:22 modbus: recv 00 13 00 00 00 0b 01 03 08 41 12 7b 58 00 00 00 00
[alfen ] TRACE 2025/02/16 21:01:22 modbus: send 00 14 00 00 00 06 01 03 01 76 00 04
[alfen ] TRACE 2025/02/16 21:01:23 modbus: recv 00 14 00 00 00 0b 01 03 08 41 12 7b 58 00 00 00 00
[alfen ] TRACE 2025/02/16 21:01:23 modbus: send 00 15 00 00 00 06 01 03 04 ba 00 02
[alfen ] TRACE 2025/02/16 21:01:23 modbus: recv 00 15 00 00 00 07 01 03 04 40 e0 00 00
[alfen ] TRACE 2025/02/16 21:01:23 modbus: send 00 16 00 00 00 06 01 06 04 bf 00 01
[alfen ] TRACE 2025/02/16 21:01:23 modbus: recv 00 16 00 00 00 03 01 86 03
[alfen ] TRACE 2025/02/16 21:01:32 modbus: send 00 17 00 00 00 06 01 03 01 58 00 02
[alfen ] TRACE 2025/02/16 21:01:32 modbus: recv 00 17 00 00 00 07 01 03 04 00 00 00 00
[alfen ] TRACE 2025/02/16 21:01:32 modbus: send 00 18 00 00 00 06 01 03 01 40 00 06
[alfen ] TRACE 2025/02/16 21:01:32 modbus: recv 00 18 00 00 00 0f 01 03 0c 00 00 00 00 00 00 00 00 00 00 00 00
[alfen ] TRACE 2025/02/16 21:01:32 modbus: send 00 19 00 00 00 06 01 03 01 76 00 04
[alfen ] TRACE 2025/02/16 21:01:32 modbus: recv 00 19 00 00 00 0b 01 03 08 41 12 7b 58 00 00 00 00
[alfen ] TRACE 2025/02/16 21:01:32 modbus: send 00 1a 00 00 00 06 01 03 04 b1 00 05
[alfen ] TRACE 2025/02/16 21:01:32 modbus: recv 00 1a 00 00 00 0d 01 03 0a 42 32 00 00 00 00 00 00 00 00
[alfen ] TRACE 2025/02/16 21:01:32 modbus: send 00 1b 00 00 00 06 01 03 04 ba 00 02
[alfen ] TRACE 2025/02/16 21:01:32 modbus: recv 00 1b 00 00 00 07 01 03 04 40 e0 00 00
[alfen ] TRACE 2025/02/16 21:01:32 modbus: send 00 1c 00 00 00 06 01 06 04 bf 00 01
[alfen ] TRACE 2025/02/16 21:01:32 modbus: recv 00 1c 00 00 00 03 01 86 03
[alfen ] TRACE 2025/02/16 21:01:35 modbus: send 00 1d 00 00 00 0b 01 10 04 ba 00 02 04 00 00 00 00
[alfen ] TRACE 2025/02/16 21:01:35 modbus: recv 00 1d 00 00 00 06 01 10 04 ba 00 02
[alfen ] TRACE 2025/02/16 21:01:42 modbus: send 00 1e 00 00 00 06 01 03 01 58 00 02
[alfen ] TRACE 2025/02/16 21:01:42 modbus: recv 00 1e 00 00 00 07 01 03 04 00 00 00 00
[... cut ...]
[alfen ] TRACE 2025/02/16 21:04:02 modbus: send 00 7a 00 00 00 06 01 03 04 b1 00 05
[alfen ] TRACE 2025/02/16 21:04:02 modbus: recv 00 7a 00 00 00 0d 01 03 0a 42 31 00 00 00 00 00 00 00 00
[alfen ] TRACE 2025/02/16 21:04:02 modbus: send 00 7b 00 00 00 06 01 03 04 ba 00 02
[alfen ] TRACE 2025/02/16 21:04:02 modbus: recv 00 7b 00 00 00 07 01 03 04 00 00 00 00
[alfen ] TRACE 2025/02/16 21:04:02 modbus: send 00 7c 00 00 00 06 01 06 04 bf 00 01
[alfen ] TRACE 2025/02/16 21:04:02 modbus: recv 00 7c 00 00 00 03 01 86 03

Hoffentlich ist das richtig

andig, 2025-02-16:
Bitte mit debug- sonst ist das nicht lesbar 😬

Steffen0707, 2025-02-16:
[alfen ] DEBUG 2025/02/16 20:43:16 detected 3p alfen

mehr ist nicht drin
hatte den Raspi aus dass ich laden konnte
kann ich die alten Debug daten nicht mehr anzeigen?

VolkerK62, 2025-02-17:
yaml ändern und dann noch mal ein Log ziehen
```
log: debug
levels:
  cache: error
  alfen: trace
```

hyperbart, 2025-02-17:
@Steffen0707 I think I had the same issue a while back.
You are using an Alfen, can you please check via ACE Installer if the setting for switching between phases has been disabled?

hyperbart, 2025-02-17:
@andig not sure yet, I just had a hunch it might be this, but we might.
I am very curious to @Steffen0707 's reply.
The issue that popped in my mind was this one, I looked it up: https://github.com/evcc-io/evcc/issues/17889.
However, it is is another modbus register/number (function 6 here versus function 16 in the issue I am referring to), that's a level of detail on which I am not familiair what the difference is between these functions.

Steffen0707, 2025-02-17:
@hyperbart 
Ich hab einen Smart und der hat den Cameleon Lader von der Renault Zoe drin da dürfen die Phasen nicht wärend dem Laden Umgestellt werden, im ACE Installer ist der Hacken gesetzt dass die ZE Eyrkennung aktiv ist.

Ich hab an der Alfen Wallbox nichts geändert.
Home Assistant wird nicht verwendet

```
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
 alfen: trace
```

ich hab die yaml geändert aber es kommt immer noch der gleiche Fehler

Steffen0707, 2025-02-17:
[alfen ] DEBUG 2025/02/17 17:19:04 detected 3p alfen

[main  ] INFO 2025/02/17 17:19:04 evcc 0.200.1
[main  ] INFO 2025/02/17 17:19:04 using config file: /etc/evcc.yaml
[db    ] INFO 2025/02/17 17:19:04 using sqlite database: /var/lib/evcc/evcc.db
[main  ] INFO 2025/02/17 17:19:04 listening at :7070
[alfen ] DEBUG 2025/02/17 17:19:04 detected 3p alfen
[lp-1  ] DEBUG 2025/02/17 17:19:04 charge total import: 313.641kWh
[site  ] INFO 2025/02/17 17:19:04 site config:
[site  ] INFO 2025/02/17 17:19:04   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2025/02/17 17:19:04     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/02/17 17:19:04     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/02/17 17:19:04     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2025/02/17 17:19:04   vehicles:
[site  ] INFO 2025/02/17 17:19:04     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2025/02/17 17:19:04 loadpoint 1:
[lp-1  ] INFO 2025/02/17 17:19:04   mode:        minpv
[lp-1  ] INFO 2025/02/17 17:19:04   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2025/02/17 17:19:04   meters:      charge ✓
[lp-1  ] INFO 2025/02/17 17:19:04     charge:    power ✓ energy ✓ currents ✓
[site  ] DEBUG 2025/02/17 17:19:04 set buffer soc: 55
[site  ] DEBUG 2025/02/17 17:19:04 set buffer start soc: 55
[site  ] DEBUG 2025/02/17 17:19:04 set priority soc: 50
[site  ] DEBUG 2025/02/17 17:19:04 set battery discharge control: false
[lp-1  ] DEBUG 2025/02/17 17:19:04 phase timer inactive
[lp-1  ] DEBUG 2025/02/17 17:19:04 pv timer inactive
[site  ] WARN 2025/02/17 17:19:04 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2025/02/17 17:19:04 ----
[lp-1  ] DEBUG 2025/02/17 17:19:04 charge power: 0W
[lp-1  ] DEBUG 2025/02/17 17:19:04 charge currents: [0 0 0]A
[site  ] DEBUG 2025/02/17 17:19:04 battery 1 power: 747W
[site  ] DEBUG 2025/02/17 17:19:04 grid power: 27W
[site  ] DEBUG 2025/02/17 17:19:04 battery 1 soc: 72%
[site  ] DEBUG 2025/02/17 17:19:05 pv 1 power: 549W
[site  ] DEBUG 2025/02/17 17:19:05 site power: 774W
[lp-1  ] DEBUG 2025/02/17 17:19:05 charge total import: 313.641kWh
[lp-1  ] DEBUG 2025/02/17 17:19:05 charger status: A
[lp-1  ] INFO 2025/02/17 17:19:05 car disconnected
[lp-1  ] DEBUG 2025/02/17 17:19:05 set battery boost: false
[lp-1  ] DEBUG 2025/02/17 17:19:05 set session soc limit: 0
[lp-1  ] DEBUG 2025/02/17 17:19:05 set session energy limit: 0
[site  ] DEBUG 2025/02/17 17:19:15 ----
[lp-1  ] DEBUG 2025/02/17 17:19:15 charge power: 0W
[lp-1  ] DEBUG 2025/02/17 17:19:15 charge currents: [0 0 0]A
[site  ] DEBUG 2025/02/17 17:19:15 battery 1 power: 752W
[site  ] DEBUG 2025/02/17 17:19:15 grid power: -2W
[site  ] DEBUG 2025/02/17 17:19:15 battery 1 soc: 72%
[site  ] DEBUG 2025/02/17 17:19:15 pv 1 power: 551W
[site  ] DEBUG 2025/02/17 17:19:15 site power: 750W
[lp-1  ] DEBUG 2025/02/17 17:19:15 charge total import: 313.641kWh
[lp-1  ] DEBUG 2025/02/17 17:19:15 charger status: A
[site  ] DEBUG 2025/02/17 17:19:25 ----
[lp-1  ] DEBUG 2025/02/17 17:19:25 charge power: 0W
[lp-1  ] DEBUG 2025/02/17 17:19:25 charge currents: [0 0 0]A
[site  ] DEBUG 2025/02/17 17:19:25 battery 1 power: 747W
[site  ] DEBUG 2025/02/17 17:19:25 grid power: 19W
[site  ] DEBUG 2025/02/17 17:19:25 battery 1 soc: 72%
[site  ] DEBUG 2025/02/17 17:19:25 pv 1 power: 608W
[site  ] DEBUG 2025/02/17 17:19:25 site power: 766W
[lp-1  ] DEBUG 2025/02/17 17:19:25 charge total import: 313.641kWh
[lp-1  ] DEBUG 2025/02/17 17:19:25 charger status: A
[site  ] DEBUG 2025/02/17 17:19:35 ----
[lp-1  ] DEBUG 2025/02/17 17:19:35 charge power: 0W
[lp-1  ] DEBUG 2025/02/17 17:19:35 charge currents: [0 0 0]A
[site  ] DEBUG 2025/02/17 17:19:35 battery 1 power: 731W
[site  ] DEBUG 2025/02/17 17:19:35 battery 1 soc: 72%
[site  ] DEBUG 2025/02/17 17:19:35 grid power: -34W
[site  ] DEBUG 2025/02/17 17:19:35 pv 1 power: 597W
[site  ] DEBUG 2025/02/17 17:19:35 site power: 697W
[lp-1  ] DEBUG 2025/02/17 17:19:35 charge total import: 313.641kWh
[... 905 lines cut ...]
[lp-1  ] DEBUG 2025/02/17 17:33:25 charge total import: 313.641kWh
[lp-1  ] DEBUG 2025/02/17 17:33:26 charger status: B
[lp-1  ] ERROR 2025/02/17 17:33:26 switch phases: modbus: exception '3' (illegal data value), function '6'
[site  ] DEBUG 2025/02/17 17:33:35 ----
[lp-1  ] DEBUG 2025/02/17 17:33:35 charge power: 0W
[lp-1  ] DEBUG 2025/02/17 17:33:35 charge currents: [0 0 0]A
[site  ] DEBUG 2025/02/17 17:33:35 battery 1 power: 1317W
[site  ] DEBUG 2025/02/17 17:33:35 grid power: 15W
[site  ] DEBUG 2025/02/17 17:33:35 battery 1 soc: 71%
[site  ] DEBUG 2025/02/17 17:33:35 pv 1 power: 117W
[site  ] DEBUG 2025/02/17 17:33:35 site power: 1332W
[lp-1  ] DEBUG 2025/02/17 17:33:35 charge total import: 313.641kWh
[lp-1  ] DEBUG 2025/02/17 17:33:36 charger status: B
[lp-1  ] ERROR 2025/02/17 17:33:36 switch phases: modbus: exception '3' (illegal data value), function '6'
[site  ] DEBUG 2025/02/17 17:33:45 ----
[lp-1  ] DEBUG 2025/02/17 17:33:45 charge power: 0W
[lp-1  ] DEBUG 2025/02/17 17:33:45 charge currents: [0 0 0]A
[site  ] DEBUG 2025/02/17 17:33:45 battery 1 power: 1302W
[site  ] DEBUG 2025/02/17 17:33:45 grid power: -25W
[site  ] DEBUG 2025/02/17 17:33:45 battery 1 soc: 71%
[site  ] DEBUG 2025/02/17 17:33:45 pv 1 power: 104W
[site  ] DEBUG 2025/02/17 17:33:45 site power: 1277W
[lp-1  ] DEBUG 2025/02/17 17:33:45 charge total import: 313.641kWh
[lp-1  ] DEBUG 2025/02/17 17:33:45 charger status: B
[lp-1  ] ERROR 2025/02/17 17:33:46 switch phases: modbus: exception '3' (illegal data value), function '6'

hyperbart, 2025-02-17:
> [@hyperbart](https://github.com/hyperbart) Ich hab einen Smart und der hat den Cameleon Lader von der Renault Zoe drin da dürfen die Phasen nicht wärend dem Laden Umgestellt werden, im ACE Installer ist der Hacken gesetzt dass die ZE Eyrkennung aktiv ist.
> 
> Ich hab an der Alfen Wallbox nichts geändert. Home Assistant wird nicht verwendet
> 
> ```
> network:
>   schema: http
>   host: evcc.local # .local suffix announces the hostname on MDNS
>   port: 7070
> 
> log: debug
> levels:
>   cache: error
>  alfen: trace
> ```
> 
> ich hab die yaml geändert aber es kommt immer noch der gleiche Fehler. oder war das in der falschen Zeile?

Sorry, these settings are not the ones I was asking for.
Apologies to also mention I don't speak German fluently so I need to rely on Google Translate (that's not an issue, but some things might get lost in translation).

Please post screenshots from ACE installer for all these settings. (yes, all of them):

[image]

[image]

[image]

[image]

Steffen0707, 2025-02-17:
[image]
[image]
[image]

Es Funktionierte mit den Einstellungen vor demUpdate

VolkerK62, 2025-02-17:
>ich hab die yaml geändert aber es kommt immer noch der gleiche Fehler.
oder war das in der falschen Zeile?
```
levels:
  cache: error
 alfen: trace
```
Du musst auf die richtige Formatierung/Einrückung achten. Vor `alfen` fehlt ein Leerzeichen.

Steffen0707, 2025-02-17:
@VolkerK62 
```

log: debug
levels:
  cache: error
  alfen: trace

```
Hab es geändert, immer noch den gleichen Fehler

hyperbart, 2025-02-17:
@Steffen0707 let's try to fix your issue first and then we can investigate the why and what.
Do me a favor please; in your third screenshot enable the option "Allow single/multiphase charging".
Save, reboot station, check if setting has persisted and report back if EVCC is working again.

andig, 2025-02-17:
...and then lets find out if older evcc _REALLY_ works. Every bet- it won't.

Steffen0707, 2025-02-17:
Vielen Dank evcc Funkioniert wieder
"Allow single/multiphase charging" war das Problem
kann mir aber nicht Erklären warum der Hacken weg war

andig, 2025-02-17:
Immer wieder schön wenn das Update schuld ist 😂. Lesson learned: immer zuerst A/B Test machen (lassen). 
Jdfls gut, dass es geht!

closed 2025-02-17 by andig