evcc-io/evcc #18879: Tasmota-3p

homber, 2025-02-16:
### Describe the bug

Beim Auslesen der Daten von einem Tasmota HW-655 PZEM mit 3 Phasen als meter, werden die Daten Strom und Spannung (Current Voltage ) vertauscht.

### Steps to reproduce

1.evcc -c evcc.yaml -l trace  meter

2.
3.
...

### Configuration details

```yaml
- type: template
  template: tasmota-3p
  usage: charge
  host: 192.168.XXX.XX
  name: pv3
```

### Log details

```text
[tasmota] TRACE 2025/02/16 20:28:28 GET http://192.168.XXX.XX/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2025/02/16 20:28:28 {"StatusSNS":{"Time":"2025-02-16T21:28:28","ENERGY":{"TotalStartTime":"2024-08-14T20:17:44","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":[1,0,0],"ApparentPower":[5,0,0],"ReactivePower":[5,0,0],"Factor":[0.16,0.00,0.00],"Frequency":[50,50,50],"Voltage":[230,230,230],"Current":[0.022,0.000,0.000]}}}
Power:          1W
Energy:         0.0kWh
Current L1..L3: 230A 230A 230A
Voltage L1..L3: 0.022V 0V 0V
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.1

andig, 2025-02-16:
/cc @thierolm

andig, 2025-02-16:
Please add a trace log. Imho this looks like a device error.

homber, 2025-02-16:
hom@evcc:~$ evcc -c evcc.yaml -l trace  meter
[main  ] INFO 2025/02/16 20:37:42 evcc 0.200.1
[main  ] INFO 2025/02/16 20:37:42 using config file: evcc.yaml
[db    ] INFO 2025/02/16 20:37:42 using sqlite database: /home/hom/.evcc/evcc.db
[db    ] TRACE 2025/02/16 20:37:42 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2025/02/16 20:37:42 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/02/16 20:37:42 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/02/16 20:37:42 SELECT * FROM `settings` 1 <nil>
[db    ] TRACE 2025/02/16 20:37:42 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2025/02/16 20:37:42 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2025/02/16 20:37:42 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2025/02/16 20:37:42 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/02/16 20:37:42 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/02/16 20:37:42 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2025/02/16 20:37:42 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2025/02/16 20:37:42 SELECT * FROM `configs` WHERE `configs`.`class` = 2 0 <nil>
pv3
---
[tasmota] TRACE 2025/02/16 20:37:42 GET http://192.168.xxx.xx/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2025/02/16 20:37:42 {"StatusSNS":{"Time":"2025-02-16T21:37:42","ENERGY":{"TotalStartTime":"2024-08-14T20:17:44","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":[1,0,0],"ApparentPower":[5,0,0],"ReactivePower":[5,0,0],"Factor":[0.16,0.00,0.00],"Frequency":[50,50,50],"Voltage":[228,229,229],"Current":[0.022,0.000,0.000]}}}
Power:          1W
Energy:         0.0kWh
Current L1..L3: 228A 229A 229A
Voltage L1..L3: 0.022V 0V 0V

grid1
-----
[grid1 ] TRACE 2025/02/16 20:37:42 GET http://192.168.xxx.xxx/solar_api/v1/GetPowerFlowRealtimeData.fcgi
[grid1 ] TRACE 2025/02/16 20:37:42 {
   "Body" : {
      "Data" : {
         "Inverters" : {
            "1" : {
               "DT" : 46,
               "E_Day" : 23244,
               "E_Total" : 13577280,
               "E_Year" : 653443.3125,
               "P" : 0
            }
         },
         "Site" : {
            "E_Day" : 23244,
            "E_Total" : 13577280,
            "E_Year" : 653443.3125,
            "Meter_Location" : "grid",
            "Mode" : "meter",
            "P_Akku" : null,
            "P_Grid" : 4845.8999999999996,
            "P_Load" : -4845.8999999999996,
            "P_PV" : null,
            "rel_Autonomy" : 0,
            "rel_SelfConsumption" : null
         },
         "Version" : "12"
      }
   },
   "Head" : {
      "RequestArguments" : {},
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2025-02-16T21:37:42+01:00"
   }
}
Power: 4846W

pv2
---
[pv2   ] TRACE 2025/02/16 20:37:42 GET http://192.168.xxx.xxx/solar_api/v1/GetPowerFlowRealtimeData.fcgi
[pv2   ] TRACE 2025/02/16 20:37:42 {
   "Body" : {
      "Data" : {
         "Inverters" : {
            "1" : {
               "DT" : 46,
               "E_Day" : 23244,
               "E_Total" : 13577280,
               "E_Year" : 653443.3125,
               "P" : 0
            }
         },
         "Site" : {
            "E_Day" : 23244,
            "E_Total" : 13577280,
            "E_Year" : 653443.3125,
            "Meter_Location" : "grid",
            "Mode" : "meter",
            "P_Akku" : null,
            "P_Grid" : 4845.8999999999996,
            "P_Load" : -4845.8999999999996,
            "P_PV" : null,
            "rel_Autonomy" : 0,
            "rel_SelfConsumption" : null
         },
         "Version" : "12"
      }
   },
   "Head" : {
      "RequestArguments" : {},
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2025-02-16T21:37:42+01:00"
   }
}
Power: 0W

premultiply, 2025-02-16:
```json
{
  "StatusSNS": {
    "Time": "2025-02-16T21:28:28",
    "ENERGY": {
      "TotalStartTime": "2024-08-14T20:17:44",
      "Total": 0,
      "Yesterday": 0,
      "Today": 0,
      "Power": [
        1,
        0,
        0
      ],
      "ApparentPower": [
        5,
        0,
        0
      ],
      "ReactivePower": [
        5,
        0,
        0
      ],
      "Factor": [
        0.16,
        0,
        0
      ],
      "Frequency": [
        50,
        50,
        50
      ],
      "Voltage": [
        230,
        230,
        230
      ],
      "Current": [
        0.022,
        0,
        0
      ]
    }
  }
}
```

PR #18883 (2025-02-17): Tasmota: fix measurands mapping
Fixes #18879

/cc @thierolm
closed 2025-02-17 by andig