evcc-io/evcc #18887: Add Victron forecast

premultiply, 2025-02-17:
> nun habe ich in meinem victron system auch einen forecast. dieser ist "intelligent" und kennt mein system. dessen werte kann ich per mqtt bekommen und das sieht so aus:
> 
> {"success":true,"records":{"solar_yield_forecast":[[1739746800000,0],[1739750400000,0],[1739754000000,0],[1739757600000,0],[1739761200000,0],[1739764800000,0],[1739768400000,0],[1739772000000,24.774508580936654],[1739775600000,231.3302736580364],[1739779200000,595.8467250053252],[1739782800000,1284.9659761270182],[1739786400000,1870.8647525406734],[1739790000000,2142.1032451454953],[1739793600000,2046.5886287830062],[1739797200000,1709.7444892312578],[1739800800000,1246.5814579655641],[1739804400000,593.2500037191538],[1739808000000,72.94314385301772],[1739811600000,0],[1739815200000,0],[1739818800000,0],[1739822400000,0],[1739826000000,0],[1739829600000,0],[1739833200000,0],[1739836800000,0]]},"totals":{"solar_yield_forecast":11818.993204609484},"options":{}}
> 
> in die evcc.yaml habe ich mit hilfe von chatgpt folgendes rein:
> ```yaml annotate
> tariffs:
>   solar:
>     type: custom
>     forecast:
>       source: mqtt
>       topic: Victron_VRM/PVForecast
>       jq: |
>         [ .records.solar_yield_forecast[] | 
>           {
>           "start": (.[0] / 1000 | strftime("%FT%TZ")), # Millisekunden in Sekunden umwandeln und dann in String umwandeln
>           "end": ((.[0] + 3600000) / 1000 | strftime("%FT%TZ")), # Millisekunden +1Stunde für Endzeit, dann in Sekunden umwandeln und dann in String umwandeln
>           "price": (.[1] | tonumber | round) # Entfernt Dezimalstellen und gibt den Wert als Ganzzahl
>           } 
>         ] | tostring
> ```
> das hat ewig gedauert (weil ich ein noob bin und chatgpt bockig ist wie ein kleines kind dem man erklären muss das es eine jacke anziehen soll weils draußen -3 grad hat obwohl die sonne scheint...).
> funktioniert nun aber anscheinend.
> danke für den anstoß!
> 
> ich denke das ist auch eine gute hilfe für andere die ihren pv forecast aus dem PV system in evcc bringen wollen :)
> 
> [image]
>  

 _Originally posted by @fletch247 in [#18857](https://github.com/evcc-io/evcc/issues/18857#issuecomment-2661744865)_

andig, 2025-02-17:
Vmtl reicht auch

    mktime

Müsste aber getestet werden.

fletch247, 2025-02-17:
danke fürs splitten in ein separates issue.

wie oft wird die vorhersage in evcc denn aktualisiert?
der victron forecast ist sehr dynamisch über den tag und ich konnte eine altualisierung in evcc bisher nur durch einen evcc restart erzwingen.

noch kurz zum victron forecast setup allgemein:
damit das victron vrm den forecast per mqtt sendet braucht es natürlich noch ein paar schritte vorweg die nichts mit evcc zu tun haben. gerne kann ich details oder ein how to für die doku liefern, hier aber mal ganz grob die steps:

- generell muss auf dem gx device das large image mit node red laufen
- man braucht einen access token für sein VRM (kann man [hier](https://vrm.victronenergy.com/access-tokens) erstellen)
- im VRM API node in node red kann man alles konfigurieren
- dann als trigger einen timestamp node der zb. stündlich die daten aus dem vrm api node in einen mqtt out node übergibt 

(ich hoffe das ist hier im issue nicht fehl am platz)

andig, 2025-02-17:
Mit `interval` so oft du willst.

fletch247, 2025-02-17:
ist damit das interval aus der allgemeinen config gemeint?
wenn ja, dann aktualisiert der allgemeine evcc regelzyklus anscheinend nicht die forecast werte oder holt sie nicht erneut aus dem mqtt topic.

sorry wenn ixh da evtl. auf dem schlauch stehe oder doof frage 😅🙈

andig, 2025-02-17:
Das interval im custom Tarif!

Joeknx, 2025-02-17:
Cool, vielen Dank für die Anleitung!

Leider funktioniert das bei mir nicht und es kommt der Error:

```
[tariff] ERROR 2025/02/17 19:40:25 outdated
[main ] ERROR 2025/02/17 19:40:25 creating tariff solar failed: cannot create tariff type 'custom': outdated
``` 

Ich habe  die Version 0.200.1

Das mit dem Interval müsste dann vermutlich so aussehen:

```
solar:
    type: custom
    interval: 5m
    forecast:
      source: mqtt
      topic: victron/forecast
      jq: |
        [ .records.solar_yield_forecast[] | 
          {
          "start": (.[0] / 1000 | strftime("%FT%TZ")), # Millisekunden in Sekunden umwandeln und dann in String umwandeln
          "end": ((.[0] + 3600000) / 1000 | strftime("%FT%TZ")), # Millisekunden +1Stunde für Endzeit, dann in Sekunden umwandeln und dann in String umwandeln
          "price": (.[1] | tonumber | round) # Entfernt Dezimalstellen und gibt den Wert als Ganzzahl
          } 
        ] | tostring
```

andig, 2025-02-17:
> interval: 1800 #refresh wert in sekunden

das sind 1800ns

fletch247, 2025-02-17:
upsi aber das steht ja auch nirgendwo 🙈😅 
danke. habs geändert. hatte es abgeleitet vom interval des system allgemein.
sorry! das hat den armen alten mac mini die letzten 5h jetzt ganz schön gestresst 🙈
das hat man davon wenn man sich nicht auskennt und bei den großen mitreden will… 🤷‍♂️

andig, 2025-02-17:
> hatte es abgeleitet vom interval des system allgemein.

Das hat auch eine Einheit. Nicht ohne Grund 😊

Joeknx, 2025-02-18:
@fletch247 

mit welchen Werten fragst du denn die VRM Api in Node Red ab? 
Ich habe:
Interval: hours
Start: beginning of day
End: +24 hours

fletch247, 2025-02-18:
hilfe, ich bin doch nur ein user, kein programmierer und schon mit der auswahl compare changes überfordert beim PR erstellen 😅🙈 so gern ich es wollte, mehr input kann ich leider nicht geben 😕

Mikka20, 2025-02-27:
> [@fletch247](https://github.com/fletch247) jetzt brauchts doch nur noch ein Template- machst Du einen PR?

@andig : Gibt es ein ähnliches Template wo man sich was abgucken könnte? Würde mein Glück mal probieren ...

andig, 2025-02-27:
Klar- jedes andere Template mit einem `custom` Tarif

andig, 2025-03-10:
Seems nobody wants to add this. Closing.

closed 2025-03-10 by andig