evcc-io/evcc #18888: Ladevorgänge: zeigt "Ladepunkt" statt des Namens

cmock, 2025-02-17:
### Describe the bug

Für Ladevorgänge, die seit dem Upgrade auf 0.200.0 stattgefunden haben, wird in im UI in der Übersicht der Ladevorgänge in der Spalte "Ladepunkt" der Text "Ladepunkt" anstatt des konfigurierten Namens des Ladepunktes angezeigt.
Im CSV-Export der Ladevorgänge ist die Spalte für diese Vorgänge leer.

[image]

Ansonsten wird im UI der Name des LP korrekt angezeigt (Hauptansicht und "Konfiguration").

### Steps to reproduce

1. Laden
2. Schauen

### Configuration details

```yaml
sponsortoken: *****

network:
  schema: http
  host: rockpro.tahina.priv.at
  port: 7070

#log: debug
log: debug
levels:
  cache: error
#  smart-hello: trace
#  cfos: trace

# unique installation id
plant: *****

interval: 30s # control cycle interval

database:
  type: sqlite
  dsn: /rockpool/evcc/evcc.sqlite

mqtt:
  broker: raspi.tahina.priv.at:1883
  topic: evcc

telemetry: true

meters:
  - name: grid
    type: custom
    power: # W
      source: calc
      add:
        - source: mqtt
          topic: smartmeter/power
          jq: .P
          timeout: 10s
        - source: mqtt
          topic: smartmeter/power
          jq: .PO
          timeout: 10s
          scale: -1
    energy: # kWh
      source: calc
      add:
        - source: mqtt
          topic: smartmeter/energy
          jq: .tPI
          timeout: 10s
        - source: mqtt
          topic: smartmeter/energy
          jq: .tPO
          timeout: 10s
          scale: -1
    currents:
      - source: mqtt
        topic: smartmeter/power
        jq: .I1
        timeout: 10s
      - source: mqtt
        topic: smartmeter/power
        jq: .I2
        timeout: 10s
      - source: mqtt
        topic: smartmeter/power
        jq: .I3
        timeout: 10s
  - name: cfos_meter
    type: template
    template: cfos
    usage: charge
    modbus: tcpip
    id: 1
    host: 192.168.57.21
    port: 4701
  - name: pv_production
    type: custom
    power:
      source: mqtt
      topic: pv_inverter/P_out
      timeout: 35s
    energy:
      source: mqtt
      topic: pv_inverter/energy_total
      scale: 0.001
      timeout: 35s
    currents:
      - source: mqtt
        topic: pv_inverter/L1_I
        timeout: 35s
      - source: mqtt
        topic: pv_inverter/L2_I
        timeout: 35s
      - source: mqtt
        topic: pv_inverter/L3_I
        timeout: 35s
  - name: battery_meter
    type: template
    template: victron-energy
    usage: battery
    host: 192.168.57.171
    port: 502
    meterid: 0
    capacity: 28

site:
  title: Hausi
  meters:
    grid: grid
    pv:
      - pv_production
    battery:
      - battery_meter
  residualPower: 20

tariffs:
  currency: EUR
  grid:
    type: awattar
    region: at
    charges: 0.12428
    tax: 0.2
  feedin:
    type: fixed
    price: 0.06

loadpoints:
  - title: Auto
    charger: wallbox
    meter: cfos_meter
    priority: 2
  - title: Roller
    charger: roller_charger
    vehicle: roller
    priority: 1

chargers:
  - name: roller_charger
    type: template
    template: tasmota
    host: 192.168.57.12
    channel: 1
    standbypower: 1
  - name: wallbox
    type: template
    template: cfos
    host: 192.168.57.21

vehicles:
  - name: roller
    type: template
    template: offline
    title: Bimie Grazie
    capacity: 3
    phases: 1
    icon: moped
  - name: smart_1
    type: smart-hello
    title: Smart #1
    user: *****
    password: *****
    vin: *****
    capacity: 62
    icon: car
    identifiers:
      - *****
  - name: guest_ev
    type: template
    template: offline
    title: Gast-EV
    icon: generic
    identifiers:
      - 46973049

messaging:
  events:
    start:
      title: Laden gestartet
      msg: Laden gestartet, ${vehicleTitle} im Modus ${mode}
    stop:
      title: Laden beendet
      msg: Laden beendet, ${vehicleTitle}, ${chargedEnergy:%.1fk} kWh in ${chargeDuration}
  services:
    - type: ntfy
      uri: https://ntfy.sh/hs27-eequohF3

Version: `0.200.1 (b987ee14)`
```

### Log details

```text
n/a
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.1 (b987ee14)

StefanSchoof, 2025-02-17:
Siehe https://github.com/evcc-io/evcc/pull/18868

andig, 2025-02-17:
Es wäre schön das nightly auszuprobieren, nicht nur abzuhaken 👍🏻

closed 2025-02-17 by andig