evcc-io/evcc #18892: forecast.solar: kleinere Anzeigefehler

StevieC121176, 2025-02-17:
### Describe the bug

es gibt einige kleinere Anzeigefehler bei forecas.solar
- fehlende Werte im Tooltip
- in der Übersicht werden einige Uhrzeiten doppelt angezeigt (in meinem Beispiel "6")
- bei der Anzeige für Morgen steht "teilweise" - hier ist mir die Bedeutung von "teilweise" unklar

[image]
[image]

### Steps to reproduce

start forecast solar

### Configuration details

```yaml
solar:
    type: template
    template: forecast-solar
    lat: xx.7987
    lon: xx.2479
    dec: 8
    az: 40
    kwp: 16.77
```

### Log details

```text
[solar ] TRACE 2025/02/17 09:34:16 GET https://api.forecast.solar/estimate/xx.7987/xx.2479/8/40/16.77?time=utc&full=1&resolution=60
[solar ] TRACE 2025/02/17 09:34:16 {"result":{"watts":{"2025-02-16T23:00:00+00:00":0,"2025-02-17T00:00:00+00:00":0,"2025-02-17T01:00:00+00:00":0,"2025-02-17T02:00:00+00:00":0,"2025-02-17T03:00:00+00:00":0,"2025-02-17T04:00:00+00:00":0,"2025-02-17T05:00:00+00:00":0,"2025-02-17T05:45:00+00:00":31,"2025-02-17T06:00:00+00:00":255,"2025-02-17T07:00:00+00:00":1229,"2025-02-17T08:00:00+00:00":2891,"2025-02-17T09:00:00+00:00":4449,"2025-02-17T10:00:00+00:00":5663,"2025-02-17T11:00:00+00:00":6080,"2025-02-17T12:00:00+00:00":5669,"2025-02-17T13:00:00+00:00":4761,"2025-02-17T14:00:00+00:00":3384,"2025-02-17T15:00:00+00:00":1875,"2025-02-17T16:00:00+00:00":872,"2025-02-17T17:00:00+00:00":26,"2025-02-17T18:00:00+00:00":0,"2025-02-17T19:00:00+00:00":0,"2025-02-17T20:00:00+00:00":0,"2025-02-17T21:00:00+00:00":0,"2025-02-17T22:00:00+00:00":0,"2025-02-17T23:00:00+00:00":0,"2025-02-18T00:00:00+00:00":0,"2025-02-18T01:00:00+00:00":0,"2025-02-18T02:00:00+00:00":0,"2025-02-18T03:00:00+00:00":0,"2025-02-18T04:00:00+00:00":0,"2025-02-18T05:00:00+00:00":0,"2025-02-18T05:45:00+00:00":46,"2025-02-18T06:00:00+00:00":284,"2025-02-18T07:00:00+00:00":1270,"2025-02-18T08:00:00+00:00":2997,"2025-02-18T09:00:00+00:00":4723,"2025-02-18T10:00:00+00:00":5999,"2025-02-18T11:00:00+00:00":6529,"2025-02-18T12:00:00+00:00":6325,"2025-02-18T13:00:00+00:00":5602,"2025-02-18T14:00:00+00:00":4205,"2025-02-18T15:00:00+00:00":2419,"2025-02-18T16:00:00+00:00":1135,"2025-02-18T17:00:00+00:00":43,"2025-02-18T18:00:00+00:00":0,"2025-02-18T19:00:00+00:00":0,"2025-02-18T20:00:00+00:00":0,"2025-02-18T21:00:00+00:00":0,"2025-02-18T22:00:00+00:00":0,"2025-02-18T23:00:00+00:00":0},"watt_hours_period":{"2025-02-16T23:00:00+00:00":0,"2025-02-17T00:00:00+00:00":0,"2025-02-17T01:00:00+00:00":0,"2025-02-17T02:00:00+00:00":0,"2025-02-17T03:00:00+00:00":0,"2025-02-17T04:00:00+00:00":0,"2025-02-17T05:00:00+00:00":0,"2025-02-17T05:45:00+00:00":12,"2025-02-17T06:00:00+00:00":36,"2025-02-17T07:00:00+00:00":742,"2025-02-17T08:00:00+00:00":2060,"2025-02-17T09:00:00+00:00":3670,"2025-02-17T10:00:00+00:00":5056,"2025-02-17T11:00:00+00:00":5872,"2025-02-17T12:00:00+00:00":5875,"2025-02-17T13:00:00+00:00":5215,"2025-02-17T14:00:00+00:00":4073,"2025-02-17T15:00:00+00:00":2630,"2025-02-17T16:00:00+00:00":1374,"2025-02-17T17:00:00+00:00":449,"2025-02-17T18:00:00+00:00":13,"2025-02-17T19:00:00+00:00":0,"2025-02-17T20:00:00+00:00":0,"2025-02-17T21:00:00+00:00":0,"2025-02-17T22:00:00+00:00":0,"2025-02-17T23:00:00+00:00":0,"2025-02-18T00:00:00+00:00":0,"2025-02-18T01:00:00+00:00":0,"2025-02-18T02:00:00+00:00":0,"2025-02-18T03:00:00+00:00":0,"2025-02-18T04:00:00+00:00":0,"2025-02-18T05:00:00+00:00":0,"2025-02-18T05:45:00+00:00":17,"2025-02-18T06:00:00+00:00":41,"2025-02-18T07:00:00+00:00":777,"2025-02-18T08:00:00+00:00":2134,"2025-02-18T09:00:00+00:00":3860,"2025-02-18T10:00:00+00:00":5361,"2025-02-18T11:00:00+00:00":6264,"2025-02-18T12:00:00+00:00":6427,"2025-02-18T13:00:00+00:00":5964,"2025-02-18T14:00:00+00:00":4904,"2025-02-18T15:00:00+00:00":3312,"2025-02-18T16:00:00+00:00":1777,"2025-02-18T17:00:00+00:00":589,"2025-02-18T18:00:00+00:00":22,"2025-02-18T19:00:00+00:00":0,"2025-02-18T20:00:00+00:00":0,"2025-02-18T21:00:00+00:00":0,"2025-02-18T22:00:00+00:00":0,"2025-02-18T23:00:00+00:00":0},"watt_hours":{"2025-02-16T23:00:00+00:00":0,"2025-02-17T00:00:00+00:00":0,"2025-02-17T01:00:00+00:00":0,"2025-02-17T02:00:00+00:00":0,"2025-02-17T03:00:00+00:00":0,"2025-02-17T04:00:00+00:00":0,"2025-02-17T05:00:00+00:00":0,"2025-02-17T05:45:00+00:00":12,"2025-02-17T06:00:00+00:00":48,"2025-02-17T07:00:00+00:00":790,"2025-02-17T08:00:00+00:00":2850,"2025-02-17T09:00:00+00:00":6520,"2025-02-17T10:00:00+00:00":11576,"2025-02-17T11:00:00+00:00":17448,"2025-02-17T12:00:00+00:00":23323,"2025-02-17T13:00:00+00:00":28538,"2025-02-17T14:00:00+00:00":32611,"2025-02-17T15:00:00+00:00":35241,"2025-02-17T16:00:00+00:00":36615,"2025-02-17T17:00:00+00:00":37064,"2025-02-17T18:00:00+00:00":37077,"2025-02-17T19:00:00+00:00":37077,"2025-02-17T20:00:00+00:00":37077,"2025-02-17T21:00:00+00:00":37077,"2025-02-17T22:00:00+00:00":37077,"2025-02-17T23:00:00+00:00":0,"2025-02-18T00:00:00+00:00":0,"2025-02-18T01:00:00+00:00":0,"2025-02-18T02:00:00+00:00":0,"2025-02-18T03:00:00+00:00":0,"2025-02-18T04:00:00+00:00":0,"2025-02-18T05:00:00+00:00":0,"2025-02-18T05:45:00+00:00":17,"2025-02-18T06:00:00+00:00":58,"2025-02-18T07:00:00+00:00":835,"2025-02-18T08:00:00+00:00":2969,"2025-02-18T09:00:00+00:00":6829,"2025-02-18T10:00:00+00:00":12190,"2025-02-18T11:00:00+00:00":18454,"2025-02-18T12:00:00+00:00":24881,"2025-02-18T13:00:00+00:00":30845,"2025-02-18T14:00:00+00:00":35749,"2025-02-18T15:00:00+00:00":39061,"2025-02-18T16:00:00+00:00":40838,"2025-02-18T17:00:00+00:00":41427,"2025-02-18T18:00:00+00:00":41449,"2025-02-18T19:00:00+00:00":41449,"2025-02-18T20:00:00+00:00":41449,"2025-02-18T21:00:00+00:00":41449,"2025-02-18T22:00:00+00:00":41449,"2025-02-18T23:00:00+00:00":0},"watt_hours_day":{"2025-02-17":37077,"2025-02-18":41449}},"message":{"code":0,"type":"success","text":"","pid":"q858n5HL","info":{"latitude":xx.7988,"longitude":xx.2477,"distance":0.018,"place":"xx, xxxxx, xxxxx, xxxxx, xxxxx, xxxxx, xxxxx, Deutschland","timezone":"Europe/Berlin","time":"2025-02-17T09:34:16+01:00","time_utc":"2025-02-17T08:34:16+00:00"},"ratelimit":{"zone":"IP 2a02:810a:c80:bf00:d612:fe6a:1e53:996f","period":3600,"limit":12,"remaining":10}}}
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.1 (b987ee14)

naltatis, 2025-02-17:
@StevieC121176 poste mal bitte das Ergebnis von `http://[evcc]:7070/api/state?jq=forecast` hier.

tantive, 2025-02-17:
Ich sehe bei mir mit forecast.solar als Anbieter ebenfalls doppelte Einträge bei 6 Uhr.

Das Ergebnis von http://[evcc]:7070/api/state?jq=.forecast zeigt bei mir unter 'solar' folgende Einträge zwischen 6 und 8 Uhr:

 {
      "start": "2025-02-17T06:00:00+01:00",
      "end": "2025-02-17T07:00:00+01:00",
      "price": 0
    },
    {
      "start": "2025-02-17T06:45:00+01:00",
      "end": "2025-02-17T07:45:00+01:00",
      "price": 15
    },
    {
      "start": "2025-02-17T07:00:00+01:00"
      "end": "2025-02-17T08:00:00+01:00",
      "price": 241
    },

julred, 2025-02-17:
Ich habe bei mir auch mal forecast.solar in die yaml eingetragen.
Wo kann ich denn die Übersicht einsehen - wie im Screenshot?

StevieC121176, 2025-02-17:
> Ich sehe bei mir mit forecast.solar als Anbieter ebenfalls doppelte Einträge bei 6 Uhr.
>
> Das Ergebnis von http://[evcc]:7070/api/state?jq=.forecast zeigt bei mir unter 'solar' folgende Einträge [...]

komisch, bei mir kommt da ein Error, obwohl die forecast Anzeige korrekt dargestellt wird, bis auf die doppelten 6 Uhr Eintrag.

tantive, 2025-02-17:
@StevieC121176 In der ersten URL von @naltatis fehlt ein Punkt vor forecast. Das hier funktioniert: http://[evcc]:7070/api/state?jq=.forecast

StevieC121176, 2025-02-17:
> Ich habe bei mir auch mal forecast.solar in die yaml eingetragen. Wo kann ich denn die Übersicht einsehen - wie im Screenshot?

oben rechts auf die 3 Striche (Menü) und dann den Punkt Vorhersage auswählen.

StevieC121176, 2025-02-17:
> [@StevieC121176](https://github.com/StevieC121176) In der ersten URL von [@naltatis](https://github.com/naltatis) fehlt ein Punkt vor forecast. Das hier funktioniert: http://[evcc]:7070/api/state?jq=.forecast

vielen Dank, daran lag es.

`{"co2":[{"end":"2025-02-17T13:00:00+01:00","price":380,"start":"2025-02-17T12:00:00+01:00"},{"end":"2025-02-17T14:00:00+01:00","price":100,"start":"2025-02-17T13:00:00+01:00"},{"end":"2025-02-17T15:00:00+01:00","price":103,"start":"2025-02-17T14:00:00+01:00"},{"end":"2025-02-17T16:00:00+01:00","price":103,"start":"2025-02-17T15:00:00+01:00"},{"end":"2025-02-17T17:00:00+01:00","price":100,"start":"2025-02-17T16:00:00+01:00"},{"end":"2025-02-17T18:00:00+01:00","price":103,"start":"2025-02-17T17:00:00+01:00"},{"end":"2025-02-17T19:00:00+01:00","price":145,"start":"2025-02-17T18:00:00+01:00"},{"end":"2025-02-17T20:00:00+01:00","price":224,"start":"2025-02-17T19:00:00+01:00"},{"end":"2025-02-17T21:00:00+01:00","price":221,"start":"2025-02-17T20:00:00+01:00"},{"end":"2025-02-17T22:00:00+01:00","price":221,"start":"2025-02-17T21:00:00+01:00"},{"end":"2025-02-17T23:00:00+01:00","price":217,"start":"2025-02-17T22:00:00+01:00"},{"end":"2025-02-18T00:00:00+01:00","price":210,"start":"2025-02-17T23:00:00+01:00"},{"end":"2025-02-18T01:00:00+01:00","price":228,"start":"2025-02-18T00:00:00+01:00"},{"end":"2025-02-18T02:00:00+01:00","price":228,"start":"2025-02-18T01:00:00+01:00"},{"end":"2025-02-18T03:00:00+01:00","price":231,"start":"2025-02-18T02:00:00+01:00"},{"end":"2025-02-18T04:00:00+01:00","price":224,"start":"2025-02-18T03:00:00+01:00"},{"end":"2025-02-18T05:00:00+01:00","price":228,"start":"2025-02-18T04:00:00+01:00"},{"end":"2025-02-18T06:00:00+01:00","price":238,"start":"2025-02-18T05:00:00+01:00"},{"end":"2025-02-18T07:00:00+01:00","price":238,"start":"2025-02-18T06:00:00+01:00"},{"end":"2025-02-18T08:00:00+01:00","price":242,"start":"2025-02-18T07:00:00+01:00"},{"end":"2025-02-18T09:00:00+01:00","price":204,"start":"2025-02-18T08:00:00+01:00"},{"end":"2025-02-18T10:00:00+01:00","price":93,"start":"2025-02-18T09:00:00+01:00"},{"end":"2025-02-18T11:00:00+01:00","price":55,"start":"2025-02-18T10:00:00+01:00"},{"end":"2025-02-18T12:00:00+01:00","price":55,"start":"2025-02-18T11:00:00+01:00"},{"end":"2025-02-18T13:00:00+01:00","price":58,"start":"2025-02-18T12:00:00+01:00"},{"end":"2025-02-18T14:00:00+01:00","price":69,"start":"2025-02-18T13:00:00+01:00"},{"end":"2025-02-18T15:00:00+01:00","price":82,"start":"2025-02-18T14:00:00+01:00"},{"end":"2025-02-18T16:00:00+01:00","price":69,"start":"2025-02-18T15:00:00+01:00"},{"end":"2025-02-18T17:00:00+01:00","price":65,"start":"2025-02-18T16:00:00+01:00"},{"end":"2025-02-18T18:00:00+01:00","price":72,"start":"2025-02-18T17:00:00+01:00"},{"end":"2025-02-18T19:00:00+01:00","price":127,"start":"2025-02-18T18:00:00+01:00"},{"end":"2025-02-18T20:00:00+01:00","price":231,"start":"2025-02-18T19:00:00+01:00"},{"end":"2025-02-18T21:00:00+01:00","price":238,"start":"2025-02-18T20:00:00+01:00"},{"end":"2025-02-18T22:00:00+01:00","price":249,"start":"2025-02-18T21:00:00+01:00"},{"end":"2025-02-18T23:00:00+01:00","price":256,"start":"2025-02-18T22:00:00+01:00"},{"end":"2025-02-19T00:00:00+01:00","price":262,"start":"2025-02-18T23:00:00+01:00"},{"end":"2025-02-19T01:00:00+01:00","price":259,"start":"2025-02-19T00:00:00+01:00"},{"end":"2025-02-19T02:00:00+01:00","price":262,"start":"2025-02-19T01:00:00+01:00"},{"end":"2025-02-19T03:00:00+01:00","price":256,"start":"2025-02-19T02:00:00+01:00"},{"end":"2025-02-19T04:00:00+01:00","price":231,"start":"2025-02-19T03:00:00+01:00"},{"end":"2025-02-19T05:00:00+01:00","price":228,"start":"2025-02-19T04:00:00+01:00"},{"end":"2025-02-19T06:00:00+01:00","price":245,"start":"2025-02-19T05:00:00+01:00"},{"end":"2025-02-19T07:00:00+01:00","price":259,"start":"2025-02-19T06:00:00+01:00"},{"end":"2025-02-19T08:00:00+01:00","price":269,"start":"2025-02-19T07:00:00+01:00"},{"end":"2025-02-19T09:00:00+01:00","price":238,"start":"2025-02-19T08:00:00+01:00"},{"end":"2025-02-19T10:00:00+01:00","price":169,"start":"2025-02-19T09:00:00+01:00"},{"end":"2025-02-19T11:00:00+01:00","price":138,"start":"2025-02-19T10:00:00+01:00"},{"end":"2025-02-19T12:00:00+01:00","price":100,"start":"2025-02-19T11:00:00+01:00"},{"end":"2025-02-19T13:00:00+01:00","price":89,"start":"2025-02-19T12:00:00+01:00"},{"end":"2025-02-19T14:00:00+01:00","price":100,"start":"2025-02-19T13:00:00+01:00"},{"end":"2025-02-19T15:00:00+01:00","price":120,"start":"2025-02-19T14:00:00+01:00"},{"end":"2025-02-19T16:00:00+01:00","price":141,"start":"2025-02-19T15:00:00+01:00"},{"end":"2025-02-19T17:00:00+01:00","price":148,"start":"2025-02-19T16:00:00+01:00"},{"end":"2025-02-19T18:00:00+01:00","price":138,"start":"2025-02-19T17:00:00+01:00"},{"end":"2025-02-19T19:00:00+01:00","price":159,"start":"2025-02-19T18:00:00+01:00"},{"end":"2025-02-19T20:00:00+01:00","price":228,"start":"2025-02-19T19:00:00+01:00"},{"end":"2025-02-19T21:00:00+01:00","price":242,"start":"2025-02-19T20:00:00+01:00"},{"end":"2025-02-19T22:00:00+01:00","price":273,"start":"2025-02-19T21:00:00+01:00"},{"end":"2025-02-19T23:00:00+01:00","price":256,"start":"2025-02-19T22:00:00+01:00"},{"end":"2025-02-20T00:00:00+01:00","price":231,"start":"2025-02-19T23:00:00+01:00"},{"end":"2025-02-20T01:00:00+01:00","price":193,"start":"2025-02-20T00:00:00+01:00"},{"end":"2025-02-20T02:00:00+01:00","price":193,"start":"2025-02-20T01:00:00+01:00"},{"end":"2025-02-20T03:00:00+01:00","price":172,"start [... cut ...] :00"},{"end":"2025-02-18T20:00:00+01:00","price":0,"start":"2025-02-18T19:00:00+01:00"},{"end":"2025-02-18T21:00:00+01:00","price":0,"start":"2025-02-18T20:00:00+01:00"},{"end":"2025-02-18T22:00:00+01:00","price":0,"start":"2025-02-18T21:00:00+01:00"},{"end":"2025-02-18T23:00:00+01:00","price":0,"start":"2025-02-18T22:00:00+01:00"},{"end":"2025-02-19T00:00:00+01:00","price":0,"start":"2025-02-18T23:00:00+01:00"},{"end":"2025-02-19T01:00:00+01:00","price":0,"start":"2025-02-19T00:00:00+01:00"}]}`

naltatis, 2025-02-17:
@tantive Der hier ist komisch verschoben:
```
{
"start": "2025-02-17T06:45:00+01:00",
"end": "2025-02-17T07:45:00+01:00",
"price": 15
},
```

Erwartet wäre hier 7:00 bis 8:00. Das würde auf jeden Fall den Effekt erklären. Sieht aus wie ein Bug in der "forecast.solar" Antwort.

naltatis, 2025-02-17:
Bitte mal das `http` Loglevel auf `trace` drehen:

```
levels:
  http: trace
```

Beim nächsten Mal wenn das auftritt den Request und die Antwort hier posten bzw. am besten direkt bei https://forecast.solar (Discord, Mail) melden. Mit den aktuellen Parametern (`?time=utc&full=1&resolution=60`) erwarten wir einheitliche Stundenslots.

StevieC121176, 2025-02-17:
> [@tantive](https://github.com/tantive) Der hier ist komisch verschoben:
>
> ```
> {
> "start": "2025-02-17T06:45:00+01:00",
> "end": "2025-02-17T07:45:00+01:00",
> "price": 15
> },
> ```
>
> Erwartet wäre hier 7:00 [...]

bei mir findet sich auch so ein komischer Eintrag:

`17T07:00:00+01:00","price":0,"start":"2025-02-17T06:00:00+01:00"},{"end":"2025-02-17T07:45:00+01:00","price":32,"start":"2025-02-17T06:45:00+01:00"},{"end":"2025-02-17T08:00:00+01:00","price":260,"start":"2025-02-17T07:00:00+01:00"},{"end":"2025-02-`

naltatis, 2025-02-17:
> ...und ich würde gerne nochmal die X-Achse ins Rennen werfen :)

@andig keine Angst, den Kommentar hab ich nicht vergessen.

tantive, 2025-02-17:
Ok, evtl. ist die Lösung doch einfacher als gedacht:

``` curl 'https://api.forecast.solar/estimate/52/12/37/0/5.67?time=utc&full=1&resolution=60&format=start-end'```
```

Gibt zumindest mal definierte Zeitbereiche zurück inkl. start und end.

Man müsste dann trotzdem noch mit jq innerhalb ganzer Stunden aufsummieren, korrekt?

tantive, 2025-02-17:
Asked the forecast.solar team in discord (invite: [Discord](https://discord.gg/AuqcSPtqKh)):
https://discord.com/channels/898084609017925633/906286197217820712

tantive — Today at 9:01 PM
Hi there, evcc (evcc.io) started to support forecast.solar, during first test an unexpected behavior in the forecast api response was observed 
Assumtion was, that using the parameters '?time=utc&full=1&resolution=60' exactly one entry per full hour would be returned
This is almost true, except for sunrise, where the 'start' timestamp denotes the actual sunrise
Is there a way to retrieve one entry per full hour?
https://github.com/evcc-io/evcc/issues/18892
GitHub
[forecast.solar: kleinere Anzeigefehler · Issue #18892 · evcc-io/evcc](https://github.com/evcc-io/evcc/issues/18892)
Describe the bug es gibt einige kleinere Anzeigefehler bei forecast.solar in der Übersicht werden einige Uhrzeiten doppelt angezeigt (in meinem Beispiel "6") bei der Anzeige für Morgen st...

Knut — Today at 9:05 PM
I'll check, full=1 should remove sunrise/sunset

tantive — Today at 9:07 PM
"2025-02-17T05:00:00+00:00":0,"2025-02-17T05:45:00+00:00":14,"2025-02-17T06:00:00+00:00":113,

kk-c, 2025-02-18:
Knut here,

_Caused by combination of `full=1` with a `resolution=...`, fixed now in v7.31.2_

closed 2025-02-18 by andig