evcc-io/evcc #18894: Keine Anzeige mehr unter iOS 12.5.7

HeikoH78, 2025-02-17:
### Describe the bug

Follow up aus #16931 

Bug tritt seit dem Update auf 0.200.1 wieder auf, ich bin sicher, dass auf 0.200.0 der Fehler nicht auftrat.

iOS Version ist 12.5.7, weitere Geräte funktionieren einwandfrei.

### Steps to reproduce

EVCC-Seite auf Safari und Firefox öffnen

### Configuration details

```yaml
keine Veränderung seit #16931
```

### Log details

```text
...
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.1

naltatis, 2025-02-17:
@HeikoH78 Weißt du ob man mit dem iPad die Ladevorgänge (mit Charts) aufrufen konnte?

HeikoH78, 2025-02-18:
@naltatis Nein, leider nicht, ich kannte die Übersicht mit den Charts noch nicht mal ;)

julian2701, 2025-02-19:
Kann den Fehler bestätigen, habe ebenfalls ein iPad mini 3 mit iOS 12.5.7 (mehr geht beim iPad mini 3 nicht) und kann seit 0.200.1 auch den Hauptscreen nicht mehr sehen. Die Diagramme waren bereits unter 0.1XX.X nicht darstellbar, was mich aber nur wenig gestört hat, da ich das iPad am Hauseingang primär zur Steuerung und zur "Auswertung" genutzt habe.

Gibt es eine Lösung bzw. Möglichkeit hier wieder zum Stand von 0.200.0 zurückzukehren bzw. kennt jemand die Ursache, die man im Zweifel selbst beheben und dann neu kompilieren kann? Eine Darstellung unter iOS 12.5.7 wäre klasse und würde mich davor bewahren ein neues Tablet inkl. Wandhalterung anzuschaffen ;)

julian2701, 2025-02-19:
> [@HeikoH78](https://github.com/HeikoH78) Weißt du ob man mit dem iPad (vor 0.200.1) die Ladevorgänge (mit Charts) aufrufen konnte?

Negativ, die Charts waren auch vorher schon nicht darstellbar, s. meine Antwort unten.

hebe2k, 2025-02-21:
@naltatis Ich habe dasselbe Problem auf meinem alten iPad mit iOS 12.5.7.

Wie @julian2701 schon geschrieben hat, die Ladevorgänge mit Charts funktionierten auch vor der 0.200.x schon nicht mehr), aber jetzt mit der 0.200.1 funktioniert bei mir auch gar nichts mehr, es kommt nur noch die leere weiße Seite wenn man evcc aufruft.
Die 0.200.0 habe ich wegen des Issues "Sessions: fix missing loadpoint names (#18868)" übersprungen.

Gibt es irgendetwas was wir zur Lösungsfindung und Analyse beitragen können?

naltatis, 2025-02-21:
Fehler gefunden. Danke für die Hinweise.

closed 2025-02-21 by naltatis