evcc-io/evcc #18896: Sungrow SHT10 mit Retro Fit / 2. Stromwandler Satz für 3. Party Inverter (ABB/Fimer) wird nicht richtig dargestellt

Bakari2020, 2025-02-17:
EVCC-Zeigt aktuell nur den Strom des Sungrow Inverters an, der 2. Inverter über Wandlersatzt 2 bzw. Retrofit Konfiguration wird nicht angezeigt.

[image]

[image]

Einbindung einer Konfiguration für SH Sungrow Inverter mit Retrofit Konfiguration oder DTSU666-20 

> [site ] DEBUG 2025/02/17 10:47:33 grid power: -295W
[db:7 ] TRACE 2025/02/17 10:47:33 modbus: send 00 11 00 00 00 06 01 04 15 e2 00 02
[db:7 ] TRACE 2025/02/17 10:47:33 modbus: recv 00 11 00 00 00 07 01 04 04 ff ff ff ff
[db:5 ] TRACE 2025/02/17 10:47:33 modbus: send 00 12 00 00 00 06 01 04 13 98 00 02
[db:5 ] TRACE 2025/02/17 10:47:33 modbus: recv 00 12 00 00 00 07 01 04 04 0d 04 00 00
[site ] DEBUG 2025/02/17 10:47:33 pv 1 power: 3332W
[db:6 ] TRACE 2025/02/17 10:47:33 modbus: send 00 13 00 00 00 06 01 04 32 c8 00 01
[db:6 ] TRACE 2025/02/17 10:47:33 modbus: recv 00 13 00 00 00 05 01 04 02 00 00
[db:7 ] TRACE 2025/02/17 10:47:33 modbus: send 00 14 00 00 00 06 01 04 13 9a 00 01
[db:7 ] TRACE 2025/02/17 10:47:33 modbus: recv 00 14 00 00 00 05 01 04 02 08 fd
[db:5 ] TRACE 2025/02/17 10:47:33 modbus: send 00 15 00 00 00 06 01 04 32 ca 00 02
[db:5 ] TRACE 2025/02/17 10:47:33 modbus: recv 00 15 00 00 00 07 01 04 04 a3 87 00 00
[db:6 ] TRACE 2025/02/17 10:47:33 modbus: send 00 16 00 00 00 06 01 04 32 dc 00 01
[db:6 ] TRACE 2025/02/17 10:47:33 modbus: recv 00 16 00 00 00 05 01 04 02 fe f7
[db:7 ] TRACE 2025/02/17 10:47:33 modbus: send 00 17 00 00 00 06 01 04 15 e4 00 02
[db:7 ] TRACE 2025/02/17 10:47:33 modbus: recv 00 17 00 00 00 07 01 04 04 ff 7d ff ff
[db:7 ] TRACE 2025/02/17 10:47:33 modbus: send 00 18 00 00 00 06 01 04 13 9b 00 01
[db:7 ] TRACE 2025/02/17 10:47:33 modbus: recv 00 18 00 00 00 05 01 04 02 09 39
[db:6 ] TRACE 2025/02/17 10:47:33 modbus: send 00 19 00 00 00 06 01 04 32 dd 00 01
[db:6 ] TRACE 2025/02/17 10:47:33 modbus: recv 00 19 00 00 00 05 01 04 02 15 9b
[site ] DEBUG 2025/02/17 10:47:33 battery 1 power: -5531W
[db:7 ] TRACE 2025/02/17 10:47:33 modbus: send 00 1a 00 00 00 06 01 04 15 e6 00 02
[db:7 ] TRACE 2025/02/17 10:47:33 modbus: recv 00 1a 00 00 00 07 01 04 04 ff 63 ff ff
[db:7 ] TRACE 2025/02/17 10:47:33 modbus: send 00 1b 00 00 00 06 01 04 13 9c 00 01
[db:7 ] TRACE 2025/02/17 10:47:33 modbus: recv 00 1b 00 00 00 05 01 04 02 09 1a
[site ] DEBUG 2025/02/17 10:47:33 grid currents: [-0.00435 -0.555 -0.674]A
[db:6 ] TRACE 2025/02/17 10:47:33 modbus: send 00 1c 00 00 00 06 01 04 32 e2 00 02
[db:6 ] TRACE 2025/02/17 10:47:33 modbus: recv 00 1c 00 00 00 07 01 04 04 33 4e 00 00
[db:7 ] TRACE 2025/02/17 10:47:33 modbus: send 00 1d 00 00 00 06 01 04 32 ec 00 02
[db:7 ] TRACE 2025/02/17 10:47:33 modbus: recv 00 1d 00 00 00 07 01 04 04 16 f5 00 00
[db:6 ] TRACE 2025/02/17 10:47:33 modbus: send 00 1e 00 00 00 06 01 04 32 de 00 01
[db:6 ] TRACE 2025/02/17 10:47:33 modbus: recv 00 1e 00 00 00 05 01 04 02 01 d0
[site ] DEBUG 2025/02/17 10:47:33 battery 1 soc: 46%
[site ] DEBUG 2025/02/17 10:47:33 battery has priority at soc 46% (< 90%)
[site ] DEBUG 2025/02/17 10:47:33 site power: -195W
> 

**Additional context**
Ich kenne die Modbusprotokolle nicht ausreichen um sagen zu können ob der Chint DTSU666-20 eigene Daten zur Verfügung stellt die lesbar sind, eine Option wäre eine Konfiguration über RS485 mit Konverter.

Bzw. wären freie Chint DTSU Konfigurationen über RS485 to TCP cool 
Abhilfe würde bei mir aber auch eine Konfiguration von Fimer / ABB Trio Wechselrichtern über RS485 to TCP schaffen.

StefanSchoof, 2025-02-17:
Ist das https://github.com/evcc-io/evcc/issues/18838?

Bakari2020, 2025-02-17:
> Ist das [#18838](https://github.com/evcc-io/evcc/issues/18838)?

ich denke da handelt es sich um externes Netzlade über Tibber, wenn es dort nicht mit Tibber geladen wird handelt es sich dort auch um eine Sungrow Anlage die aber nicht auf Retrofit Konfiguriert ist und keinen 2. Wandlersatz hat.

Es gibt die möglichkeit das die SH so konfiguriert werden dass sie jeglichen Strom der zusätzlich über den DTSU negativeingespeißt wird als Ertrag einfach zum gesamt Ertrag hinzurechnen das es in der iSolarcloud richtig angezeigt wird, das ist aber nur eine suboptimale Lösung.

Die einzig wahre Lösung ist von DTSU666 Inline oder indirekte Messung mit einem Wandlersatzt den DTSU666-20 zu verbauen dieser hat dann 2 Wandlersätze für den Sungrow und einen für den Fremdinverter. 

Isolarcloud hat seperate Werte für den 2. Wandlersatz:
Wandlersatz 1 Hauptmessstelle

[image]

Wandlersatz 2 Fremdwechselrichter

[image]

premultiply, 2025-02-17:
Könnte ggf. hier ab Register 5602 zu finden sein?

[image]

Probier mal als custom meter aus.

Bakari2020, 2025-02-18:
> Könnte ggf. hier ab Register 5602 zu finden sein?
> 
> [image]
> 
> Probier mal als custom meter aus.

Das sind die Hauptphasen von Wandler Satz 1 das spiegelt nur solange den Wert vom 2. Wechselrichter richtig wieder solange auch die Batterie über AC zu DC geladen wird. 

[image]

Sobald Netzennahme / Batterieladung über AC stattfindet ist der Wert dann auch um den passiven Stromverbrauch falsch.

Es gibt im Chint 2 Register bei der ich für 2 aber nicht das RTU Register kenne bzw. rausgefunden habe:

[image]

[image]

premultiply, 2025-02-27:
Ich kann dir ehrlich gesagt nicht ganz folgen.

Kannst mal eine Skizze zeichnen was da wo genau elektrisch angeschlossen ist und wie und durch was gemessen werden soll?

Bakari2020, 2025-03-14:
[image]

closed 2025-04-08 by github-actions[bot]