evcc-io/evcc #18903: Forecast Solar - viel zu niedrige Prognose

theoldphilip, 2025-02-17:
### Describe the bug

Sobald ich Forecast Solar inkl. API Key einbinde, wird mir eine Prognose gezeigt. Diese fällt aber viel zu gering aus.
Ich habe die selbe API auch in HomeAssistant eingebunden mit ganz anderen Werten.

Beispiel heute:

EVCC Integration: 2,9 kWh verbleiben
HomeAssistant: 18.1 kWh

[image]

[image]

`solar:
  type: template
  template: forecast-solar
  lat: 51.669491
  lon: 7.693407
  dec: 29 # 0 = horizontal, 90 = vertikal
  az: 160 # -180 = Norden, -90 = Osten, 0 = Süden, 90 = Westen, 180 = Norden
  kwp: 6
  apikey: XXXXXXXXXXXXXXXX`

und in HA

[image]

### Steps to reproduce

1. Lässt sich immer wieder durch das eintragen der Parameter reproduzieren
2.
3.
...

### Configuration details

```yaml
Hab ich oben angehängt
```

### Log details

```text
keine Auffälligkeiten beim Start
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

200.1

goebelmeier, 2025-02-17:
Du musst `-20` bei `az` eintragen. In HA wird der Azimuth anders ausgelegt:

> As a full circle is 360°, a value of 0 is facing North, 90 East, 180 South and 270 is facing West (or any value in between).

theoldphilip, 2025-02-17:
@goebelmeier Danke dir, Schande auf mein Haupt 👍

closed 2025-02-17 by theoldphilip