evcc-io/evcc #18916: Add the Cupra Tavascan to documentaion about CupraConnect Gen4

lars-enytime, 2025-02-17:
**Is your feature request related to a problem? Please describe.**
The documentation (https://docs.evcc.io/docs/devices/vehicles#cupraconnect-gen4-born) only lists the Born as being compatible with CupraConnect Gen4. I successfully added a Cupra Tavascan with the cupra template.

**Describe the solution you'd like**
It should also list the Tavascan in the title

**Describe alternatives you've considered**
-

**Additional context**
The tavascan seems to be fully supported as it can be seen in this screenshot:

[image]

0x3d13f, 2025-02-17:
Die Doku ist auch Open Source, Du kannst Änderungen über den Link "Diese Seite bearbeiten" ganz unten selbst einreichen.

andig, 2025-02-17:
Genau genommen muss das im Cupra Template ergänzt werden. Gerne PR!

closed 2025-02-17 by andig