evcc-io/evcc #18929: Ostrom - errors when configuring

ibuprofen600, 2025-02-18:
### Describe the bug

Adding Ostrom as documented triggers an error:  "Error during startup. Check your configuration and restart.
tariff [grid] cannot create tariff type 'template': invalid key: price"

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
![Image](https://github.com/user-attachments/assets/6dd893d4-1630-4886-b839-3d22c2676eb5)
```

### Log details

```text
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/02/18 09:50:41 evcc 0.200.1
[main  ] INFO 2025/02/18 09:50:41 using config file: /config/evcc.yaml
[db    ] INFO 2025/02/18 09:50:41 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2025/02/18 09:50:41 connecting evcc-2044051639 at tcp://core-mosquitto.local.hass.io:1883
[mqtt  ] DEBUG 2025/02/18 09:50:41 tcp://core-mosquitto.local.hass.io:1883 connected
[main  ] INFO 2025/02/18 09:50:41 listening at :7070
[main  ] FATAL 2025/02/18 09:50:41 tariff tariff [grid] cannot create tariff type 'template': invalid key: price
[main  ] FATAL 2025/02/18 09:50:41 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.1

ibuprofen600, 2025-02-18:
a bit of online research shows other having success merging 'type' and 'template' like this:

[image]

this will cause a different error for me:

"Error during startup. Check your configuration and restart.
tariff [grid] cannot create tariff type 'ostrom': decoding failed due to the following error(s): '' has invalid keys: price"

andig, 2025-02-18:
Pls share trace log

ibuprofen600, 2025-02-18:
> Pls share trace log

[evcc-20250218-130234-trace.log](https://github.com/user-attachments/files/18844832/evcc-20250218-130234-trace.log)

attached

ibuprofen600, 2025-02-19:
@andig actually, I was using the (experimental) UI configuration to change from a fixed tariff to Ostrom (comment fixed out, add Ostrom in). Doing it in this way was causing the conflict. Once I went and edited the evcc.yaml file under /addon_configs/49686a9f_evcc/ manually, the issue got resolved (my evcc lives in Homeassistant)