evcc-io/evcc #18933: Sofar HYD-20KTL Second Battery Block seen as consumer

ep4711, 2025-02-18:
### Describe the bug

I installed a second Sofar Battery Block to my Sofar inverter. In evcc this block is seen as a consumer, not a battery. The solarman app can differenciate. Nevertheless the battery SOC ist reported right (total battery load)
any suggestions? Thanks!

### Steps to reproduce

1. Compare Solarman values vs evcc values
2.
3.
...

### Configuration details

```yaml
meters:
  - name: sofar-grid
    type: template
    template: sofarsolar-g3
    usage: grid
   
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.XXX.YY
    port: 8899
 
  - name: sofar-pv
    type: template
    template: sofarsolar-g3
    usage: pv
   
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.XXX.YY
    port: 8899
 
  - name: sofar-battery-1
    type: template
    template: sofarsolar-g3
    usage: battery
   
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.XXX.YY
    port: 8899
    capacity: 30
```

### Log details

```text
no log problem
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.1

ep4711, 2025-02-23:
> Do you need to configure the second battery separetely?

good point! But how do I differentiate the two blocks in the config? Can the authors of the Sofar integration help here?

andig, 2025-02-23:
Seems we're missing a storageunit number. Maybe @cschlipf has an idea?

cschlipf, 2025-02-23:
What I found:

https://github.com/evcc-io/evcc/blob/c59f1b1ca76cd5031eb866a91849f76c94ebaaf6/templates/definition/meter/sofarsolar-g3.yaml#L136

This is referencing the first battery only. The inverter can support up to 8 batteries at the successive registers (0x0607 is the second). 

[image]

But I think referencing all individual batteries would be wrong. Instead we should use register 0x0667, Power_Bat_Total

[image]

The same issue here:

https://github.com/evcc-io/evcc/blob/c59f1b1ca76cd5031eb866a91849f76c94ebaaf6/templates/definition/meter/sofarsolar-g3.yaml#L152

This should be replaced by register 0x0668:

[image]

Sorry, I am in bed with fever today. Would test that and create a PR otherwise. Or you can wait a few days?

ep4711, 2025-02-23:
> What I found:
> 
> [evcc/templates/definition/meter/sofarsolar-g3.yaml](https://github.com/evcc-io/evcc/blob/c59f1b1ca76cd5031eb866a91849f76c94ebaaf6/templates/definition/meter/sofarsolar-g3.yaml#L136)
> 
> Line 136 in [c59f1b1](/evcc-io/evcc/commit/c59f1b1ca76cd5031eb866a91849f76c94ebaaf6)
> 
>        address: 0x0606 # Power_Bat1 
> This is referencing the first battery only. The inverter can support up to 8 batteries at the successive registers (0x0607 is the second).
> 
> <img alt="Image" width="912" src="https://private-user-images.githubusercontent.com/894150/415968671-be0c49c8-6c52-49aa-b2d0-4583bd864016.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDAzMDA4NTAsIm5iZiI6MTc0MDMwMDU1MCwicGF0aCI6Ii84OTQxNTAvNDE1OTY4NjcxLWJlMGM0OWM4LTZjNTItNDlhYS1iMmQwLTQ1ODNiZDg2NDAxNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMjIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDIyM1QwODQ5MTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02NGJkNDE3OWRkOTRmYTAyZjEwMGQ3Y2Q5YjE0YzA0YjBjNjc3MjE4ZjdjODM0ODAyYjg5MmE4NzUwYWJhOGU3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Dqd6cZkWcp6WaBBFOJ5YO4lI4VyrCibvIU_V7zvRZNc">
> But I think referencing all individual batteries is not what we want to do. Instead we should use register 0x0667, Power_Bat_Total
> 
> <img alt="Image" width="918" src="https://private-user-images.githubusercontent.com/894150/415968716-8c4473bf-0e47-4759-908f-cf75a7c7273c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDAzMDA4NTAsIm5iZiI6MTc0MDMwMDU1MCwicGF0aCI6Ii84OTQxNTAvNDE1OTY4NzE2LThjNDQ3M2JmLTBlNDctNDc1OS05MDhmLWNmNzVhN2M3MjczYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMjIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDIyM1QwODQ5MTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zYmNlNTA1Yjg3N2E2Zjc3MGVlYjQ4YmVjYTRhYzM0NmI5OGY2MTI5Y2MxZDc3NGY5MjI1NGFmOGE4YWYwOWRmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.inESUHg1k0f9JMvwogJvJ-W_QEGnBNPzgQ4pS9n-UI8">
> The same issue here:
> 
> [evcc/templates/definition/meter/sofarsolar-g3.yaml](https://github.com/evcc-io/evcc/blob/c59f1b1ca76cd5031eb866a91849f76c94ebaaf6/templates/definition/meter/sofarsolar-g3.yaml#L152)
> 
> Line 152 in [c59f1b1](/evcc-io/evcc/commit/c59f1b1ca76cd5031eb866a91849f76c94ebaaf6)
> 
>        address: 0x0608 # SOC_Bat1 
> This should be replaced by register 0x0668:
> 
> <img alt="Image" width="914" src="https://private-user-images.githubusercontent.com/894150/415968858-2b1e3e65-e42e-46c1-a39b-79215dc8e0a4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDAzMDA4NTAsIm5iZiI6MTc0MDMwMDU1MCwicGF0aCI6Ii84OTQxNTAvNDE1OTY4ODU4LTJiMWUzZTY1LWU0MmUtNDZjMS1hMzliLTc5MjE1ZGM4ZTBhNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMjIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDIyM1QwODQ5MTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00ZmY5MGYwYjdmYzI1MzAxYzA5NDk0NjQwNWVkZjlkMDMxNzQ5MDg4Mjc3ZmZhM2M0NjA5ZjI1MWNkZDEyNWYwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.6sZmK40o6iWRVZRew5uL8SlOS2h2BL62h45ly2SOkUM">
> Sorry, I am in bed with fever today. Would test that and create a PR otherwise. Or you can wait a few days?

no rush from my side, wish a fast recovery!

andig, 2025-02-23:
@ep4711 just make a PR with updating those 2 registers in the template- nothing simpler than that!

ep4711, 2025-02-23:
> [@ep4711](https://github.com/ep4711) just make a PR with updating those 2 registers in the template- nothing simpler than that!

I will take care about the PR. I never created one before but will do my best ;-)

ep4711, 2025-02-23:
> > [@ep4711](https://github.com/ep4711) just make a PR with updating those 2 registers in the template- nothing simpler than that!
> 
> I will take care about the PR. I never created one before but will do my best ;-)

I got an error message, I guess I need the collaborator role. I have to read the instructions in more depth.:

Pull request creation failed. Validation failed: must be a collaborator

cschlipf, 2025-02-23:
You need to create a fork of the repository. But nevermind, let me quickly do it. I just can't test it right now

cschlipf, 2025-02-23:
PR created. @ep4711  Maybe you can test it with the next nightly build after it got merged?

ep4711, 2025-02-23:
> PR created. [@ep4711](https://github.com/ep4711) Maybe you can test it with the next nightly build after it got merged?

sure I will, thank you for your support and get better soon!

closed 2025-02-23 by premultiply

cschlipf, 2025-02-24:
So with my single battery tower it seems to work after updating. @ep4711 What about multiple battery units?

ep4711, 2025-02-24:
> So with my single battery tower it seems to work after updating. [@ep4711](https://github.com/ep4711) What about multiple battery units?

I have nightly 0.200.2 running now. Since it is night now, I will continue to observe. Will feedback tomorrow, hopefully some sun...

ep4711, 2025-02-25:
@cschlipf : unfortunately not enough sun today that I can confirm the right behaviour. will try again tomorrow.

cschlipf, 2025-02-25:
We found another issue today. Please test the nightly tomorrow

ep4711, 2025-02-25:
> We found another issue today. Please test the nightly tomorrow

will do, thank you!

wardwygaerts, 2025-02-25:
I also noticed strange (low) values of the power coming from my battery in v0.200.3, reverted back to v0.200.2

cschlipf, 2025-02-25:
> I also noticed strange (low) values of the power coming from my battery in v0.200.3, reverted back to v0.200.2

That's the issue I mentioned.

cschlipf, 2025-02-26:
Gerade mit der 200.4 getestet. Es tut jetzt.

Wermutstropfen ist nun leider, dass die Batterie nur noch in 100W Schritten angezeigt wird und damit zieht auch eine größere Ungenauigkeit in den Hausverbrauch ein. Und zwischen 10W und 100W ist in den üblichen Bereichen im Haus leider schon ein deutlicher Unterschied.

@premultiply - fällt Dir langfristig eine bessere Lösung ein? Zum Beispiel eine Battery ID (Value 1-8) als optionaler Parameter (default: 1) und man kann dann mehrere Batterien mit dem Template mit verschiedenen IDs konfigurieren und man zählt die ID zum Register Wert 0x605 dazu? (Battery ID 1 wird dann 0x606)

andig, 2025-02-26:
Wie wäre >1 Bat als Option oder einfach separat mit storageid konfigurieren? Wäre dann breaking, aber so lange gibts das ja noch nicht.

premultiply, 2025-02-26:
Würde auch für die Variante analog zu anderen Implementierungen via `storageunit` plädieren.
Die passenden Summenbildungen macht evcc an der `site` automatisch.

cschlipf, 2025-02-26:
Hört sich auch super an. Mich wundert, dass das jetzt erst kommt, also scheinen die, die mehr als eine Batterie am Sofar betreiben ja eher die Ausnahme zu sein.

Gibt es irgendwo ein Beispiel das zeigt wie man einen Teil des Templates konditional gegen einen anderen Teil austauscht? Müsste ja Register und Scale betreffen.

cschlipf, 2025-02-26:
Ok, @premultiply war schneller. Hast Du ein Beispiel template an dem ich mich orientieren könnte?

andig, 2025-02-26:
https://github.com/search?q=repo%3Aevcc-io%2Fevcc%20storageunit&type=code

cschlipf, 2025-02-26:
Perfekt. Danke!

ep4711, 2025-02-27:
> Gerade mit der 200.4 getestet. Es tut jetzt.
>
> Wermutstropfen ist nun leider, dass die Batterie nur noch in 100W Schritten angezeigt wird und damit zieht auch eine größere Ungenauigkeit in den [...]

Ich kann bestätigen, dass es mit 200.4 funktioniert. 
Wenn das mit den Storage units eingebaut ist werde ich es auch probieren. Ich habe ja 2 units am Sofar Wechselrichter dran. Danke erstmal!

PR #19102 (2025-02-23): Sofar: support multiple battery storage units
Use total values instead of BAT1 values

Fixes #18933