evcc-io/evcc #18934: Speichernutzung zum Abfedern von Preisspitzen (Peak Shaving)

gimmel, 2025-02-18:
**Is your feature request related to a problem? Please describe.**
Ich nutze evcc zum Überschussladen zweier Elektroautos und als Modbus Proxy für meinen Huawei Wechselrichter mit 10 kWh Speicher.
Außerdem nutze ich Tibber als dynamischen Stromtarif. Bei hohen Preisspitzen lade ich vorher (über evcc) den Speicher, indem ich manuell für das Netzladen eine Preisgrenze angebe. Ich darf dann aber nicht vergessen, das Laden wieder abzuschalten, sonst kann es passieren, dass der Speicher auch ohne Spitzen vollgeladen wird.
_I use evcc for surplus charging of two electric cars and as a Modbus proxy for my Huawei inverter with 10 kWh battery.
I also use Tibber as a dynamic electricity tariff. In the event of high price peaks, I charge the battery beforehand (via evcc) by manually specifying a price limit for grid charging. However, I must not forget to switch off charging again, otherwise the storage system may be fully charged even without peaks._ 

**Describe the solution you'd like**
Ich wünsche mir ein automatisches Peak Shaving. Sobald Spitzen in der Preisvorschau erscheinen, plant evcc, wann der Speicher wie weit geladen werden muss, um die Peaks zu überbrücken. Dabei soll sowohl der Verlust des Speichers mit einkalkuliert werden als auch die zu erwartende Solar-Leistung, bei genügend Solarstrom muss man Spitzen natürlich nicht abfangen. Ich will also nicht mehr ständig im Blick haben müssen, wie ich am meisten CO₂ und Energiekosten spare, das soll evcc für mich übernehmen.
_I would like to see automatic peak shaving. As soon as peaks appear in the price preview, evcc plans when and how far the battery needs to be charged in order to shave the peaks. This should take into account both the power loss of the battery and the expected solar power; if there is sufficient solar power, there is of course no need to intercept peaks. So I no longer want to have to constantly keep an eye on how I can save the most CO₂ and energy costs, evcc should do that for me._

**Describe alternatives you've considered**
Ich habe bereits versucht, in Homeassistant eine Lösung zu programmieren. Leider lässt sich das Laden des Speichers mit evcc  nicht vollständig über/MQTT steuern. Ich steuere nun den Speicher direkt aus Homeassistant, habe aber Sorge, dass sich das mit der Steuerung aus evcc eventuell überlagert. Außerdem ist YAML für so ein komplexes System mit vielem Faktoren nicht besonders gut geeignet.
_I have already tried to program a solution in Homeassistant. Unfortunately, loading the home battery with evcc cannot be fully controlled via/MQTT. I now control the home battery directly from Homeassistant, but I'm worried that this might overlap with the control from evcc. In addition, YAML is not particularly suitable for such a complex system with many factors._

Ich weiß, das ist ein bisschen viel verlangt für eine Lösung, die eigentlich mal für das Überschussladen entwickelt wurde, aber nach und nach scheint evcc ja zu einer kompletten Haus-Energie-Steuerung zu wachsen, selbst Wärmepumpen sind neuerdings in der Geräteliste. Es erscheint mir sinnvoll, die Steuerung in Homeassistant zu lassen, aber die Intelligenz in evcc. So halte ich es auch in meinem Dashboard. (Das Sternchen hinter PV bedeutet, dass der Strom gerade unter der Preisgrenze ist. Und ja, zur Zeit ist bei dynamischen Strompreisen alles unter 35 ct schon günstig.)

[image]
[image]
[image]

andig, 2025-02-18:
> Ich wünsche mir ein automatisches Peak Shaving. Sobald Spitzen in der Preisvorschau erscheinen, plant evcc, wann der Speicher wie weit geladen werden muss, um die Peaks zu überbrücken. Dabei soll [...]

Was soll das denn konkret bedeuten? Fahrzeugladung im PV Modus beenden um den Speicher laden zu können? Also eine Art dynamische Batteriepriorisierung?

Das gleiche Problem hast Du ja beim Hausverbrauch auch. Geht es hier wirklich um Peakshaving oder nicht eher darum, den Batterievorrang besser als mit fixer Grenze zu steuern?

Wir brauchen einen implementierbaren Algorithmus- bisher sehe ich noch keinen.

/cc @premultiply

tugsi, 2025-02-23:
Victron hat sowas als Dynamic ESS in seinem System mit eingebaut und kann aktiviert werden.
Der kalkuliert die Akkuabnutzung anhand der Kosten und Zyklenzahl, die PV-Prognose, den dynamischen Strompreis, bisheriger Hausverbrauch nach Stunden... 
Da lädt er dann den Akku im niedrigen Preissegment vor (sofern die Differenz zur Spitze groß genug ist).
Oder wenn der SOC für den prognostizierten Tag ausreicht und der Preis nachts günstiger ist, dann sperrt das System den Akku und man bezieht Grid.
An sich funktioniert dies schon recht gut, Ausnahmen gibt es immer, denn die Verbrauchsprognose kann ja nur über die Vergangenheit gehen und woher weiß er, dass die Frau dann doch mehr baden oder backen möchte ... 
Ich weiß nicht, ob Victron sich da in die Karten schauen lässt, zum nachbauten oder sofern Du zufällig ein Victron-System hast, schau Dir Dynamic ESS mal an. 
[https://www.victronenergy.com/live/drafts:dynamic_ess](url)

batmanvane, 2025-02-23:
Das DESS von Victron in Verbindung mit EVCC funktioniert für uns sehr gut. Günstiges Netzladen in EVCC nutzen wir damit aktuell gar nicht mehr.
Wenn ich es richtig beachte, dann nutzt DESS auch den minsoc als Steuergröße - kann man bei aktiviertem DESS in Backend nachschauen, der Wert ändert sich hier regelmäßig (wenn denn Preisdifferenzen entsprechend greifen). Im Anhang mal die Energieflüsse von heute (vergangene und forecast).[image]

tugsi, 2025-02-23:
Für mich funktioniert es auch sehr gut, aber nicht jeder hat ein Victron-System ;-) 
Daher finde ich den Ansatz von evcc sehr gut, dass so viele wie mögliche Systeme dort angebunden werden können und evcc zentral die Steuerung übernehmen kann. 
In meinem Fall nutze ich das dynamische EV-Laden, der Hausakku wird vom DESS Übernomen, somit ergänzt sich alles sehr gut.

andig, 2025-02-23:
Wir experimentieren gerade in Richtung https://github.com/evcc-io/evcc/issues/18934#issuecomment-2676844046. Bitte ein bisschen Geduld!

closed 2025-03-07 by github-actions[bot]