evcc-io/evcc #18942: Charging amount stopps at 23,9kWh

ep4711, 2025-02-18:
### Describe the bug

In the GUI the charging amount stopps counting with 23,9 kWh, even if it further charges. The Load Percentage of the car is still increading… has anybody the same issue?

### Steps to reproduce

1.Charge mor than 23,9kWh on the same day
2.
3.
...

### Configuration details

```yaml
nothing extraordinary, sofar Inverter with Sofar Batteries, e-Go Wallbox, Mercedes EQA
```

### Log details

```text
nothing visible in logs, seems to be a GUI problem
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.1

andig, 2025-02-18:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2025-02-18 by andig

ep4711, 2025-02-19:
2025/02/18 14:10:56	stdout	[site  ] DEBUG 2025/02/18 14:10:56 grid currents: [-3.09 -3.77 -3.16]A
2025/02/18 14:10:56	stdout	[site  ] DEBUG 2025/02/18 14:10:56 grid powers: [-30 -20 -10]W
2025/02/18 14:10:55	stdout	[site  ] DEBUG 2025/02/18 14:10:55 battery 1 soc: 100%
2025/02/18 14:10:54	stdout	[site  ] DEBUG 2025/02/18 14:10:54 battery 1 power: 10W
2025/02/18 14:10:54	stdout	[site  ] DEBUG 2025/02/18 14:10:54 pv 1 power: 10300W
2025/02/18 14:10:54	stdout	[site  ] DEBUG 2025/02/18 14:10:54 grid power: -140W
2025/02/18 14:10:53	stdout	[lp-1  ] DEBUG 2025/02/18 14:10:53 charge currents: [8 8 8]A
2025/02/18 14:10:53	stdout	[lp-1  ] DEBUG 2025/02/18 14:10:53 charge power: 5710W
2025/02/18 14:10:53	stdout	[site  ] DEBUG 2025/02/18 14:10:53 ----
2025/02/18 14:10:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:10:27 pv charge current: 8.59A = 8A + 0.594A (-410W @ 3p)
2025/02/18 14:10:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:10:27 !! pvScalePhases EXIT
2025/02/18 14:10:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:10:27 !! pvScalePhases UP maxPhases: 3, available power: 6100W, scalable: false
2025/02/18 14:10:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:10:27 !! pvScalePhases DOWN activePhases: 3, available power: 6100W, scalable: false
2025/02/18 14:10:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:10:27 vehicle range: 264km
2025/02/18 14:10:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:10:27 vehicle soc limit: 100%
2025/02/18 14:10:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:10:27 vehicle soc: 77%
2025/02/18 14:10:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:10:27 soc estimated: 77.32% (vehicle: 77.00%)
2025/02/18 14:10:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:10:27 charger status: C
2025/02/18 14:10:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:10:27 charge total import: 1376.038kWh
2025/02/18 14:10:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:10:27 detected active phases: 3p
2025/02/18 14:10:27	stdout	[site  ] DEBUG 2025/02/18 14:10:27 site power: -410W
2025/02/18 14:10:26	stdout	[site  ] DEBUG 2025/02/18 14:10:26 grid currents: [-3.21 -3.88 -6.42]A
2025/02/18 14:10:26	stdout	[site  ] DEBUG 2025/02/18 14:10:26 battery 1 soc: 100%
2025/02/18 14:10:25	stdout	[site  ] DEBUG 2025/02/18 14:10:25 grid powers: [-170 -170 -170]W
2025/02/18 14:10:24	stdout	[site  ] DEBUG 2025/02/18 14:10:24 pv 1 power: 10800W
2025/02/18 14:10:24	stdout	[site  ] DEBUG 2025/02/18 14:10:24 battery 1 power: -0W
2025/02/18 14:10:24	stdout	[site  ] DEBUG 2025/02/18 14:10:24 grid power: -510W
2025/02/18 14:10:23	stdout	[lp-1  ] DEBUG 2025/02/18 14:10:23 charge currents: [8 8 7.9]A
2025/02/18 14:10:23	stdout	[lp-1  ] DEBUG 2025/02/18 14:10:23 charge power: 5690W
2025/02/18 14:10:23	stdout	[site  ] DEBUG 2025/02/18 14:10:23 ----
2025/02/18 14:09:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:57 pv charge current: 8.68A = 8A + 0.681A (-470W @ 3p)
2025/02/18 14:09:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:57 !! pvScalePhases EXIT
2025/02/18 14:09:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:57 !! pvScalePhases UP maxPhases: 3, available power: 6190W, scalable: false
2025/02/18 14:09:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:57 !! pvScalePhases DOWN activePhases: 3, available power: 6190W, scalable: false
2025/02/18 14:09:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:57 vehicle range: 264km
2025/02/18 14:09:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:57 vehicle soc limit: 100%
2025/02/18 14:09:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:57 vehicle soc: 77%
2025/02/18 14:09:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:57 soc estimated: 77.32% (vehicle: 77.00%)
2025/02/18 14:09:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:57 charger status: C
2025/02/18 14:09:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:57 charge total import: 1375.991kWh
2025/02/18 14:09:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:57 detected active phases: 3p
2025/02/18 14:09:57	stdout	[site  ] DEBUG 2025/02/18 14:09:57 site power: -470W
2025/02/18 14:09:56	stdout	[site  ] DEBUG 2025/02/18 14:09:56 grid currents: [-3.22 -3.89 -6.22]A
2025/02/18 14:09:56	stdout	[site  ] DEBUG 2025/02/18 14:09:56 battery 1 soc: 100%
2025/02/18 14:09:55	stdout	[site  ] DEBUG 2025/02/18 14:09:55 grid powers: [-190 -200 -210]W
2025/02/18 14:09:54	stdout	[site  ] DEBUG 2025/02/18 14:09:54 battery 1 power: -0W
2025/02/18 14:09:54	stdout	[site  ] DEBUG 2025/02/18 14:09:54 pv 1 power: 10900W
2025/02/18 14:09:54	stdout	[site  ] DEBUG 2025/02/18 14:09:54 grid power: -570W
2025/02/18 14:09:53	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:53 charge currents: [8 8.1 8]A
2025/02/18 14:09:53	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:53 charge power: 5720W
2025/02/18 14:09:53	stdout	[site  ] DEBUG 2025/02/18 14:09:53 ----
2025/02/18 14:09:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:27 pv charge current: 8.83A = 8A + 0.826A (-570W @ 3p)
2025/02/18 14:09:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:27 !! pvScalePhases EXIT
2025/02/18 14:09:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:27 !! pvScalePhases UP maxPhases: 3, available power: 6270W, scalable: false
2025/02/18 14:09:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:27 !! pvScalePhases DOWN activePhases: 3, available power: 6270W, scalable: false
2025/02/18 14:09:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:27 vehicle range: 264km
2025/02/18 14:09:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:27 vehicle soc limit: 100%
2025/02/18 14:09:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:27 vehicle soc: 77%
2025/02/18 14:09:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:27 soc estimated: 77.32% (vehicle: 77.00%)
[... cut ...]
2025/02/18 14:08:55	stdout	[site  ] DEBUG 2025/02/18 14:08:55 grid powers: [-200 -220 -220]W
2025/02/18 14:08:54	stdout	[site  ] DEBUG 2025/02/18 14:08:54 pv 1 power: 11000W
2025/02/18 14:08:54	stdout	[site  ] DEBUG 2025/02/18 14:08:54 battery 1 power: 10W
2025/02/18 14:08:54	stdout	[site  ] DEBUG 2025/02/18 14:08:54 grid power: -530W
2025/02/18 14:08:53	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:53 charge currents: [7.9 8 8]A
2025/02/18 14:08:53	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:53 charge power: 5700W
2025/02/18 14:08:53	stdout	[site  ] DEBUG 2025/02/18 14:08:53 ----

ep4711, 2025-02-19:
I analyzed the logfile. Around 14.08 the charger status changes from B to C and three logs with "!!" appear:

2025/02/18 14:08:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:27 !! pvScalePhases EXIT
2025/02/18 14:08:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:27 !! pvScalePhases UP maxPhases: 3, available power: 6110W, scalable: false
2025/02/18 14:08:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:27 !! pvScalePhases DOWN activePhases: 3, available power: 6110W, scalable: false

This is the time when evcc stopps counting higher than 23,9.

here is the rest of the log around this time:

2025/02/18 14:09:53	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:53 charge currents: [8 8.1 8]A
2025/02/18 14:09:53	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:53 charge power: 5720W
2025/02/18 14:09:53	stdout	[site  ] DEBUG 2025/02/18 14:09:53 ----
2025/02/18 14:09:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:27 pv charge current: 8.83A = 8A + 0.826A (-570W @ 3p)
2025/02/18 14:09:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:27 !! pvScalePhases EXIT
2025/02/18 14:09:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:27 !! pvScalePhases UP maxPhases: 3, available power: 6270W, scalable: false
2025/02/18 14:09:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:27 !! pvScalePhases DOWN activePhases: 3, available power: 6270W, scalable: false
2025/02/18 14:09:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:27 vehicle range: 264km
2025/02/18 14:09:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:27 vehicle soc limit: 100%
2025/02/18 14:09:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:27 vehicle soc: 77%
2025/02/18 14:09:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:27 soc estimated: 77.32% (vehicle: 77.00%)
2025/02/18 14:09:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:27 charger status: C
2025/02/18 14:09:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:27 charge total import: 1375.945kWh
2025/02/18 14:09:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:27 detected active phases: 3p
2025/02/18 14:09:27	stdout	[site  ] DEBUG 2025/02/18 14:09:27 site power: -570W
2025/02/18 14:09:26	stdout	[site  ] DEBUG 2025/02/18 14:09:26 grid currents: [-3.31 -3.92 -3.44]A
2025/02/18 14:09:26	stdout	[site  ] DEBUG 2025/02/18 14:09:26 battery 1 soc: 100%
2025/02/18 14:09:25	stdout	[site  ] DEBUG 2025/02/18 14:09:25 grid powers: [-250 -250 -270]W
2025/02/18 14:09:24	stdout	[site  ] DEBUG 2025/02/18 14:09:24 pv 1 power: 10900W
2025/02/18 14:09:24	stdout	[site  ] DEBUG 2025/02/18 14:09:24 battery 1 power: -0W
2025/02/18 14:09:24	stdout	[site  ] DEBUG 2025/02/18 14:09:24 grid power: -670W
2025/02/18 14:09:23	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:23 charge currents: [8 7.9 8]A
2025/02/18 14:09:23	stdout	[lp-1  ] DEBUG 2025/02/18 14:09:23 charge power: 5700W
2025/02/18 14:09:23	stdout	[site  ] DEBUG 2025/02/18 14:09:23 ----
2025/02/18 14:08:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:57 pv charge current: 8.61A = 8A + 0.609A (-420W @ 3p)
2025/02/18 14:08:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:57 !! pvScalePhases EXIT
2025/02/18 14:08:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:57 !! pvScalePhases UP maxPhases: 3, available power: 6120W, scalable: false
2025/02/18 14:08:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:57 !! pvScalePhases DOWN activePhases: 3, available power: 6120W, scalable: false
2025/02/18 14:08:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:57 vehicle range: 264km
2025/02/18 14:08:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:57 vehicle soc limit: 100%
2025/02/18 14:08:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:57 vehicle soc: 77%
2025/02/18 14:08:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:57 soc estimated: 77.32% (vehicle: 77.00%)
2025/02/18 14:08:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:57 charger status: C
2025/02/18 14:08:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:57 charge total import: 1375.897kWh
2025/02/18 14:08:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:57 detected active phases: 3p
2025/02/18 14:08:57	stdout	[site  ] DEBUG 2025/02/18 14:08:57 site power: -420W
2025/02/18 14:08:56	stdout	[site  ] DEBUG 2025/02/18 14:08:56 grid currents: [-3.3 -3.92 -6.24]A
2025/02/18 14:08:56	stdout	[site  ] DEBUG 2025/02/18 14:08:56 battery 1 soc: 100%
2025/02/18 14:08:55	stdout	[site  ] DEBUG 2025/02/18 14:08:55 grid powers: [-200 -220 -220]W
2025/02/18 14:08:54	stdout	[site  ] DEBUG 2025/02/18 14:08:54 pv 1 power: 11000W
2025/02/18 14:08:54	stdout	[site  ] DEBUG 2025/02/18 14:08:54 battery 1 power: 10W
2025/02/18 14:08:54	stdout	[site  ] DEBUG 2025/02/18 14:08:54 grid power: -530W
2025/02/18 14:08:53	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:53 charge currents: [7.9 8 8]A
2025/02/18 14:08:53	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:53 charge power: 5700W
2025/02/18 14:08:53	stdout	[site  ] DEBUG 2025/02/18 14:08:53 ----
2025/02/18 14:08:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:27 max charge current: 8A
2025/02/18 14:08:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:27 pv timer inactive
2025/02/18 14:08:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:27 pv timer reset
2025/02/18 14:08:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:27 pv charge current: 8.8A = 6A + 2.8A (-1930W @ 3p)
2025/02/18 14:08:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:27 !! pvScalePhases EXIT
2025/02/18 14:08:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:27 !! pvScalePhases UP maxPhases: 3, available power: 6110W, scalable: false
2025/02/18 14:08:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:27 !! pvScalePhases DOWN activePhases: 3, available power: 6110W, scalable: false
2025/02/18 14:08:27	stdout	[lp-1  ] DEBUG 2025/02/18 14:08:27 vehicle range: 264km
[... cut ...]
2025/02/18 14:07:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:07:57 max charge current: 6A
2025/02/18 14:07:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:07:57 pv enable timer elapsed
2025/02/18 14:07:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:07:57 pv enable in 0s
2025/02/18 14:07:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:07:57 site power -6040W <= 0W enable threshold
2025/02/18 14:07:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:07:57 pv charge current: 8.75A = 0A + 8.75A (-6040W @ 3p)
2025/02/18 14:07:57	stdout	[lp-1  ] DEBUG 2025/02/18 14:07:57 charger status: B

ep4711, 2025-02-19:
is this sufficient to re-open the issue?

ep4711, 2025-02-19:
> Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

I pasted the log now, sorry for missing it the first time.

andig, 2025-02-19:
Logs in reverse order are hard to read. Afaikt charge total import continues to increase. It would help to see /api/state when this happens and again maybe 10 minutes later while charging continues.

ep4711, 2025-02-19:
> Logs in reverse order are hard to read. Afaikt charge total import continues to increase. It would help to see /api/state when this happens and again maybe 10 minutes later while charging continues.

This is the Log order that Docker delivers for me, sorry. Even if it continues to increase in the log, the GUI stays on 23,9 kWh. It is still connected and even next day when it starts charging again, the value in the GUI ist still stable. I can try to bring more info when it happens next time, can you describe the commands I have to submit in the terminal of the container (I run on Docker)...

andig, 2025-02-21:
> This is the Log order that Docker delivers for me

You can use

    docker logs

> It is still connected and even next day when it starts charging again, the value in the GUI ist still stable.

When this happens again, please send the log and the output of /api/state in the browser at the same time.

ep4711, 2025-03-02:
> You can use
>
> ```
> docker logs
> ```
>
>
> When this happens again, please send the log and the output of /api/state in the browser at the same time.

We can close this issue, it turned out, that the wallbox has a problem and stopps working from a certain temerature on.It goes on error. When it continues later evcc ist not updating the charging amout anymore. I will obseve, no need for action in the moment.