evcc-io/evcc #18950: Use home battery should be global not per charger

Cheppetto, 2025-02-18:
### Describe the bug

Currently you can set use gone battery yes/no for every charger/load point.

This makes absolutely no sense, cause if deactivated for a single one battery will globally be forbidden - Strom hat halt keine Masche

### Steps to reproduce

1. Try it
2. Try it again
3. 
...

### Configuration details

```yaml
Migrated
```

### Log details

```text
...
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.1

andig, 2025-02-19:
> Currently you can set use gone battery yes/no for every charger/load point.

You can still decide for that specific loadpoint to conteibute to that decision or not.

closed 2025-02-19 by andig

Cheppetto, 2025-02-19:
> > Currently you can set use gone battery yes/no for every charger/load point.
> 
> You can still decide for that specific loadpoint to conteibute to that decision or not.

This makes sense for vehicle connected loadpoints, I get that, there it might have minimal benefits 

But it also effects integrated devices like heaters ...