evcc-io/evcc #18951: Integrated devices

Cheppetto, 2025-02-18:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when adding an running loadpoints which are not actual load points and those need to be charging (PV, min + PV, fast)

As well as (keep current status) leads to disabling the switch when adding the device and restarting. 

Device should have the capability to be in show consumption only mode. 

Furthermore it should be possible that they ate not shown in UI except in the consumption bar.

VolkerK62, 2025-02-18:
something like this? https://github.com/evcc-io/evcc/discussions/7235

Cheppetto, 2025-02-18:
Kind of, but no not like that.
should be a own category - consumer 
Had nothing todo with a loadpoint

Might as well be hidden in user-UI. - only shown within the consumers bar on top and used for load control with per default higher prio then a charger, as it will normally be heating, network stuff, freezers, fridges, laundry stuff, stove ...

maybe A/C

andig, 2025-02-19:
You can already add extra meters to track consumption. They are for time being visible in the api alone.