evcc-io/evcc #18958: Var not charged

timpetersson81, 2025-02-19:
### Describe the bug

Car not charged, planner outdated. Plan was added and confirmed. I cannot reach work!

### Steps to reproduce

1. Wait for night time
2.
3.
...

### Configuration details

```yaml
Config, is in database now and was migrated there
```

### Log details

```text
[site ] DEBUG 2025/02/19 05:48:03 battery 1 soc: 66%
[site ] DEBUG 2025/02/19 05:48:03 grid currents: [3.24 2.33 2.33]A
[site ] DEBUG 2025/02/19 05:48:03 battery has priority at soc 66% (< 100%)
[site ] DEBUG 2025/02/19 05:48:03 site power: 1981W
[lp-1 ] DEBUG 2025/02/19 05:48:03 charge voltages: [231 250 228]V
[lp-1 ] DEBUG 2025/02/19 05:48:03 detected connected phases: 3p
[lp-1 ] DEBUG 2025/02/19 05:48:03 charge total import: 3486.347kWh
[lp-1 ] DEBUG 2025/02/19 05:48:03 charger status: B
[lp-1 ] ERROR 2025/02/19 05:48:03 planner: outdated
[lp-1 ] DEBUG 2025/02/19 05:48:03 pv charge current: 0A = 0A + -2.87A (1981W @ 3p)
[site ] DEBUG 2025/02/19 05:48:33 ----
[lp-1 ] DEBUG 2025/02/19 05:48:33 charge power: 0W
[lp-1 ] DEBUG 2025/02/19 05:48:33 charge currents: [0 0 0]A
[site ] WARN 2025/02/19 05:48:33 planner: outdated
[site ] DEBUG 2025/02/19 05:48:33 grid power: 1885W
[site ] DEBUG 2025/02/19 05:48:33 pv 1 power: 0W
[site ] DEBUG 2025/02/19 05:48:33 battery 1 power: -1556W
[site ] DEBUG 2025/02/19 05:48:33 grid currents: [3.28 2.33 2.32]A
[site ] DEBUG 2025/02/19 05:48:33 battery 1 soc: 66%
[site ] DEBUG 2025/02/19 05:48:33 battery has priority at soc 66% (< 100%)
[site ] DEBUG 2025/02/19 05:48:33 site power: 1985W
[lp-1 ] DEBUG 2025/02/19 05:48:33 charge voltages: [231 250 229]V
[lp-1 ] DEBUG 2025/02/19 05:48:33 detected connected phases: 3p
[lp-1 ] DEBUG 2025/02/19 05:48:33 charge total import: 3486.347kWh
[lp-1 ] DEBUG 2025/02/19 05:48:33 charger status: B
[lp-1 ] ERROR 2025/02/19 05:48:33 planner: outdated
[lp-1 ] DEBUG 2025/02/19 05:48:33 pv charge current: 0A = 0A + -2.88A (1985W @ 3p)
[site ] DEBUG 2025/02/19 05:49:03 ----
[lp-1 ] DEBUG 2025/02/19 05:49:03 charge power: 0W
[lp-1 ] DEBUG 2025/02/19 05:49:03 charge currents: [0 0 0]A
[site ] WARN 2025/02/19 05:49:03 planner: outdated
[site ] DEBUG 2025/02/19 05:49:03 grid power: 1885W
[site ] DEBUG 2025/02/19 05:49:03 pv 1 power: 0W
[site ] DEBUG 2025/02/19 05:49:03 battery 1 power: -1556W
[site ] DEBUG 2025/02/19 05:49:03 grid currents: [3.3 2.33 2.32]A
[site ] DEBUG 2025/02/19 05:49:03 battery 1 soc: 66%
[site ] DEBUG 2025/02/19 05:49:03 battery has priority at soc 66% (< 100%)
[site ] DEBUG 2025/02/19 05:49:03 site power: 1985W
[lp-1 ] DEBUG 2025/02/19 05:49:03 charge voltages: [230 249 229]V
[lp-1 ] DEBUG 2025/02/19 05:49:03 detected connected phases: 3p
[lp-1 ] DEBUG 2025/02/19 05:49:03 charge total import: 3486.347kWh
[lp-1 ] DEBUG 2025/02/19 05:49:03 charger status: B
[lp-1 ] ERROR 2025/02/19 05:49:03 planner: outdated
[lp-1 ] DEBUG 2025/02/19 05:49:03 pv charge current: 0A = 0A + -2.88A (1985W @ 3p)
[site ] DEBUG 2025/02/19 05:49:33 ----
[lp-1 ] DEBUG 2025/02/19 05:49:33 charge power: 0W
[lp-1 ] DEBUG 2025/02/19 05:49:33 charge currents: [0 0 0]A
[site ] WARN 2025/02/19 05:49:33 planner: outdated
[site ] DEBUG 2025/02/19 05:49:33 grid power: 1879W
[site ] DEBUG 2025/02/19 05:49:33 pv 1 power: 0W
[... 567 lines cut ...]
[site ] DEBUG 2025/02/19 06:06:33 grid currents: [4.97 2.33 2.33]A
[site ] DEBUG 2025/02/19 06:06:33 battery 1 soc: 71%
[site ] DEBUG 2025/02/19 06:06:33 battery has priority at soc 71% (< 100%)
[site ] DEBUG 2025/02/19 06:06:33 site power: 2367W
[lp-1 ] DEBUG 2025/02/19 06:06:33 charge voltages: [229 249 228]V
[lp-1 ] DEBUG 2025/02/19 06:06:33 detected connected phases: 3p
[lp-1 ] DEBUG 2025/02/19 06:06:33 charge total import: 3486.347kWh
[lp-1 ] DEBUG 2025/02/19 06:06:33 charger status: B
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.1

andig, 2025-02-19:
Wow, that sucks and clearly shouldn‘t happen. Can you please show the output of /api/state? Even if tariff doesn‘t have data the planner should still charge your car.

timpetersson81, 2025-02-19:
The energy Price api? That Mighty be what failed. It output code 500 at the moment, but as you say it should still charge. Allso it had the prices for the charging prior to the 500 response, but savning the prices before trying to get new seems to be the issue the the api fails.

> Wow, that sucks and clearly shouldn‘t happen. I don’t think this ever has happened before. Can you please show the output of /api/state? Even if tariff doesn‘t have data the planner should still charge your car.

timpetersson81, 2025-02-19:
`{"result":{"auth":{},"aux":[],"battery":[{"power":0,"energy":1104.7,"capacity":0,"soc":96,"controllable":true}],"batteryCapacity":1,"batteryDischargeControl":false,"batteryEnergy":1104.7,"batteryGridChargeActive":false,"batteryMode":"unknown","batteryPower":0,"batterySoc":96,"bufferSoc":0,"bufferStartSoc":0,"currency":"DKK","eebus":false,"ext":[],"fatal":null,"forecast":{},"greenShareHome":1,"greenShareLoadpoints":0.876,"grid":{"power":349,"energy":6347.8,"currents":[4.7625,1.1240149315636665,-4.394693200663349]},"gridConfigured":true,"hems":{},"homePower":785.1,"influx":{"url":"","database":"","org":"","user":"","insecure":false},"interval":30,"loadpoints":[{"batteryBoost":false,"chargeCurrent":6,"chargeCurrents":[5.9,5.7,0],"chargeDuration":7590,"chargePower":2810.9,"chargeRemainingDuration":5783,"chargeRemainingEnergy":4115.35,"chargeTotalImport":3494.412,"chargeVoltages":[232,253,238],"chargedEnergy":5908.4,"chargerFeatureHeating":false,"chargerFeatureIntegratedDevice":false,"chargerIcon":null,"chargerPhases1p3p":false,"chargerSinglePhase":false,"chargerStatusReason":"unknown","charging":true,"connected":true,"connectedDuration":0,"disableDelay":180,"disableThreshold":0,"effectiveLimitSoc":100,"effectiveMaxCurrent":16,"effectiveMinCurrent":6,"effectivePlanId":2,"effectivePlanSoc":100,"effectivePlanTime":"2025-02-20T07:00:00+01:00","effectivePriority":0,"enableDelay":60,"enableThreshold":0,"enabled":true,"limitEnergy":0,"limitSoc":0,"maxCurrent":16,"minCurrent":6,"mode":"minpv","phaseAction":"inactive","phaseRemaining":0,"phasesActive":2,"phasesConfigured":3,"planActive":false,"planEnergy":0,"planOverrun":0,"planProjectedEnd":null,"planProjectedStart":null,"planTime":null,"priority":0,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":null,"sessionEnergy":5908.4,"sessionPrice":null,"sessionPricePerKWh":null,"sessionSolarPercentage":87.381,"smartCostActive":false,"smartCostLimit":null,"smartCostNextStart":null,"title":"Carport","vehicleClimaterActive":null,"vehicleDetectionActive":false,"vehicleIdentity":"NoAuthorization","vehicleLimitSoc":0,"vehicleName":"db:4","vehicleOdometer":53646,"vehicleRange":133,"vehicleSoc":90.968,"vehicleWelcomeActive":false}],"messaging":true,"modbusproxy":null,"mqtt":{"broker":"","topic":"evcc"},"network":{"schema":"http","host":"evcc.local","port":7070},"prioritySoc":100,"pv":[{"power":3247,"energy":1372}],"pvEnergy":1372,"pvPower":3247,"residualPower":0,"siteTitle":"Hjemme","smartCostType":"priceforecast","sponsor":{"name":"timpetersson81","expiresAt":"2028-01-01T18:00:00Z"},"statistics":{"30d":{"avgCo2":0,"avgPrice":1.1172806563100137,"chargedKWh":745.8797,"solarPercentage":1.478684400822212},"365d":{"avgCo2":0,"avgPrice":0.846556714872531,"chargedKWh":2879.696735535435,"solarPercentage":0.806593114426403},"thisYear":{"avgCo2":0,"avgPrice":1.025933143443944,"chargedKWh":1226.549899999999,"solarPercentage":1.158161273577017},"total":{"avgCo2":0,"avgPrice":0.846556714872531,"chargedKWh":2879.696735535435,"solarPercentage":0.806593114426403}},"tariffGrid":2.6,"tariffPriceHome":0,"tariffPriceLoadpoints":2.6,"tariffSolar":221.8,"vehicles":{"Xpeng":{"title":"Xpeng","icon":"car","capacity":100,"features":["Offline"],"repeatingPlans":[]},"db:3":{"title":"Nannas golf","icon":"car","capacity":38,"maxCurrent":13,"features":["Offline"],"repeatingPlans":[]},"db:4":{"title":"Nanna's golf","icon":"car","capacity":38,"repeatingPlans":[{"weekdays":[1,2,3,4,5,6,0],"time":"07:00","tz":"Europe/Copenhagen","soc":100,"active":true}]}},"version":"0.200.1"}} `

benesolar, 2025-02-19:
That may be another Problem, but you should check your Phase 2 Voltage:
chargeVoltages":[232,**253**,238]

Allowd Maximum is 254 V.

benesolar, 2025-02-19:
in that case it would be best to contact an electrician or as a first step someone who can use a multimeter.

timpetersson81, 2025-02-20:
> in that case it would be best to contact an electrician or as a first step someone who can use a multimeter.

The inverter says that fases go no higher that 241.  Fases should be the same allover the house? I do Think that the inverter gets its data from some clamps the the incoming wires to the house.

Ill try to contact the electriciant, thanks for notising

closed 2025-02-21 by andig