evcc-io/evcc #18961: Tibber Grid not updated - 429 Too Many Requests

chrstmnn, 2025-02-19:
### Describe the bug

Still with version 0.200.0, Tibber API gets blocked when using EVCC.
As soon as I shut down EVCC and refresh my external IP, the API is working again as expected (e.g. in Home Assistant).
After turning of EVCC on again, it takes some hours or days until the API is blocked again.

### Steps to reproduce

1. Set up Tibber in EVCC
2. Let it run

### Configuration details

```yaml
meters:
- type: template
  template: tibber-pulse 
  usage: grid  
  token: <token>
  homeid: <homeid>
  name: grid1
```

### Log details

```text
[main ] INFO 2025/02/19 09:32:04 evcc 0.200.0
[main ] INFO 2025/02/19 09:32:04 using config file: /etc/evcc.yaml
[db ] INFO 2025/02/19 09:32:04 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2025/02/19 09:32:04 listening at :7070
[main ] FATAL 2025/02/19 09:32:04 meter [grid1] cannot create meter 'grid1': cannot create meter type 'template': cannot create meter type 'tibber-pulse': Message: 429 Too Many Requests, Locations: [], Extensions: map[code:request_error], Path: []
[main ] FATAL 2025/02/19 09:32:04 will attempt restart in: 15m0s
[main ] INFO 2025/02/19 09:32:04 new version available: 0.200.1
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.0

andig, 2025-02-19:
> I have verified that the issue is reproducible with the latest nightly build

Anscheinend nicht :( Es braucht ein trace log für Tibber. Bitte auf aktueller Version.

closed 2025-02-19 by andig

whisky2k2, 2025-03-18:
Ich habe seit wenigen Wochen auch das Problem und bin nicht sicher, ob es mit meinem Sprung von Version 0.130 auf 0.200.9 zusammenhängt. Hat sich irgendwas geändert, was ich in der Konfiguration anpassen muss?
Abhilfe schaffte es, den Token bei Tibber zu erneuern. Das ging 10 Tage gut. Nun habe ich das Problem schon nach zwei Tagen.

Konfiguration:
network:
  schema: http
  host: *****
  port: 7070
log: error
levels:
  cache: error
plant: *****
interval: 30s # control cycle interval
sponsortoken: *****
telemetry: false
meters:
- type: template
  template: huawei-dongle 
  id: 1  
  host: *****
  port: 502  
  usage: pv  
  modbus: tcpip  
  timeout: 15s  
  name: pv1
- type: template
  template: tibber-pulse 
  usage: grid  
  token: *****
  name: grid1
chargers:
- type: template
  template: go-e-v3 
  host: *****
  name: wallbox3
vehicles:
- type: template
  template: tesla 
  title: MY  
  icon: car  
  clientId: *****
  accessToken: *****
  refreshToken: *****
  vin: *****
  capacity: 77  
  phases: 3  
  commandProxy: https://tesla.evcc.io/  
  mode: pv  
  minCurrent: 6  
  maxCurrent: 16  
  name: ev2
loadpoints:
- title: Garage
  charger: wallbox3
  vehicle: ev2
  mode: pv
site:
  title: My home
  meters:
    grid: grid1
    pv:
    - pv1
tariffs:
  grid:
    type: tibber
    token: *****
  feedin:
    type: fixed
    price: 0.082 # EUR/kWh

Log:
[site ] DEBUG 2025/03/18 09:02:07 ----
[lp-1 ] DEBUG 2025/03/18 09:02:07 charge power: 0W
[lp-1 ] DEBUG 2025/03/18 09:02:07 charge currents: [0 0 0]A
[site ] DEBUG 2025/03/18 09:02:08 pv 1 power: 5491W
[site ] ERROR 2025/03/18 09:02:08 grid power: outdated

Mar 15 18:18:03 evcc evcc[3726]: [pulse ] ERROR 2025/03/15 18:18:03 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent go-graphql-client/0.9.0"

Ich finde im Log leider nicht mehr den Ursprung. 
Ich will jetzt zeitnah versuchen über Home Assistent die lokalen Werte an evcc weiterzuleiten, anstatt über die Tibber API.

chrstmnn, 2025-03-18:
Hi, auch wenn es hier anders gesehen wird, bin ich ziemlich sicher, dass das Problem bei EVCC liegt.
Seitdem ich die Daten per Tibber Home Assistant Integration abfrage und per MQTT an EVCC weiterleite, wurde mein API Token nicht ein einziges Mal gesperrt und funktioniert bereits seit etlichen Tagen problemlos.

whisky2k2, 2025-03-19:
> Anscheinend nicht :( Es braucht ein trace log für Tibber. Bitte auf aktueller Version.
Kannst Du mir helfen einzugrenzen, was Du benötigst?

[pulse ] TRACE 2025/03/18 13:18:38 {"id":"915b2dc6-ae2e-4ba5-bfce-dd68863e8d43","type":"next","payload":{"data":{"liveMeasurement":{"power":0,"powerProduction":2054,"lastMeterConsumption":5371.125,"lastMeterProduction":6788.3731,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/03/18 13:18:41 {"id":"915b2dc6-ae2e-4ba5-bfce-dd68863e8d43","type":"next","payload":{"data":{"liveMeasurement":{"power":0,"powerProduction":2052,"lastMeterConsumption":5371.125,"lastMeterProduction":6788.3743,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/03/18 13:18:44 {"id":"915b2dc6-ae2e-4ba5-bfce-dd68863e8d43","type":"next","payload":{"data":{"liveMeasurement":{"power":0,"powerProduction":1980,"lastMeterConsumption":5371.125,"lastMeterProduction":6788.3765,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/03/18 13:18:54 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/18 13:18:59 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/18 13:18:59 {"type":"connection_ack"} server
[pulse ] TRACE 2025/03/18 13:18:59 {"id":"cb2e1052-36a8-4e0e-833c-609d9628a8bd","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,lastMeterProduction,currentL1,currentL2,currentL3}}","variables":{"homeId":"5c5eea54-c342-4bb3-80a7-fe6adab180da"}}} client
[pulse ] TRACE 2025/03/18 13:18:59 {"id":"cb2e1052-36a8-4e0e-833c-609d9628a8bd","type":"next","payload":{"errors":[{"message":"unable to start stream api:a1d68ecb64cc4c8caa75bc4ab6fbf24c for device a1d68ecb-64cc-4c8c-aa75-bc4ab6fbf24c; userId e6d2fab0-ee12-4ee0-bee0-c250f0ec4768","locations":[{"line":1,"column":28}],"path":["liveMeasurement"]}]}} server
[pulse ] ERROR 2025/03/18 13:18:59 Message: unable to start stream api:a1d68ecb64cc4c8caa75bc4ab6fbf24c for device a1d68ecb-64cc-4c8c-aa75-bc4ab6fbf24c; userId e6d2fab0-ee12-4ee0-bee0-c250f0ec4768, Locations: [{Line:1 Column:28}], Extensions: map[], Path: [liveMeasurement]
[pulse ] TRACE 2025/03/18 13:18:59 failed to read JSON message: failed to read: use of closed network connection client
[pulse ] TRACE 2025/03/18 13:19:05 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/18 13:19:05 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/18 13:19:10 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/18 13:19:10 failed to read JSON message: failed to get reader: use of closed network connection client

nc-kristema, 2025-04-28:
Ich habe das gleiche Problem.

failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = \"Too many open connections on this server; count 2; user agent evcc/v0.0.0 hasura/go-graphql-client/v0.13.2\

Mit Tibber aktuell. 
Lief bis dato ok aktuellste Version 0.203.4

VG, Markus

whisky2k2, 2025-04-28:
Das issue wird hier weiter bearbeitet https://github.com/evcc-io/evcc/issues/20502#issuecomment-2831477665