evcc-io/evcc #18974: Wallbox Pulsar: charger logic error: current mismatch

mmargi, 2025-02-19:
### Describe the bug

Hi, when I try charge with OCPP connected Wallbox Pulsar Max in Solar or Min+Solar mode, there are continuos errors in log like : charger logic error: current mismatch (got 32A, expected 16.2A)..

Are there probles on EVCC or Wallbox side? In wallbox app i can see requested value for some time (seconds...) for example 16A, but 

I had isolated every possible other device, even disabled chargng station access to cloud. EVCC was the only thing that could communicate with the charger.

Maybe important: I cant set this in my app:  Enable “Improved charger control” (Profile -> Experimental functions) (myWallbox app)

### Steps to reproduce

1. Connect EVCC with Wallbox OCPP
2. Set Solar or Min+Solar mode and start charging
3.
...

### Configuration details

```yaml
chargers:
- type: template
  template: pulsarplus.
  stationid: sloupek..
  name: wallbox-sloupek
  remotestart: true

- type: template
  template: pulsarplus.
  stationid: garaz..
  name: wallbox-garaz
  remotestart: true

loadpoints:
- title: Sloupek
  charger: wallbox-sloupek
  mode: off

- title: Garaz
  charger: wallbox-garaz
  mode: off

site:
  title: Doma
  meters:
    grid: grid_estimated
    pv:
    - pv_estimated
    battery:
    - battery
```

### Log details

```text
lp-2 ] DEBUG 2025/02/19 10:44:48 charge voltages: [235 244 237]V
[lp-2 ] DEBUG 2025/02/19 10:44:48 detected connected phases: 3p
[lp-2 ] DEBUG 2025/02/19 10:44:48 detected active phases: 1p
[lp-2 ] DEBUG 2025/02/19 10:44:48 charge total import: 3117.800kWh
[lp-2 ] DEBUG 2025/02/19 10:44:48 charger status: C
[lp-2 ] DEBUG 2025/02/19 10:44:48 soc estimated: 93.91% (vehicle: 92.00%)
[lp-2 ] DEBUG 2025/02/19 10:44:48 vehicle soc: 94%
[lp-2 ] DEBUG 2025/02/19 10:44:48 vehicle range: 0km
[lp-2 ] WARN 2025/02/19 10:44:48 charger logic error: current mismatch (got 32A, expected 26.7A)
[lp-2 ] DEBUG 2025/02/19 10:44:48 pv charge current: 27A = 24A + 2.99A (-688W @ 1p)
[ocpp ] TRACE 2025/02/19 10:44:48 send garaz: [2,"125413954","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":30,"stackLevel":10,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2025-02-19T09:43:48Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":26.9}]}}}]
[ocpp ] TRACE 2025/02/19 10:44:50 recv garaz: [3,"125413954",{"status": "Accepted"}]
[lp-2 ] DEBUG 2025/02/19 10:44:50 max charge current: 27A
[ocpp ] TRACE 2025/02/19 10:44:54 recv sloupek: [2,"1615546590","Heartbeat",{}]
[ocpp ] TRACE 2025/02/19 10:44:54 send sloupek: [3,"1615546590",{"currentTime":"2025-02-19T09:44:54Z"}]
[... 223 lines cut ...]
[proxy-5021] TRACE 2025/02/19 10:46:19 read input: 0018
[battery] TRACE 2025/02/19 10:46:19 modbus: recv 01 39 00 00 00 05 01 04 02 00 18
[site ] DEBUG 2025/02/19 10:46:19 battery 1 soc: 24%
[site ] DEBUG 2025/02/19 10:46:19 site power: 896W
[lp-1 ] DEBUG 2025/02/19 10:46:19 charge voltages: [240 250 243]V
[lp-1 ] DEBUG 2025/02/19 10:46:19 detected connected phases: 3p
[lp-1 ] DEBUG 2025/02/19 10:46:19 charge total import: 787.376kWh
[lp-1 ] DEBUG 2025/02/19 10:46:19 charger status: A
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.1

mmargi, 2025-02-19:
thank you. It helped.

closed 2025-02-20 by andig

mmargi, 2025-02-20:
So unfortunately the problem still persists. After some investigation I think that something (or the wallbox itself) is irregularly changing the charging current to 32A. Attached are the logs and trace. If anyone can, please look at the time between CET 2025/02/20 11:27:15 and 2025/02/20 11:27:45. I tried writing to Wallbox support, but they are not willing to look into their logs for me why this happened. They only tell me that they do not recommend this type of control :-/.

at 2025/02/20 11:27:06 was current value 21.8A
`[ocpp  ] TRACE 2025/02/20 11:27:06 recv sloupek: [2,"1033896826","MeterValues",{"meterValue": [{"timestamp": "2025-02-20T10:27:05Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Active.Import","unit": "W","value": "5123.0"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "807065.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1","measurand": "Current.Import","unit": "A","value": "21.8"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1-N","measurand": "Voltage","unit": "V","value": "235.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2-N","measurand": "Voltage","unit": "V","value": "255.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3-N","measurand": "Voltage","unit": "V","value": "243.0"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Offered","unit": "W","value": "15242"} ]} ],"connectorId": 1,"transactionId": 1739990005}]
`
At CET 2025/02/20 11:27:15, the charger received a command to change the maximum power to 21A

`[ocpp  ] TRACE 2025/02/20 11:27:15 send sloupek: [2,"4046667528","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":30,"stackLevel":10,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2025-02-20T10:26:15Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":21}]} } }]

[ocpp  ] TRACE 2025/02/20 11:27:16 recv sloupek: [3,"4046667528",{"status": "Accepted"}]`

in next second it send meter current 21.8A:

`ocpp  ] TRACE 2025/02/20 11:27:16 recv sloupek: [2,"1265328494","MeterValues",{"meterValue": [{"timestamp": "2025-02-20T10:27:15Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Active.Import","unit": "W","value": "5144.8"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "807080.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1","measurand": "Current.Import","unit": "A","value": "21.8"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1-N","measurand": "Voltage","unit": "V","value": "236.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2-N","measurand": "Voltage","unit": "V","value": "255.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3-N","measurand": "Voltage","unit": "V","value": "243.0"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Offered","unit": "W","value": "22170"} ]} ],"connectorId": 1,"transactionId": 1739990005}]
`
10 second after it was current 27.2

`[ocpp  ] TRACE 2025/02/20 11:27:26 recv sloupek: [2,"30921225","MeterValues",{"meterValue": [{"timestamp": "2025-02-20T10:27:25Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Active.Import","unit": "W","value": "6364.8"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "807096.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1","measurand": "Current.Import","unit": "A","value": "27.2"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1-N","measurand": "Voltage","unit": "V","value": "234.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2-N","measurand": "Voltage","unit": "V","value": "255.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3-N","measurand": "Voltage","unit": "V","value": "244.0"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Offered","unit": "W","value": "22170"} ]} ],"connectorId": 1,"transactionId": 1739990005}]`

10 second after it was 30.9

`[ocpp  ] TRACE 2025/02/20 11:27:36 recv sloupek: [2,"2013528219","MeterValues",{"meterValue": [{"timestamp": "2025-02-20T10:27:35Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "T [... cut ...] nit": "V","value": "243.0"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Offered","unit": "W","value": "22170"} ]} ],"connectorId": 1,"transactionId": 1739990005}]
`
In attachement are full logs of ocpp comunication and shorted logs.

[evcc-20250220-113340-trace.log](https://github.com/user-attachments/files/18895596/evcc-20250220-113340-trace.log)
[evcc-20250220-113334-debug.log](https://github.com/user-attachments/files/18895597/evcc-20250220-113334-debug.log)

mfuchs1984, 2025-02-20:
Wallbox.com unfortunately has numerous issues with the ocpp implementation. It feels like each time they release a new firmware, they fix one thing but mess up two others.

mmargi, 2025-02-21:
yes :-/ ... i was able to isolate problem - i can set value from 6A to 18A without problem. I case i set 19A or more, its set to 18A.... It looks like i will need to make custom charger through homeassistant and wallbox integration :-/.

premultiply, 2025-02-21:
Quick fix: Just set `maxcurrent` to 18 in config?

Broken firmware will not get fixed by just using other software.

mmargi, 2025-02-21:
yes, of course. But that's not a exactly right  in this case. Because Homeassistant is using wallbox integration with their cloud... and max power regulation from their cloud is working.... so i can use HA as bridge between evcc and stupid wallbox cloud... 
I cant see other option, when wallbox support do not want to help me.

andig, 2025-02-21:
Seems they really don't want. 

> They only tell me that they do not recommend this type of control

Essentially means that they don't trust their smart charging implementation.

mmargi, 2025-02-21:
I tried downgrading to version 5 (5.13.17, which the charger originally came with). The issue remains the same. Can't change max charge value correctly. I find this strange—surely someone within the EVCC community must be using chargers from this manufacturer, and if this were an inherent feature, it would have been documented somewhere, right?

I'm worried that I might be doing something wrong, but EVCC seems so straightforward that I have no idea where I could be making a mistake. :)

mmargi, 2025-03-05:
so i was able to root one of my chargers and i have logs from inside. Can anybody look on it? Regulation is working almost time without problems, but sometimes changed to 32A....  

log from evcc and wallbox attached , OCPP command is at **TRACE 2025/03/05 08:57:59 send garaz:** and on wallbox site is line starting at **Mar 05 08:57:58 WB973539 ocppwallbox[1956]: OCPP_STACK|**

And the wallbox is doing magic...but why :) - maybe is decimal number? Can i round it somewhere on evcc site? I'm just guessing...

[evcc-20250305-091401-trace.log](https://github.com/user-attachments/files/19085089/evcc-20250305-091401-trace.log)

[wallbox-garaz.log](https://github.com/user-attachments/files/19085097/wallbox-garaz.log)

```
Mar 05 08:57:59 WB973539 ocppwallbox[1956]: OCPP_STACK|2025-03-05|07:57:59.927|INFO |1956|PeriodsRunner.cpp|58|apply_current_limit::Informing hardware::limit connector ID 1, limit 32.000000, numberPhases 3
Mar 05 08:57:59 WB973539 ocppwallbox[1956]: OCPPWALLBOX/WALLBOX_OCPP|2025-03-05|07:57:59.929|INFO |492086|OCPPServiceStack.cpp|886|setChargingLimit -      ConnectorId: 1
Mar 05 08:57:59 WB973539 ocppwallbox[1956]: OCPP_STACK|2025-03-05|07:57:59.926|INFO |1956|PeriodsRunner.cpp|68|wait_for_seconds_or_exit::Wait for 0 seconds
Mar 05 08:57:59 WB973539 ocppwallbox[1956]: OCPP_STACK|2025-03-05|07:57:59.929|INFO |1956|PeriodsRunner.cpp|58|apply_current_limit::Informing hardware::limit connector ID 1, limit 11.600000, numberPhases 3
Mar 05 08:57:59 WB973539 ocppwallbox[1956]: OCPPWALLBOX/WALLBOX_OCPP|2025-03-05|07:57:59.930|INFO |492087|OCPPServiceStack.cpp|886|setChargingLimit -      ConnectorId: 1
Mar 05 08:57:59 WB973539 ocppwallbox[1956]: OCPPWALLBOX/WALLBOX_OCPP|2025-03-05|07:57:59.931|INFO |492087|OCPPServiceStack.cpp|887|setChargingLimit -            limit: 11.6
Mar 05 08:57:59 WB973539 ocppwallbox[1956]: OCPPWALLBOX/WALLBOX_OCPP|2025-03-05|07:57:59.931|INFO |492087|OCPPServiceStack.cpp|888|setChargingLimit - chargingRateUnit: A
Mar 05 08:57:59 WB973539 ocppwallbox[1956]: OCPPWALLBOX/WALLBOX_OCPP|2025-03-05|07:57:59.932|INFO |492087|OCPPServiceStack.cpp|889|setChargingLimit -     numberPhases: 3
Mar 05 08:57:59 WB973539 ocppwallbox[1956]: OCPPWALLBOX/WALLBOX_OCPP|2025-03-05|07:57:59.932|INFO |492087|OCPPServiceStack.cpp|890|setChargingLimit -  minChargingRate: 0
Mar 05 08:57:59 WB973539 ocppwallbox[1956]: OCPPWALLBOX/WALLBOX_OCPP|2025-03-05|07:57:59.931|INFO |492086|OCPPServiceStack.cpp|887|setChargingLimit -            limit: 32
Mar 05 08:57:59 WB973539 ocppwallbox[1956]: OCPPWALLBOX/WALLBOX_OCPP|2025-03-05|07:57:59.936|INFO |492086|OCPPServiceStack.cpp|888|setChargingLimit - chargingRateUnit: A
Mar 05 08:57:59 WB973539 ocppwallbox[1956]: OCPPWALLBOX/WALLBOX_OCPP|2025-03-05|07:57:59.937|INFO |492086|OCPPServiceStack.cpp|889|setChargingLimit -     numberPhases: 3
Mar 05 08:57:59 WB973539 ocppwallbox[1956]: OCPPWALLBOX/WALLBOX_OCPP|2025-03-05|07:57:59.938|INFO |492086|OCPPServiceStack.cpp|890|setChargingLimit -  minChargingRate: 0
Mar 05 08:57:59 WB973539 ocppwallbox[1956]: OCPP_STACK|2025-03-05|07:57:59.943|INFO |1956|JsonEngineLib.cpp|740|ProcessingResponse::Sending Response to CS ([3,"2040326906",{"status": "Accepted"}])
Mar 05 08:57:59 WB973539 ocppwallbox[1956]: OCPPWALLBOX/WALLBOX_OCPP|2025-03-05|07:57:59.950|INFO |492087|WallboxChargePointServerImpl.cpp|380|Setting max charging current to: 12
Mar 05 08:57:59 WB973539 ocppwallbox[1956]: OCPPWALLBOX/WALLBOX_OCPP|2025-03-05|07:57:59.979|INFO |492086|WallboxChargePointServerImpl.cpp|380|Setting max charging current to: 32
```

mfuchs1984, 2025-03-05:
Well, even the successfully executed profiles have a decimal numbers. But if you want to try full Ampere steps, use https://docs.evcc.io/en/docs/reference/configuration/vehicles#features-coarsecurrent in your vehicle configuration.

mfuchs1984, 2025-03-05:
You could try to reduce your interval to 15 seconds and monitor if the frequency of the problem is reduced or increased.

mmargi, 2025-03-12:
> Could be a timing issue in the wallbox, you could try to reduce your interval to 15 seconds (or even 17 to not always hit 1 minute boundaries) and monitor if the frequency of the problem is reduced or increased.

yes,i think it helped. It is working last 7 days without problem....thak you!

typxxi, 2025-11-13:
Just to let you know regarding wallbox firmware

they seem to have different firmware which they give the same release number but the software and the hardware are different.
I have 4 Pulsar Plus and they are hardware wise different which they explained to me they day I told them that i have 4 Pulsar Plus and 3 different firmware versions. 
At the end they told me that due to chip shortages they had sold up to 3 different hardware versions in 2021/2022/2023 especially for the german market to supply the demand during chip shortages and afterwards. That was caused by the shortage of the pi compute models and their way to get something different. 

And hence I have to live with the different firmware versions and sorted them out, the older ones with late firmware updates are mainly sorted out to serve as dedicated Octopus Wallbox for charging at night.  
And I was able to replace 1 to get rid down to 2 hardware- and software versions.