evcc-io/evcc #18979: Cupra Born offline

Loos-Fuxsteiner, 2025-02-19:
### Describe the bug

Meldung lautet seit dem Update auf V0.200.1

cannot create vehicle type 'template': cannot create vehicle type 'cupra': unexpected status: 503 (Service Unavailable)

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
vehicles:
  - name: ev3
    type: template
    template: cupra
    title: Cupra Born # Wird in der Benutzeroberfläche angezeigt (Optional)
    user: **** # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: **** # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
```

### Log details

```text
[main  ] ERROR 2025/02/19 16:18:08 creating vehicle ev3 failed: cannot create vehicle type 'template': cannot create vehicle type 'cupra': unexpected status: 503 (Service Unavailable)
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

thierolm, 2025-02-19:
I have the same issue, but looks like the cura API has a problem ... => service unavailable

kackvogel456, 2025-02-19:
same here,

cupra formentor

android app working, but evcc get no data

[image]

thierolm, 2025-02-19:
Could somebody provide the output of `evcc -l trace vehicle`?

Monki2005, 2025-02-19:
I´ve the same after update v0.200.1
Cupra iPhone App and HA integration is still working

kackvogel456, 2025-02-19:
> Could somebody provide the output of `evcc -l trace vehicle`?

```
[main  ] ERROR 2025/02/19 19:52:06 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'cupra': unexpected status: 503 (Service Unavailable)
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'cupra': unexpected status: 503 (Service Unavailable)
Capacity: 19.7kWh
Features: [Offline Retryable]
```

Bucky2k, 2025-02-19:
Same here (OpenWB SoC module EVCC)

TazerReloaded, 2025-02-20:
> Could somebody provide the output of `evcc -l trace vehicle`?

Here is the full output:

```
[main  ] INFO 2025/02/20 01:24:30 evcc 0.200.1
[main  ] INFO 2025/02/20 01:24:30 using config file: /etc/evcc.yaml
[db    ] INFO 2025/02/20 01:24:30 using sqlite database: /var/lib/evcc/evcc.db
[db    ] TRACE 2025/02/20 01:24:30 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2025/02/20 01:24:30 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/02/20 01:24:30 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/02/20 01:24:30 SELECT * FROM `settings` 44 <nil>
[db    ] TRACE 2025/02/20 01:24:30 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2025/02/20 01:24:30 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2025/02/20 01:24:30 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2025/02/20 01:24:30 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/02/20 01:24:30 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/02/20 01:24:30 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2025/02/20 01:24:30 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2025/02/20 01:24:30 SELECT * FROM `configs` WHERE `configs`.`class` = 3 0 <nil>
[cupra ] TRACE 2025/02/20 01:24:30 GET https://identity.vwgroup.io/oidc/v1/authorize?client_id=30e33736-c537-4c72-ab60-74a7b92cfe83%40apps_vw-dilab_com&code_challenge=4IsB2-w__7B5nQYrKMj1EcMKJOFXHvEIzJNorDgREQ8&code_challenge_method=S256&redirect_uri=cupraconnect%3A%2F%2Fidentity-kit%2Flogin&response_type=code+id_token&scope=openid+profile+mbb&nonce=UzWyssbriwOHtPzyiAXKqfavoFdQSTsbogFuWJDNjcR&state=c934d23f-c986-4c92-88b9-092e5c077178
[cupra ] TRACE 2025/02/20 01:24:30 GET https://identity.vwgroup.io/signin-service/v1/signin/30e33736-c537-4c72-ab60-74a7b92cfe83@apps_vw-dilab_com?relayState=611d725a2a3e06b241354b1ffff9df3b47c785fc
[cupra ] TRACE 2025/02/20 01:24:31 <!DOCTYPE html>
    <html lang="en" xmlns="http://www.w3.org/1999/html">
<head>
    <meta charset="utf-8"/>
    <meta name="viewport"
          content="width=device-width, height=device-height, initial-scale=1, maximum-scale=1, user-scalable=no"/>
    <meta name="identitykit" content="loginIdentifier"/>
    
<script type="text/javascript">window.NREUM||(NREUM={});NREUM.info={"errorBeacon":"bam.eu01.nr-data.net","licenseKey":"NRJS-ecd5929e8859410bbd9","agent":"","beacon":"bam.eu01.nr-data.net","applicationTime":28,"applicationID":"460225982","transactionName":"MhBSZQoZCEdVUxFYCgtaY0EKEQhTd18LRRcKGVxUClcdV1hZAF8RLBFNHgsRAVpdXkUZIiAhGQ==","queueTime":0};
(window.NREUM||(NREUM={})).init={privacy:{cookies_enabled:true},ajax:{deny_list:["bam.eu01.nr-data.net"]},distributed_tracing:{enabled:true}};(window.NREUM||(NREUM={})).loader_config={agentID:"535875755",accountID:"2547829",trustKey:"2217462",xpid:"VgMDVl5RARAEUlFQBQQOXlY=",licenseKey:"NRJS-ecd5929e8859410bbd9",applicationID:"460225982"};;/*! For license information please see nr-loader-spa-1.282.0.min.js.LICENSE.txt */
(()=>{var e,t,r={8122:(e,t,r)=>{"use strict";r.d(t,{a:()=>i});var n=r(944);function i(e,t){try{if(!e||"object"!=typeof e)return(0,n.R)(3);if(!t||"object"!=typeof t)return(0,n.R)(4);const r=Object.create(Object.getPrototypeOf(t),Object.getOwnPropertyDescriptors(t)),o=0===Object.keys(r).length?e:r;for(let a in o)if(void 0!==e[a])try{if(null===e[a]){r[a]=null;continue}Array.isArray(e[a])&&Array.isArray(t[a])?r[a]=Array.from(new Set([...e[a],...t[a]])):"object"==typeof e[a]&&"object"==typeof t[a]?r[a]=i(e[a],t[a]):r[a]=e[a]}catch(e){(0,n.R)(1,e)}return r}catch(e){(0,n.R)(2,e)}}},2555:(e,t,r)=>{"use strict";r.d(t,{Vp:()=>c,fn:()=>s,x1:()=>u});var n=r(384),i=r(8122);const o={beacon:n.NT.beacon,errorBeacon:n.NT.errorBeacon,licenseKey:void 0,applicationID:void 0,sa:void 0,queueTime:void 0,applicationTime:void 0,ttGuid:void 0,user:void 0,account:void 0,product:void 0,extra:void 0,jsAttributes:{},userAttributes:void 0,atts:void 0,transactionName:void 0,tNamePlain:void 0},a={};function s(e){try{const t=c(e);return!!t.licenseKey&&!!t.errorBeacon&&!!t.applicationID}catch(e){return!1}}function c(e){if(!e)throw new Error("All info objects require an agent identifier!");if(!a[e])throw new Error("Info for ".concat(e," was never set"));return a[e]}function u(e,t){if(!e)throw new Error("All info objects require an agent identifier!");a[e]=(0,i.a)(t,o);const r=(0,n.nY)(e);r&&(r.info=a[e])}},9417:(e,t,r)=>{"use strict";r.d(t,{D0:()=>p,gD:()=>m,xN:()=>g});var n=r(3333),i=r(993);const o=e=>{if(!e||"string"!=typeof e)return!1;try{document.createDocumentFragment().querySelector(e)}catch{return!1}return!0};var a=r(2614),s=r(944),c=r(384),u=r(8122);const d="[data-nr-mask]",l=()=>{const e={feature_flags:[],experimental:{marks:!1,measures:!1,resources:!1},mask_selector:"*",block_selector:"[data-nr-block]",mask_input_options:{color:!1,date:!1,"datetime-local":!1,email:!1,month:!1,number:!1,range:!1,search:!1,tel:!1,text:!1,time:!1,url:!1,week:!1,textarea:!1,select:!1,password:!0}};return{ajax:{deny_list:void 0,block_internal:!0,enabled:!0,autoStar [... cut ...]
<head><title>503 Service Temporarily Unavailable</title></head>
<body>
<center><h1>503 Service Temporarily Unavailable</h1></center>
</body>
</html>
[main  ] ERROR 2025/02/20 01:24:32 creating vehicle ev5 failed: cannot create vehicle type 'template': cannot create vehicle type 'cupra': unexpected status: 503 (Service Unavailable)
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'cupra': unexpected status: 503 (Service Unavailable)
Capacity: 55.0kWh
Features: [Offline Retryable]
```

RogerHartmann, 2025-02-20:
same issue here. evcc 0.200.1
`DEBUG
Alle Bereiche
[main ] INFO 2025/02/20 07:02:27 evcc 0.200.1
[main ] INFO 2025/02/20 07:02:27 using config file: /etc/evcc.yaml
[db ] INFO 2025/02/20 07:02:27 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2025/02/20 07:02:28 listening at :7070
[main ] ERROR 2025/02/20 07:02:31 creating vehicle Cupra_Born failed: cannot create vehicle type 'template': cannot create vehicle type 'cupra': unexpected status: 503 (Service Unavailable)
[lp-1 ] WARN 2025/02/20 07:02:31 poll mode '{connected 15m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
...
`

aspengler, 2025-02-20:
Betrifft auch 0.133.0

Kommunikation sieht aus wie bei @TazerReloaded

Bucky2k, 2025-02-20:
Es dürfte unabhängig des EVCC Updates sein, da die Binary bei openWB auch nicht geändert wurde. Es liegt also am Service.

Ich befüttere die OpenWB nun behelfsweise über MQTT von seiten der Home Assistant Cupra Integration

WilMa5301, 2025-02-20:
Ich hänge mich hier mal mit dran (trotz Skoda Octavia):

[main ] ERROR 2025/02/20 08:53:07 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'skoda': unexpected status: 503 (Service Unavailable)

Bin noch auf v0.132.1. Zugriff über MySkôda funktioniert.

andig, 2025-02-20:
Looks like VAG Login is broken.

andig, 2025-02-20:
Pls also check T&C

Loos-Fuxsteiner, 2025-02-20:
Cupra App funktioniert einwandfrei. Login auf Website funktioniert, keine neuen AGB's zu bestätigen/vorhanden.

niederradenberg, 2025-02-20:
Ist es möglich, mal zu gucken wie die Cupra „we connect“ integration mit der cupra api spricht? Die funktioniert in der gleichen Homeassistant Instanz weiterhin problemlos.

zachelnet, 2025-02-20:
So wie es aus sieht, ist `https://tokenrefreshservice.apps.emea.vwapps.io/exchangeAuthCode` down. Anschein nutzt die "we connect" Integration ([WeConnect-python](https://github.com/tillsteinbach/WeConnect-python)) ein andere weg den Token zu refreshen.  [Refresh-Url](https://github.com/tillsteinbach/WeConnect-python/blob/main/weconnect/auth/my_cupra_session.py#L24C46-L24C102) wird `https://identity.vwgroup.io/oidc/v1/token` verwendet.

Keine Ahnung, wie dies in Go umgesetzt werden kann.

andig, 2025-02-20:
Wäre es möglich mal den ganzen HTTP Flow zu tracen?

zachelnet, 2025-02-20:
> Wäre es möglich mal den ganzen HTTP Flow zu tracen?

Hab entsprechen ein Example aus [WeConnect-Cupra-python](https://github.com/daernsinstantfortress/WeConnect-Cupra-python) getraced. Kann ich dir die Logs als age (per ssh-key) Verschlüsselt zukommen lassen? 

Verschlüsselte Logs: [debug.cupra.txt](https://github.com/user-attachments/files/18894537/debug.cupra.txt)
Example:  [allVehicles.txt](https://github.com/user-attachments/files/18894591/allVehicles.txt)

zum Entschlüsseln einfach
```
age -d -i .ssh/id_ed25519 out.debug.cupra.text -o out.txt 
```

zachelnet, 2025-02-20:
Anbei ein Trace von evcc, wenns du noch was brauchst sag Bescheid.
`./evcc vehicle --database ~/Downloads/evcc.db --config ~/Downloads/evcc.yaml --log trace`

[evcc.trace.vehicle.txt](https://github.com/user-attachments/files/18894757/evcc.trace.vehicle.txt)

repo-fredo, 2025-02-20:
Cupra WeConnect geht bei mir mittlerweile auch nicht mehr.

andig, 2025-02-21:
Das klingt sehr nach T&C, bitte nichmal schauen

aspengler, 2025-02-21:
Ok, sorry, das war mir nicht bewusst. Jetzt kommt wieder die 503...

andig, 2025-02-21:
@zachelnet Log hab ich, aber für mich unlesbar. Gibts ne Chance das einfach durch Proxyman laufen zu lassen so dass es auch wie HTTP aussieht statt eines Python Logs?

zachelnet, 2025-02-21:
@andig anbei die Logs über Proxyman

[cupra-access.proxymanlogv2.txt](https://github.com/user-attachments/files/18902733/cupra-access.proxymanlogv2.txt)

Vogelbecker, 2025-02-21:
Affects me also with a Cupra Formentor eHybird.

I've just created a Proxyman log with ssl proxy on my iOS device. 
Since it contains some personal data, I would not like to upload it here. Can I send it to someone by mail?

loeffelpan, 2025-02-21:
Das gleiche hier für meinen Skoda Superb iV.
Ich schätze die Ursache ist identisch.
Werden Logs für Skoda benötigt oder ein extra Issue?

N-DEE1, 2025-02-22:
Das gleiche bei unserem Skoda Citigo e-iv

Simpson474, 2025-02-22:
My knowledge of Go is close to zero, therefore, I can just provide a dirty hack to get it back working based on the Python implementation mentioned by @zachelnet. With this patch the call to `Exchange()` is working for My CUPRA (tested with CUPRA Born). I did not test whether the call to `Refresh()`  is also working. For a real fix the forwarding of the `AuthParams` to the `tokenrefreshservice` would be needed to make it work for other VAG services like MyŠkoda. I also don't know whether the `client_secret` is the same for all VAG services, I have taken it from the Python implementation.
```diff
diff --git a/vehicle/seat/cupra/params.go b/vehicle/seat/cupra/params.go
index ce38d946c..6c0858759 100644
--- a/vehicle/seat/cupra/params.go
+++ b/vehicle/seat/cupra/params.go
@@ -5,8 +5,8 @@ import "net/url"
 // Authorization parameters
 var AuthParams = url.Values{
 	"response_type": {"code id_token"}, // token
-	"client_id":     {"30e33736-c537-4c72-ab60-74a7b92cfe83@apps_vw-dilab_com"},
-	"redirect_uri":  {"cupraconnect://identity-kit/login"},
+	"client_id":     {"3c756d46-f1ba-4d78-9f9a-cff0d5292d51@apps_vw-dilab_com"},
+	"redirect_uri":  {"cupra://oauth-callback"},
 	"scope":         {"openid profile mbb"}, // cars birthdate nickname address phone
 }
 
diff --git a/vehicle/vag/tokenrefreshservice/endpoint.go b/vehicle/vag/tokenrefreshservice/endpoint.go
index ecd5b5f34..cf1a483ea 100644
--- a/vehicle/vag/tokenrefreshservice/endpoint.go
+++ b/vehicle/vag/tokenrefreshservice/endpoint.go
@@ -12,9 +12,7 @@ import (
 )
 
 const (
-	BaseURL         = "https://tokenrefreshservice.apps.emea.vwapps.io"
-	CodeExchangeURL = BaseURL + "/exchangeAuthCode"
-	RefreshTokenURL = BaseURL + "/refreshTokens"
+	BaseURL         = "https://identity.vwgroup.io/oidc/v1/token"
 )
 
 var _ vag.TokenExchanger = (*Service)(nil)
@@ -32,20 +30,22 @@ func New(log *util.Logger, q url.Values) *Service {
 }
 
 func (v *Service) Exchange(q url.Values) (*vag.Token, error) {
-	if err := urlvalues.Require(q, "id_token", "code"); err != nil {
+	if err := urlvalues.Require(q, "id_token", "code", "state"); err != nil {
 		return nil, err
 	}
 
 	data := url.Values{
-		"auth_code": {q.Get("code")},
-		"id_token":  {q.Get("id_token")},
+		"client_id":     {"3c756d46-f1ba-4d78-9f9a-cff0d5292d51@apps_vw-dilab_com"},
+		"client_secret": {"eb8814e641c81a2640ad62eeccec11c98effc9bccd4269ab7af338b50a94b3a2"},
+		"grant_type":    {"authorization_code"},
+		"redirect_uri":  {"cupra://oauth-callback"},
 	}
 
 	urlvalues.Merge(data, v.data, q)
 
 	var res vag.Token
 
-	req, err := request.New(http.MethodPost, CodeExchangeURL, strings.NewReader(data.Encode()), request.URLEncoding)
+	req, err := request.New(http.MethodPost, BaseURL, strings.NewReader(data.Encode()), request.URLEncoding)
 	if err == nil {
 		err = v.DoJSON(req, &res)
 	}
@@ -55,6 +55,8 @@ func (v *Service) Exchange(q url.Values) (*vag.Token, error) {
 
 func (v *Service) Refresh(token *vag.Token) (*vag.Token, error) {
 	data := url.Values{
+		"client_id":     {"3c756d46-f1ba-4d78-9f9a-cff0d5292d51@apps_vw-dilab_com"},
+		"client_secret": {"eb8814e641c81a2640ad62eeccec11c98effc9bccd4269ab7af338b50a94b3a2"},
 		"grant_type":    {"refresh_token"},
 		"refresh_token": {token.RefreshToken},
 	}
@@ -63,7 +65,7 @@ func (v *Service) Refresh(token *vag.Token) (*vag.Token, error) {
 
 	var res vag.Token
 
-	req, err := request.New(http.MethodPost, RefreshTokenURL, strings.NewReader(data.Encode()), request.URLEncoding)
+	req, err := request.New(http.MethodPost, BaseURL, strings.NewReader(data.Encode()), request.URLEncoding)
 	if err == nil {
 		err = v.DoJSON(req, &res)
 	}
```

zachelnet, 2025-02-22:
For Cupra it works, but the Client_ID is only specific for Seat.
I think we need to wait for Andy to change the code, because it's not too easy to implement for all brands.

andig, 2025-02-22:
@Simpson474 thank you for the diff. It looks to me as if the tokenrefreshservice is no longer used and instead the VW Identity service's token endpoint is being used.

Monki2005, 2025-02-22:
> [@Simpson474](https://github.com/Simpson474) thank you for the diff. It looks to me as if the tokenrefreshservice is no longer used and instead the VW Identity service's token endpoint is being used.

Kann ich hier nun etwas ändern, damit es wieder funktioniert, oder soll man auf ein evcc Update warten?

TazerReloaded, 2025-02-22:
@Monki2005 Du kannst den Patch anwenden und EVCC selber kompilieren.
Da das ziemlich viel Aufwand ist, hab ich es mit der aktuellen Git-Version (0.200.2) gemacht und das Ergebnis hier angehängt.
Entpacken und nach /usr/bin/evcc kopieren.

In meinem kurzen Test sah alles gut aus, mein Born wird wieder korrekt angezeigt.
Benutzung auf eigene Gefahr! Entspricht einem Nightly-Build.
[evcc.zip](https://github.com/user-attachments/files/18925894/evcc.zip)

gnujuba, 2025-02-22:
> [@Monki2005](https://github.com/Monki2005) Du kannst den Patch anwenden und EVCC selber kompilieren. Da das ziemlich viel Aufwand ist, hab ich es mit der aktuellen Git-Version (0.200.2) gemacht und [...]

Gibts davon zufällig auch ein ARM binary? 😇 😃

TazerReloaded, 2025-02-23:
Ich habe noch ein paar andere Varianten gebaut. Jetzt ist hoffentlich für jeden as dabei :wink:

[evcc_arm64.zip](https://github.com/user-attachments/files/18926479/evcc_arm64.zip)
[evcc_armhf.zip](https://github.com/user-attachments/files/18926481/evcc_armhf.zip)
[evcc_armv6.zip](https://github.com/user-attachments/files/18926480/evcc_armv6.zip)
[evcc_mac-amd64.zip](https://github.com/user-attachments/files/18926483/evcc_mac-amd64.zip)
[evcc_windows-amd64.zip](https://github.com/user-attachments/files/18926482/evcc_windows-amd64.zip)

gnujuba, 2025-02-23:
> Ich habe noch ein paar andere Varianten gebaut. Jetzt ist hoffentlich für jeden was dabei 😉
> 
> [evcc_arm64.zip](https://github.com/user-attachments/files/18926479/evcc_arm64.zip) [evcc_armhf.zip](https://github.com/user-attachments/files/18926481/evcc_armhf.zip) [evcc_armv6.zip](https://github.com/user-attachments/files/18926480/evcc_armv6.zip) [evcc_mac-amd64.zip](https://github.com/user-attachments/files/18926483/evcc_mac-amd64.zip) [evcc_windows-amd64.zip](https://github.com/user-attachments/files/18926482/evcc_windows-amd64.zip)

Super, vielen Dank für die Builds. Bei mir will leider meine tolle elli Wallbox nicht mit dem arm64 binary zusammenarbeiten - eebus timeout. Will jetzt aber den Thread nicht noch mehr aufblähen.

andig, 2025-02-23:
Ich bin dran, die Vorlage hat sehr geholfen, muss aber so umgebaut werden, dass sie auch außerhalb Seat weiter funktionert.

Eine Frage: trifft es wirklich nur Seat oder auch Skoda?

N-DEE1, 2025-02-23:
Also bei uns geht der Skoda citigo auch nicht mehr.

bigdog-carsten, 2025-02-23:
+1 (Skoda Citigo not working)

AlexZ1402, 2025-02-23:
> Ich bin dran, die Vorlage hat sehr geholfen, muss aber so umgebaut werden, dass sie auch außerhalb Seat weiter funktionert. Anscheinend haben sich auch einige Datenpunkte verschoben, dazu habe ich aber noch kein klares Bild.
> 
> Eine Frage: trifft es wirklich nur Seat oder auch Skoda?

Hallo,
Betrifft komplette VAG Gruppe / also alle Fahrzeuge die online-Verbindung haben 
Ich schätze wegen Daten leak was VW hatte 

👋🏻

loeffelpan, 2025-02-23:
> Das Gleiche hier für meinen Skoda Superb iV. Ich schätze die Ursache ist identisch. Werden Logs für Skoda benötigt oder ein extra Issue?

Wie zuvor schon gemeldet. Betrifft Skoda ebenfalls.
Ob es VW selber auch trifft, konnte ich bisher nicht rauslesen.

pampelschuster, 2025-02-23:
VW scheint nicht betroffen zu sein. Bei unserem ID.3 funktioniert noch alles.

andig, 2025-02-23:
I've added a PR for Seat- would be great if this could be tested. It's only a quick hack rather than a clean solution.

andig, 2025-02-23:
> Wie zuvor schon gemeldet. Betrifft Skoda ebenfalls.

In der Annahme dass die Änderung hier identisch ist brauchen wir die neuen client id/secret um das analog anzupassen.

aspengler, 2025-02-23:
Ich werde mal versuchen, evcc selbst zu bauen. Ich wäre dennoch sehr dankbar, wenn jemand die Änderung auch in 0.133 einpflegen könnte...

thierolm, 2025-02-23:
PR #19094 ist schon in der Mache.

markimarc66, 2025-02-23:
Mit unserem Cupra Born funktioniert's wieder (v0.200.2, in Konfiguration bei Status prüfen: erfolgreich)

N-DEE1, 2025-02-23:
> > Wie zuvor schon gemeldet. Betrifft Skoda ebenfalls.
> 
> In der Annahme dass die Änderung hier identisch ist brauchen wir die neuen client id/secret um das analog anzupassen.

Wie kommt man zu dieser client id/secret ?

Simpson474, 2025-02-23:
The Python implementation [CarConnectivity](https://github.com/tillsteinbach/CarConnectivity) (successor of [WeConnect-python](https://github.com/tillsteinbach/WeConnect-python)) does not use a "secret" for Skoda, but a different URL directly from Skoda instead of the one from Volkswagen. It is really strange why VAG hosts so many different services even though the cars are all built using the same parts. Another possibility would be tracing the MySkôda app for example using Proxyman to see which URL and secret is used by the app.

BHeuk, 2025-02-23:
> Mit unserem Cupra Born funktioniert's wieder (v0.200.2, in Konfiguration bei Status prüfen: erfolgreich)

Bei unserem Born leider nicht.

N-DEE1, 2025-02-23:
> The Python implementation [CarConnectivity](https://github.com/tillsteinbach/CarConnectivity) (successor of [WeConnect-python](https://github.com/tillsteinbach/WeConnect-python)) does not use a [...]

As proxyman is not available for Android - does anyone have a similar program?

Monki2005, 2025-02-23:
> > Mit unserem Cupra Born funktioniert's wieder (v0.200.2, in Konfiguration bei Status prüfen: erfolgreich)
> 
> Bei unserem Born leider nicht.

>> Bei meinem Tavascan unktioniert 0.200.2 leider auch nicht - nach wie vor der selbe Fehler bei "prüfen"
      "cannot create vehicle type 'template': cannot create vehicle type 'cupra': unexpected status: 503 (Service Unavailable)"
     Die Cupra-Integrtation im HA funktioniert jedoch weiterhin

closed 2025-02-23 by andig

l3szy, 2025-02-23:
bei Mein Cupra Bron geht es weiterhin nicht:

[main ] ERROR 2025/02/23 15:01:32 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'cupra': unexpected status: 503 (Service Unavailable)

Vogelbecker, 2025-02-23:
> bei Mein Cupra Bron geht es weiterhin nicht:
> 
> [main ] ERROR 2025/02/23 15:01:32 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'cupra': unexpected status: 503 (Service Unavailable)

Bei welcher Version? v0.200.2 oder der Master?

l3szy, 2025-02-23:
> > bei Mein Cupra Bron geht es weiterhin nicht:
> > [main ] ERROR 2025/02/23 15:01:32 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'cupra': unexpected status: 503 (Service Unavailable)
> 
> Bei welcher Version? v0.200.2 oder der Master?

Hi sorry nicht geschrieben frisch 0.200.2 drauf gespielt danach Home Assistant komplett neu gestartet.

Jam-EX, 2025-02-23:
Bei Skoda bekomme ich ebenfalls die gleiche Fehlermeldung (inmer noch).Bin ebenfalls bei v0.200.2.

MiniOh, 2025-02-23:
Danke an alle.
Funktioniert für den Cupra Born.

Schlichi83, 2025-02-23:
Wie kann man das Update bei der Nutzung von evcc als HA Addon kurzfristig einspielen?

l3szy, 2025-02-23:
> Wie kann man das Update bei der Nutzung von evcc als HA Addon kurzfristig einspielen?

Bei mir war es einfach als Updatet verfügbar und könnte einfach installieren.

Max112233445566, 2025-02-23:
Funktioniert hier bei jemandem die Verbindung zum Cupra Born mit EVCC als HA Addon?

Vogelbecker, 2025-02-23:
Ich kenne das Verhalten des HA Plugins nicht. 

Der Fix ist gerade einen Tag alt und ist aktuell in den Master gemerged. 

Ich denke wenn eine neue Version geleased wird, dann ist dort der fix drin.

[45 more replies]