evcc-io/evcc #18984: Car information

Monki2005, 2025-02-19:
### Describe the bug

Today, I'm updated the latest v0.200.0 to v0.200.1. Everything worked fine, but after the update evcc was suddenly no longer able to connect and query the SOC or any information from my Cupra Tavascan.

The MyCupra App on my iphone and the HA integration is still working.

**Log entry:**
Request failed with status code 400.
cannot create vehicle type 'template': cannot create vehicle type 'cupra': unexpected status: 503 (Service Unavailable).

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Configuration over UI
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/02/19 16:26:01 evcc 0.200.1
[main  ] INFO 2025/02/19 16:26:01 using config file: /config/evcc.yaml
[db    ] INFO 2025/02/19 16:26:01 using sqlite database: /data/evcc.db
[main  ] INFO 2025/02/19 16:26:02 listening at :7070
[main  ] ERROR 2025/02/19 16:26:03 creating vehicle db:3 failed: cannot create vehicle type 'template': cannot create vehicle type 'cupra': unexpected status: 503 (Service Unavailable)
[site  ] INFO 2025/02/19 16:26:04 site config:
[site  ] INFO 2025/02/19 16:26:04   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2025/02/19 16:26:04     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2025/02/19 16:26:04     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/02/19 16:26:04     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2025/02/19 16:26:04   vehicles:
[site  ] INFO 2025/02/19 16:26:04     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2025/02/19 16:26:04 loadpoint 1:
[lp-1  ] INFO 2025/02/19 16:26:04   mode:        off
[lp-1  ] INFO 2025/02/19 16:26:04   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2025/02/19 16:26:04   meters:      charge ✓
[lp-1  ] INFO 2025/02/19 16:26:04     charge:    power ✓ energy ✓ currents ✓
[site  ] WARN 2025/02/19 16:26:04 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[lp-1  ] INFO 2025/02/19 16:26:04 car connected
[lp-1  ] INFO 2025/02/19 16:27:45 vehicle updated: unknown -> Cupra Tavascan
[lp-1  ] INFO 2025/02/19 16:30:04 car disconnected
[lp-1  ] INFO 2025/02/19 16:30:04 vehicle updated: Cupra Tavascan -> unknown
[lp-1  ] INFO 2025/02/19 18:36:38 vehicle updated: unknown -> Cupra Tavascan
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.200.1

FantOEM, 2025-02-20:
Hallo zusammen,

hier dasselbe. Aber ich bin mir nicht sicher, ob das nicht auch schon bei der v0.200.0 auftrat. Ich habe gestern ein Update gemacht, weil EVCC mit besagtem Cupra nicht funktioniert hat. Ich habe vorsichtshalber die AGBs nochmals abgesegnet (Umstellung auf Österreich -> Bestätigung -> zurück auf Deutschland -> Bestätigung), weil das beim letzten mal geholfen hat.

Es könnte sich also auch um ein Problem mit der Cupra-API handeln. App geht und zeigt alles richtig an.

Hawedieehre,
FantOEM

StefanSchoof, 2025-02-20:
Siehe https://github.com/evcc-io/evcc/issues/18979

closed 2025-02-20 by andig