evcc-io/evcc #18989: Sessions UI: make charging points editable

naltatis, 2025-02-19:
Ja sollten wir über die UI analog Fahrzeuge änderbar machen.

_Originally posted by @naltatis in https://github.com/evcc-io/evcc/discussions/18949#discussioncomment-12241912_

andig, 2025-02-21:
Ich sehe keinen normalen UC dafür.

StevieC121176, 2025-02-21:
> Ich sehe keinen normalen UC dafür.

ich habe vor kurzem meine Ladepunkte über die UI neu angelegt und sie genau so benannt wie die alten in der yaml. Nun werden mir allerdings die alten in der Übersicht der Ladevorgänge nur noch als `Ladepunkt` angezeigt. Somit hab ich jetzt 4 anstatt 2 verschiedener Ladepunkte in der Übersicht was die ganzen Auswertungen durcheinander bringt.
So könnte man den Ladepunkt wieder ändern.

naltatis, 2025-02-21:
Wir hatten im vergangenen Release auch den Bug, bei dem wir die Ladepunktnamen nicht gespeichert haben. Damit gibts dann zumindest eine Manuelle Recover Möglichkeit (ohne DB Wissen).

andig, 2025-02-21:
Ist ein Edge Case für den ich keine permanente Lösung einbauen würde. 

@StevieC121176 neues Thema, bitte Issue.

PR #19016 (2025-02-22): Sessions: make chargingpoint changeable
Hi,

fix #18989 

Make loadpoint changeable:
[image]
[image]

TODO:
- [ ] (decide whether to) add test(s)
closed 2025-02-22 by andig