evcc-io/evcc #18991: Wieder Invalid Response / Cmd Response Errors auf RCT Power Hardware

AcidCells, 2025-02-19:
### Describe the bug

Hi zusammen,

ich habe leider seit neustem wieder den schon länger gefixten Fehler auf RCT Power Hardware, der im Folgendem beschrieben wird. Das war bereits gelöst und ist schon ewig nicht mehr in meiner Umgebung aufgetreten.
Konkret tauchen im Log folgende Fehler auf (hier als Beispiel exportiert) (siehe angehängtem Log):

[site ] ERROR 2025/02/17 20:45:09 pv 1 power: parsing failed in state 0
[site ] ERROR 2025/02/17 20:45:09 battery 1 power: parsing failed in state 0
[site ] ERROR 2025/02/17 20:45:10 grid power: parsing failed in state 0
[site ] ERROR 2025/02/17 20:48:11 pv 1 power: invalid response to read of 23494E56414C4944: Cmd Response (05) Id #INVALID (FBF3CE97) Data [70 7 159 123]

Ich kann hierzu noch folgende Beobachtung mitteilen:

- Das Problem tritt bei 0.133.0, 0.200.0 und 0.200.1 auf.
- Es war mit 0.124.5 gefixt, seitdem ist der Fehler bis auf jetzt nicht mehr aufgetreten.
- Was kann ich berichten, was "anders" ist: Ich habe vor einiger Zeit ein Firmware-Update auf der RCT-Hardware gemacht (Screenshot der Version), aber ich bin der Meinung, dass mir die Fehler zum ersten Mal nach dem Update auf 0.200.0 aufgefallen sind. Allerdings hat auch das Downgrade auf die 0.133.0 die Fehler nicht mehr beseitigt. Deswegen bin ich etwas verwirrt. Im aktuellen nightly 0.200.1 ist der Fehler auch noch enthalten

Im Anhang noch config, log und RCT Firmware. Ich habe meine "Geheimnisse" manuell aus der config entfernt.

TAUSEND Dank dafür, dass ich euch das anschaut. Evcc ist mein fester Alltagsbestandteil, und ich habe schon viele begeistern können :-)

### Steps to reproduce

1. Evcc auf Rhaspberry Pie starten. RTC-Power DC Wechselrichter muss verbunden sein
2. Nach einiger Zeit tauchen die Fehler im Log auf

Hier noch die Firmware Version meiner RCT-Hardware
[image]

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: XX

interval: 10s # control cycle interval

meters:
- type: template
  template: rct-power 
  usage: grid  
  host: 1xx 
  name: grid1
  minsoc: 7 # Untere Grenze beim Entladen der Batterie (normaler Betrieb) (optional)
  maxsoc: 97 # Oberes Limit beim Laden der Batterie aus dem Netz (optional)
- type: template
  template: rct-power 
  usage: pv  
  host: 1xx
  name: pv2
- type: template
  template: rct-power 
  usage: battery  
  host: 1xx
  name: battery3

vehicles:
- name: kona
  type: template
[... 273 lines cut ...]
[lp-1  ] DEBUG 2025/02/17 21:15:26 charge currents: [0 0 0]A
[site  ] DEBUG 2025/02/17 21:15:27 battery 1 power: 9706W
[site  ] DEBUG 2025/02/17 21:15:27 grid power: 4072W
[site  ] DEBUG 2025/02/17 21:15:27 battery 1 soc: 7%
[site  ] DEBUG 2025/02/17 21:15:27 pv 1 power: 0W
[site  ] DEBUG 2025/02/17 21:15:27 site power: 13878W
[lp-1  ] DEBUG 2025/02/17 21:15:27 charge total import: 5329.375kWh
[lp-1  ] DEBUG 2025/02/17 21:15:27 charger status: A
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.1 (0ef999ef)

TobiasHuber1980, 2025-02-20:
1. ja, ich kann auch feststellen, dass die Fehlermeldungen wieder stark zugenommen haben.

2. 0.124.5 (https://github.com/evcc-io/evcc/commit/4b8fa29963edcfeccb2a4f45174a92424aaa1f55)  bis 0.124.7 (https://github.com/evcc-io/evcc/commit/db3f355ab6fd0c4b72c0a9a94549c044f3a0bb86) waren die einzigen beiden Commits bzgl. RecoverableError.

3. Die letzten großen Änderungen waren "battery control" mit 0.200.0 (0.133.0 ist ja das release davor), gut möglich dass dadurch ein altes Problem wieder zum Leben erweckt wurde. Da können aber nur @andig / @Maschga ein Auge drauf werfen :kissing_heart:. 
  
4. Muss die RTC Libary um "async handling" ergänzt/umgeschrieben werden. https://github.com/mlnoga/rct/issues/19 - leider hat sich bis dato noch niemand gefunden, der das in Angriff nehmen will. (ping @Maschga :rocket:)

andig, 2025-02-20:
Geht denn die 124.5 noch? Falls nein liegts am Gerät.

AcidCells, 2025-02-20:
Hi,

habe gerade ein Downgrade auf 0.124.5 gemacht. Fehler sofort da :-/ Scheint ggf. an der neuen RCT Firmware zu liegen. Wir RCTler sind sehr für einen Fix dankbar. Ich teste auch gerne noch alles weiter, wenn es hilft.

@Maschga : Können wir Motivations-Bier spenden?

Maschga, 2025-02-20:
Hi,
vielen Dank für das Angebot. Geld will ich dafür aber nicht annehmen.
Ich habe bereits mehrfach (mit Hilfe von KI) versucht, die RCT-Go-Library zu verbessern, habe es aber dann wieder gelassen.

Ich müsste mich dafür in die [RCT-Python-Implementierung](https://github.com/svalouch/python-rctclient) einlesen und besser verstehen, wie sich eine asynchrone Implementierung realisieren liese. Das will ich aber derzeit nicht angehen. Dafür fehlt mir derzeit die Zeit und die Lust.
Auf dem ToDo-Zettel ist es, aber nicht sehr weit oben. Dieses Thema darf sich auch gerne wer anderes annehmen.

Bis jetzt bekomme ich nicht viele `Invalid Response / Cmd Response Errors`, deshalb ist das Thema für mich auch nicht so wichtig, sorry. Ich vermeide es, in die RCT-App zu sehen, wenn evcc läuft. Das vermeidet viele Fehler, vielleicht hilft das auch bei dir.

AcidCells, 2025-02-20:
Hi, danke dir für deine Antwort.

Verständlich. Wäre es vielleicht möglich, dass ihr einen Workaround einbaut, so was wie "on error repeat last query after one second"? Dann wäre das Problem zumindest umgangen.

Ich benutze die RCT App nicht gleichzeitig und es erscheinen trotzdem wieder die Fehler, sie damals ja schon gefixt wurden. Das ist ja das mysteriöse. 

VG an euch.

Maschga, 2025-02-21:
Genau das passiert bereits: https://github.com/evcc-io/evcc/blob/master/meter%2Frct.go#L233
Aber anscheinend mit der neuen RCT Version weniger erfolgreich.

AcidCells, 2025-02-22:
Verdammt... okay, also vielleicht hat ja doch noch jemand Lust, sich dem Problem zu widmen. Es ist zumindest so, dass die Fehler deutlich weniger auftauchen, als in den älteren. 
Ich würde auch mein schönstes Bettelgesicht aufsetzten. Ich verstehe aber auch, dass dieses Projekt den Invest der Freizeit bedeutet.

AcidCells, 2025-02-23:
Hi zusammen 

Ich habe einen Workaround gefunden, durch den die Fehler verschwinden. Durch das aktuelle Firmwareupdate (siehe Screenshot) oben, muss es eine Veränderung mit den Abfragen des Portal-Service gegeben haben. Meine Lösung ist: deaktiviert man den Portal-Service in den Einstellungen in der RCT App, dann verschwinden die Fehler. Da ich damit gut leben kann, ist das für mich zumindest ein Workaround. Ich würde das Issue trotzdem noch auf open lassen, da das eigentliche Problem ja noch existiert.

VG

TobiasHuber1980, 2025-02-24:
Nope, das war mit den vorherigen Firmware Versionen auch schon so.

AcidCells, 2025-03-08:
@andig : Ich hoffe, ihr seid nicht total genervt, aber mit Version 200.6 sind leider wieder neue "Fehler" gekommen:

[site  ] ERROR 2025/03/08 19:04:43 pv 1 power: timeout
[site  ] ERROR 2025/03/08 19:04:43 battery 1 power: timeout
[site  ] ERROR 2025/03/08 19:04:53 battery 1 soc: timeout
[site  ] ERROR 2025/03/08 19:04:53 grid power: timeout

Die sind mit 200.5 nicht da (dafür habe ich das Audi/VW-Problem, dass bei mir auch mit 200.8 gefixt ist, Vielen Dank dafür!!) 

Im Anhang mein Log, und vielen Dank, dass ihr und du euch diesen Problemen so schnell annehmt! (Die Infos vom Anfang des Issues sind ja noch vorhanden, falls mehr Infos über meine Konfig gebraucht werden)

[evcc-20250308-195028-debug.log](https://github.com/user-attachments/files/19145380/evcc-20250308-195028-debug.log)

TobiasHuber1980, 2025-03-08:
@AcidCells habe ich bei mir auch, sind aber nur im log, das rote dreieck kommt bei mir mittlerweile sehr selten (aber es kommt manchmal). Spiel dich mal bisschen mit dem Cache-Timing.

@andig vielleicht kann man ja die zeitspanne für den timeout (ich glaube ist momentan 15 sekunden) noch erhöhen....? wir wissen ja mittlerweile wie langsam die rct dinger sind....

AcidCells, 2025-03-08:
@TobiasHuber1980 Ich habe sie leider relativ viel und oft... Spielen mit dem Cache-Timing? Was meinst du damit? Ich habe nur das Timing für den Kontakt zum Auto gefunden.

TobiasHuber1980, 2025-03-09:
[image]

Das da...

AcidCells, 2025-03-09:
Verstehe, aber im der Beschreibung steht ja, dass sich das auf das Fahrzeug bezieht. Ich habe das daher nicht im Bezug zu den Fehlern zu PV, Batterie und Grid genommen. Aktuell habe ich die gleichen Einstellungen, wie du, und die Fehler sind da, wohlgemerkt, mit 200.5 waren sie nicht da.

TobiasHuber1980, 2025-03-09:
Ja die Beschreibung ist noch falsch. 

Vielleicht kann @andig ja das timeout noch auf z.B. 30-40 Sekunden ändern?

andig, 2025-03-09:
Leider haben mehrere Arbeitstage Aufwand für das komplette Umschreiben der Bibliothek in asynchrones Handling nicht den erwünschten Erfolg gebracht. Fakt ist, dass dieses Gerät aus der Hölle manchmal einfach nicht auf Anfrage antwortet, selbst wenn man 30s darauf wartet. Der neue PR ist nochmal ein Versuch, das mittels Retry zu verbessern. Probehalber wäre es gut, ein bisschen mit `cache` zu spielen- zwischen 2 und 10s.

Feedback welcome.

AcidCells, 2025-03-09:
Ich teste so viel und alles was geht und gebe dann Feedback. Vielen Dank für euren Einsatz!