evcc-io/evcc #19002: IDM Wärmepumpen

motze92, 2025-02-20:
**Is your feature request related to a problem? Please describe.**
IDM Wärmepumpen per Modbus anbinden

**Describe the solution you'd like**
Bitte ein Template für IDM Wärmepumpen erstellen. Sie können per Modbus angesteuert werden.

Interresant ist vor allem Register 74. Damit kann der WP der aktuelle Überschuss in Watt mitgeteilt werden.
Untenstehend die vollständige Tabelle.

Welche 

**Additional context**
<table id="idm_info"><tbody><tr><th>ID</th><th>Name</th><th>Access</th><th align="center">Value</th><th align="right">GLT Value</th></tr><tr><td>74</td><td>Überschuss</td><td>RW</td><td align="center">0</td><td align="right"></td></tr><tr><td>76</td><td>Leistung E-Heizstab</td><td>RW</td><td align="center">0</td><td align="right"></td></tr><tr><td>78</td><td>PV Leistung</td><td>RW</td><td align="center">0</td><td align="right"></td></tr><tr><td>82</td><td>Hausverbrauch</td><td>RW</td><td align="center">0</td><td align="right"></td></tr><tr><td>84</td><td>Batterieentladung</td><td>RW</td><td align="center">0</td><td align="right"></td></tr><tr><td>86</td><td>Batterie Füllstand</td><td>RW</td><td align="center">-1</td><td align="right"></td></tr><tr><td>1000</td><td>Außentemperatur</td><td>RO</td><td align="center">6.7</td><td align="right"></td></tr><tr><td>1002</td><td>Mittelwert Außentem.</td><td>RO</td><td align="center">-1.5</td><td align="right"></td></tr><tr><td>1004</td><td>interne Meldung</td><td>RO</td><td align="center">126.0</td><td align="right"></td></tr><tr><td>1005</td><td>Betriebsart System</td><td>RW</td><td align="center">1</td><td align="right"></td></tr><tr><td>1008</td><td>Wärmespeichertemperatur</td><td>RO</td><td align="center">33.8</td><td align="right"></td></tr><tr><td>1010</td><td>Kältespeichertemperatur</td><td>RO</td><td align="center">25.7</td><td align="right"></td></tr><tr><td>1012</td><td>Trinkwarmwassererwärmerfühler unten</td><td>RO</td><td align="center">28.2</td><td align="right"></td></tr><tr><td>1014</td><td>TW-Erw. Temp.oben</td><td>RO</td><td align="center">44.5</td><td align="right"></td></tr><tr><td>1033</td><td>Warmwasserladung Einschalttemperatur</td><td>RW</td><td align="center">35</td><td align="right"></td></tr><tr><td>1034</td><td>Warmwasserladung Ausschalttemperatur</td><td>RW</td><td align="center">42</td><td align="right"></td></tr><tr><td>1050</td><td>Wärmepumpenvorlauftemperatur</td><td>RO</td><td align="center">28.4</td><td align="right"></td></tr><tr><td>1052</td><td>Rücklauffühler</td><td>RO</td><td align="center">23.9</td><td align="right"></td></tr><tr><td>1060</td><td>Ansauglufttemperatur</td><td>RO</td><td align="center">6.4</td><td align="right"></td></tr><tr><td>1090</td><td>Betriebsart Wärmepumpe</td><td>RO</td><td align="center">0.0</td><td align="right"></td></tr><tr><td>1091</td><td>Heizanforderung</td><td>RO</td><td align="center">0.0</td><td align="right"></td></tr><tr><td>1092</td><td>Kühlanforderung</td><td>RO</td><td align="center">0.0</td><td align="right"></td></tr><tr><td>1093</td><td>Vorranganforderung</td><td>RO</td><td align="center">0.0</td><td align="right"></td></tr><tr><td>1098</td><td>EW/EVU Sperrkontakt</td><td>RO</td><td align="center">1.0</td><td align="right"></td></tr><tr><td>1099</td><td>Summenstörung Regler</td><td>RO</td><td align="center">0.0</td><td align="right"></td></tr><tr><td>1100</td><td>Verdichter 1</td><td>RO</td><td align="center">0.0</td><td align="right"></td></tr><tr><td>1104</td><td>Ladepumpe</td><td>RO</td><td align="center">-1.0</td><td align="right"></td></tr><tr><td>1110</td><td>Ventil Heizen/Kühlen</td><td>RO</td><td align="center">0.0</td><td align="right"></td></tr><tr><td>1111</td><td>Ventil Wärme/Kälte</td><td>RO</td><td align="center">0.0</td><td align="right"></td></tr><tr><td>1112</td><td>Ventil Heizen/Vorrang</td><td>RO</td><td align="center">0.0</td><td align="right"></td></tr><tr><td>1120</td><td>Bivalenzpunkt 1</td><td>RW</td><td align="center">-5</td><td align="right"></td></tr><tr><td>1121</td><td>Bivalenzpunkt 2</td><td>RW</td><td align="center">-20</td><td align="right"></td></tr><tr><td>1122</td><td>Bivalenzpunkt 1</td><td>RW</td><td align="center">-5</td><td align="right"></td></tr><tr><td>1123</td><td>Bivalenzpunkt 2</td><td>RW</td><td align="center">-20</td><td align="right"></td></tr><tr><td>1124</td><td>2./3. Wärmeerzeuger</td><td>RO</td><td align="center">0.0</td><td align="right"></td></tr><tr><td>1350</td><td>Vorlauftemperatur HK A</td><td>RO</td><td align="center">28.9</td><td align="right"></td></tr><tr><td>1378</td><td>Sollvorlauftemp. HK A</td><td>RO</td><td align="center">28.7</td><td align="right"></td></tr><tr><td>1392</td><td>Feuchtesensor</td><td>RO</td><td align="center">Kurzschluss</td><td align="right"></td></tr><tr><td>1393</td><td>Betriebsart Heizkreis A</td><td>RW</td><td align="center">2</td><td align="right"></td></tr><tr><td>1401</td><td>Raumtemperatur Heizen Normal</td><td>RW</td><td align="center">22</td><td align="right"></td></tr><tr><td>1415</td><td>Raumtemperatur Heizen Eco</td><td>RW</td><td align="center">18</td><td align="right"></td></tr><tr><td>1429</td><td>Heizkurve</td><td>RW</td><td align="center">0.3</td><td align="right"></td></tr><tr><td>1442</td><td>Heizgrenze</td><td>RW</td><td align="center">15</td><td align="right"></td></tr><tr><td>1457</td><td>Raumtemperatur Kühlen Normal</td><td>RW</td><td align="center">15</td><td align="right"></td></tr><tr><td>1471</td><td>Raumtemperatur Kühlen Eco</td><td>RW</td><td align="center">15</td><td align="right"></td></tr><tr><td>1484</td><td>Kühlgrenze</td><td>RW</td><td align="center">20</td><td align="right"></td></tr><tr><td>1491</td><td>Vorlauftemperatur Kühlen</td><td>RW</td><td align="center">15</td><td align="right"></td></tr><tr><td>1498</td><td>aktive Betriebsart HK A</td><td>RO</td><td align="center">1.0</td><td align="right"></td></tr><tr><td>1505</td><td>Parallelverschiebung</td><td>RW</td><td align="center">2</td><td align="right"></td></tr><tr><td>1650</td><td>externe Raumtemperatur HKA</td><td>W</td><td align="center">-1</td><td align="right"></td></tr><tr><td>1652</td><td>externe Raumtemperatur HKB</td><td>W</td><td align="center">-1</td><td align="right"></td></tr><tr><td>1654</td><td>externe Raumtemperatur HKC</td><td>W</td><td align="center">-1</td><td align="right"></td></tr><tr><td>1656</td><td>externe Raumtemperatur HKD</td><td>W</td><td align="center">-1</td><td align="right"></td></tr><tr><td>1658</td><td>externe Raumtemperatur HKE</td><td>W</td><td align="center">-1</td><td align="right"></td></tr><tr><td>1660</td><td>externe Raumtemperatur HKF</td><td>W</td><td align="center">-1</td><td align="right"></td></tr><tr><td>1662</td><td>externe Raumtemperatur HKG</td><td>W</td><td align="center">-1</td><td align="right"></td></tr><tr><td>1690</td><td>externe Außentemperatur</td><td>W</td><td align="center">-1</td><td align="right"></td></tr><tr><td>1692</td><td>externe Feuchte</td><td>W</td><td align="center">-1</td><td align="right"></td></tr><tr><td>1694</td><td>Externe Ansteuerung Heiztemperatur</td><td>RW</td><td align="center">40</td><td align="right"></td></tr><tr><td>1695</td><td>Externe Ansteuerung Kühltemperatur</td><td>RW</td><td align="center">10</td><td align="right"></td></tr><tr><td>1696</td><td>GLT Temperaturanforderung Heizen</td><td>W</td><td align="center">0</td><td align="right"></td></tr><tr><td>1698</td><td>GLT Temperaturanforderung Kühlen</td><td>W</td><td align="center">1e+02</td><td align="right"></td></tr><tr><td>1710</td><td>Externe Ansteuerung Heizen</td><td>RW</td><td align="center">0</td><td align="right"></td></tr><tr><td>1711</td><td>Externe Ansteuerung Kühlen</td><td>RW</td><td align="center">0</td><td align="right"></td></tr><tr><td>1712</td><td>Externe Warmwasserladung</td><td>RW</td><td align="center">0</td><td align="right"></td></tr><tr><td>1713</td><td>Externe Ansteuerung einmalige Warmwasserladung</td><td>RW</td><td align="center">0</td><td align="right"></td></tr><tr><td>1714</td><td>externe Anforderung Sole-/Zwischenkreispumpe</td><td>RW</td><td align="center">-2</td><td align="right"></td></tr><tr><td>1715</td><td>externe Anforderung Grundwasserpumpe</td><td>RW</td><td align="center">-2</td><td align="right"></td></tr><tr><td>1716</td><td>GLT Wärmespeichertemperatur</td><td>W</td><td align="center">-1</td><td align="right"></td></tr><tr><td>1718</td><td>GLT Kältespeichertemperatur</td><td>W</td><td align="center">-1</td><td align="right"></td></tr><tr><td>1720</td><td [... cut ...] d><td align="right"></td></tr><tr><td>4124</td><td>elektrische Gesamtleistung (falls kein Wert vom Inverter vorhanden ist, wird jener vom Polynom verwendet)</td><td>RO</td><td align="center">0.0</td><td align="right"></td></tr><tr><td>4126</td><td>thermische Leistung (basierend auf Durchflusssensor)</td><td>RO</td><td align="center">0.0</td><td align="right"></td></tr><tr><td>4128</td><td>Wärmemenge gesamt</td><td>RO</td><td align="center">38511.6</td><td align="right"></td></tr></tbody></table>

andig, 2025-02-20:
@premultiply das wäre analog zur Lambda

motze92, 2025-02-20:
Folgende config funktioniert bei mir um der Pumpe den Überschuss mitzuteilen, sowie den aktuellen verbrauch auszulesen. 

```
meters:
  - name: WP
    type: custom
    power:
      source: modbus
      id: 1
      uri: "192.168.3.243:502"
      register:
        address: 4122
        type: holding
        decode: float32s
      scale: 1000

chargers:
  - name: WP
    type: custom
    icon: waterheater
    features: [integrateddevice, heating]
    status:
      source: const
      value: C
    enabled:
      source: const
      value: 1
    enable:
      source: js
      script: |
        console.log(enable);
    maxcurrent:
      source: modbus
      id: 1
      uri: "192.168.3.243:502"
      register:
        address: 74
        type: writeholding
        decode: float32s
      scale: 220
```

andig, 2025-02-20:
Das meter brauchst Du gar nicht- der custom Charger kann mittlerweile auch `power`.

motze92, 2025-02-20:
Ok danke. Hab jetzt nach dem Vorbild von Striebel folgende config gemacht:

```
- name: WP
    type: sgready
    setmode: 
      source: js
      script: |
        console.log();
    getmode:
      source: map
      values:
        0: 1 # Frostschutz
        1: 2 # Normal
        2: 2 # Forcierter Betrieb
        3: 2 # Sofortige Ansteuerung
        4: 2 # Sofortige Ansteuerung
      get:
        source: modbus
        id: 1
        uri: "192.168.3.243:502"
        register:
          address: 1090
          type: holding
          decode: uint16
    maxpower:
      source: modbus
      id: 1
      uri: "192.168.3.243:502"
      register:
        address: 74
        type: writeholding
        decode: float32s
      scale: 0.001
    power:
      source: modbus
      id: 1
      uri: "192.168.3.243:502"
      register:
        address: 4122
        type: holding
        decode: float32s
      scale: 1000
    temp:
      source: modbus
      id: 1
      uri: "192.168.3.243:502"
      register:
        address: 1014
        type: holding
        decode: float32s
    energy:
      source: modbus
      id: 1
      uri: "192.168.3.243:502"
      register:
        address: 4128
        type: holding
        decode: float32s
```

gibts irgendwo eine doku was die Werte für getmode bzw. setmode bedeuten? Hab nichts gefunden..

motze92, 2025-02-20:
0 ... Aus
1 ... Heizbetrieb
2 ... Kühlbetrieb
4 ... Warmwasser
8 ... Abtauung

[Modbus TCP_Navigator 2.0.pdf](https://github.com/user-attachments/files/18892881/Modbus.TCP_Navigator.2.0.pdf)

andig, 2025-02-20:
Ich schau grad- wie kommst du auf `Forcierter Betrieb`?

motze92, 2025-02-20:
Ah, sry ist copy/paste von hier. Also ignorieren.

https://github.com/evcc-io/evcc/blob/master/templates/definition/charger/stiebel-lwa.yaml

Das setmode wird für diese WP eigentlich nicht benötigt. Kann man das ändern, dass dies nicht required ist?

mathiasmora, 2025-02-21:
@motze92 
Ich wäre auch sehr an deiner Config interessiert wenn es läuft. 
Vielleicht kannst du den finalen Stand hier ja verteilen. Würde mich auch dazu bereit erklären es zu testen :)

andig, 2025-02-22:
Schaut Euch mal bitte an, ob https://github.com/evcc-io/evcc/pull/19008 funktionieren kann. Da wird einfach der WP die verfügbare Leistung mitgeteilt. Kann das jemand testen ob sich dieser `heatpump` Charger für die IDM eignen würde?

closed 2025-02-24 by andig

mathiasmora, 2025-02-27:
Hi @andig, 

erstmal vielen Dank für die Integration der IDM Wärmepumpe. Bin schon sehr darauf gespannt was da in Zukunft noch so alles möglich sein wird. 

Ich hatte heute endlich Zeit EVCC an meine IDM WP anzubinden. Das Auslesen von Leistung WP,  Energie und Temperatur funktioniert. Für mich persönlich würde die Temperatur für Trinkwasserspeicher unten (Register 1012) interessanter sein als der Wert für den Trinkwasserspeicher oben (Register 1014). Das ist aber vermutlich eine persönliche Präferenz. 

[image]

Der PV-Überschüss (-0,6 kW) der an der WP ankommt scheint für mich auch plausibel zu sein.

[image]

[image]

Ich werde es die kommenden Tage weiter beobachten. Sollten noch irgendwelche Tests notwendig sein gerne melden.

Viele Grüße
Mathias

mathiasmora, 2025-02-27:
PR ist erstellt #19249

dubtec, 2025-02-28:
Ich klinke mich hier mal mit ein, als Interessent mit IDM-Wärmepumpe und dem unmittelbaren Wunsch diese auch in EVCC zu steuern abhängig vom PV-Überschuß. Bitte bedenkt, dass ältere WPn mit Navigator 1.7 ausgeliefert worden (bis 2020 glaube ich) und dort sind meines Wissens nach die Modbus-Adressen im Detail andere. PV-Überschuss ist aber weiterhin auf der 74. Für eine gute Abdeckung sollten diese Versionsunterschiede (v1.7 bzw. v2.0) berücksichtigt werden.

andig, 2025-02-28:
Fixed in b9863ad49ed712884bbaf14da052772825994b4f

mathiasmora, 2025-03-02:
Hi @andig,

hab die Version v.200.5 mit den Bugfixes von oben jetzt am laufen. Die Fehlermeldung ist verschwunden und es wird ein PV-Überschuss an die WP kommuniziert. 

Leider ist der PV Überschuss der Übertragen wird ist viel zu hoch (PV hat 13 kWp): 
[image]

Ich habe dir zwei Logs angehängt. 
- Einmal nur den Log für lp-2 (Ladepunkt WP) & idm_alm (Charger IDM) 
[evcc-20250302-104242-trace.log](https://github.com/user-attachments/files/19042427/evcc-20250302-104242-trace.log)

- Einmal den kompletten Log von meiner EVCC Instanz 
[evcc-20250302-104037-trace.log](https://github.com/user-attachments/files/19042432/evcc-20250302-104037-trace.log)

Ich hab nochmal in die genaue Doku vom Modus TCP für die IDM nachgeschaut. Der Datentyp wäre hier float: 
[image]
Das sollte somit passen. 

Viele Grüße
Mathias

andig, 2025-03-02:
Hast Du mal ins Log geschaut, welcher Wert für reg 74 geschrieben wird? Stimmt der?

andig, 2025-03-02:
Du musst nur ins Log schauen ;)

motze92, 2025-03-02:
```
[lp-2 ] DEBUG 2025/03/02 12:27:50 charge total import: 39129.773kWh
[lp-2 ] DEBUG 2025/03/02 12:27:50 charger status: B
[lp-2 ] DEBUG 2025/03/02 12:27:50 pv charge current: 7.95A = 0A + 7.95A (-5483W @ 3p)
[lp-2 ] DEBUG 2025/03/02 12:27:50 site power -5483W <= 0W enable threshold
[lp-2 ] DEBUG 2025/03/02 12:27:50 pv enable in 0s
[lp-2 ] DEBUG 2025/03/02 12:27:50 pv enable timer elapsed
[circuit-main] TRACE 2025/03/02 12:27:50 validate power: -1W + (1885.8W -> 4140W) <= 4500W ok
[WP ] TRACE 2025/03/02 12:27:50 modbus: send 00 71 00 00 00 06 01 06 00 4a 00 04
[WP ] TRACE 2025/03/02 12:27:50 modbus: recv 00 71 00 00 00 06 01 06 00 4a 00 04
[lp-2 ] DEBUG 2025/03/02 12:27:50 max charge current: 6A
[WP ] TRACE 2025/03/02 12:27:50 modbus: send 00 72 00 00 00 06 01 06 00 4a 00 04
[WP ] TRACE 2025/03/02 12:27:50 modbus: recv 00 72 00 00 00 06 01 06 00 4a 00 04
[lp-2 ] DEBUG 2025/03/02 12:27:50 charger enable
[lp-2 ] DEBUG 2025/03/02 12:27:50 wake-up timer: start
[site ] DEBUG 2025/03/02 12:28:09 ----
[WP ] TRACE 2025/03/02 12:28:09 modbus: send 00 73 00 00 00 06 01 03 10 1a 00 02
[WP ] TRACE 2025/03/02 12:28:09 modbus: close connection and retry, because of EOF
[WP ] TRACE 2025/03/02 12:28:24 modbus: send 00 73 00 00 00 06 01 03 10 1a 00 02
[WP ] TRACE 2025/03/02 12:28:24 modbus: recv 00 73 00 00 00 07 01 03 04 12 c1 3f fb
[lp-2 ] DEBUG 2025/03/02 12:28:24 charge power: 1962W
```

andig, 2025-03-02:
Nochmal neues Nightly in 20min, kann sein, dass noch nicht als float32 geschrieben wird

dubtec, 2025-03-02:
Wäre es bitte möglich, dass jemand seine / ihre Konfig. postet? Ich würde gerne auch beisteuern (mit Navigator v1.7). Wenn ich wie in der Doku. beschrieben das Template nutze, komme ich nicht weit…

andig, 2025-03-03:
> komme ich nicht weit…

Das ist sehr unspezifisch.

dubtec, 2025-03-03:
> > komme ich nicht weit…
> 
> Das ist sehr unspezifisch.

Ich habe nur das template als charger in der Konfiguration angegeben. In der UI sehe ich davon aber nichts. 

[image]

Mir ist leider nicht klar - nach Lesen der Wärmepumpen-Doku. als auch dem Inhalt hier im Thread - was genau notwendig ist, um a) die Wärmepumpe in EVCC einzubetten und b) weiterführend PV-Überschuss priorisiert. Beispiele enthalten u.a. mal meters, mal nicht. 

Off topic, ich bin Sponsor, der nicht via Github gezahlt hat. Kann man das hier als Badge trotzdem deutlich machen?

andig, 2025-03-03:
> der UI sehe ich davon aber nichts.

Ladepunkt konfiguriert?!

dubtec, 2025-03-03:
Danke, ja, jetzt habe ich einen zweiten Verbraucher bzw. Ladepunkt:

[image]

Inwiefern muß / kann ich die Konfiguration weiter spezifizieren, um Temperaturen auszulesen? Ist es nicht die Idee eines Templates, dass man nur den Typ der Wärmepumpe angibt und dann der Rest wie bspw. oben angegeben automatisch übernommen wird? (https://github.com/evcc-io/evcc/issues/19002#issuecomment-2672285056)

andig, 2025-03-03:
Ich versteh die Frage nicht. Das Template hat alles was es braucht. Warum wählst du nicht einfach den Temperatursensor aus??? Leider funktioniert es nicht, wenn wir in jedem issue wieder 

    evcc configure

erklären müssen oder wie man das im UI machen kann. Dafür gibts die Doku.

andig, 2025-03-03:
Dann mach es doch bitte im UI oder mit configure. Niemand zwingt dich zur yaml datei wenn du die nicht verstehst.

dubtec, 2025-03-03:
Es ist nicht so, dass ich YAML nicht verstehe. Ich ziehe eine Textkonfiguration sogar vor. Meine Rückmeldung ist hier: Anhand der Doku alleine geht’s nicht. 
Ich möchte aber auch nicht beratungsresistent rüberkommen und werde den Weg über `evcc configure` mal probieren.

motze92, 2025-03-04:
> Nochmal neues Nightly in 20min, kann sein, dass noch nicht als float32 geschrieben wird

Ok jetzt passts.

mathiasmora, 2025-03-07:
@motze92: Stimmen bei dir auch die Werte die an der WP ankommen? 

Ich habe jetzt die v.0200.8 drauf. Aktuell habe ich um die 4 kW Überschuss. An der WP kommen im Register aber 11kW an: 

[image]

```
[lp-2 ] DEBUG 2025/03/07 09:09:13 pv charge current: 18.4A = 0A + 18.4A (-4223W @ 1p)
[idm_alm] TRACE 2025/03/07 09:09:13 modbus: send 02 ef 00 00 00 0b 01 10 00 4a 00 02 04 7a e1 40 84
[idm_alm] TRACE 2025/03/07 09:09:13 modbus: recv 02 ef 00 00 00 06 01 10 00 4a 00 02
[idm_alm] TRACE 2025/03/07 09:09:13 modbus: send 02 f0 00 00 00 0b 01 10 00 4a 00 02 04 7a e1 40 84
[idm_alm] TRACE 2025/03/07 09:09:13 modbus: recv 02 f0 00 00 00 06 01 10 00 4a 00 02
[idm_alm] TRACE 2025/03/07 09:09:43 modbus: send 02 f7 00 00 00 0b 01 10 00 4a 00 02 04 a3 d7 41 30
[idm_alm] TRACE 2025/03/07 09:09:43 modbus: recv 02 f7 00 00 00 06 01 10 00 4a 00 02
```

mathiasmora, 2025-03-07:
Was mir gerade auch noch aufgefallen ist, dass der Wert nur kurz nach dem "Start Charging" um 09:09:13 zweimal geschickt wird und dann um 09:09:43 das letzte und einzige Update kommt vom PV-Überschuss. Ist das so beabsichtigt? 

```
[lp-2 ] INFO 2025/03/07 09:09:43 start charging ->
[lp-2 ] DEBUG 2025/03/07 09:09:43 wake-up timer: stop
[idm_alm] TRACE 2025/03/07 09:09:43 modbus: send 02 f6 00 00 00 06 01 03 03 f4 00 02
[idm_alm] TRACE 2025/03/07 09:09:43 modbus: recv 02 f6 00 00 00 07 01 03 04 8e f4 42 4e
[lp-2 ] DEBUG 2025/03/07 09:09:43 pv charge current: 24.3A = 6A + 18.3A (-4211W @ 1p)
[lp-2 ] DEBUG 2025/03/07 09:09:43 pv timer reset
[lp-2 ] DEBUG 2025/03/07 09:09:43 pv timer inactive
[idm_alm] TRACE 2025/03/07 09:09:43 modbus: send 02 f7 00 00 00 0b 01 10 00 4a 00 02 04 a3 d7 41 30
[idm_alm] TRACE 2025/03/07 09:09:43 modbus: recv 02 f7 00 00 00 06 01 10 00 4a 00 02
```

Mittlerweile sind knapp 30 Minute vergangen und es wurde kein weiterer Wert geschickt: 

[image]
[image]

andig, 2025-03-07:
Irgendwie schaffe ich es nicht mir einen Reim auf die `04 7a e1 40` zu machen. Das scheint kein passendes Float32 zu sein. Könnte mir mal jemand seine WP per SSH oder VPN verfügbar machen (bitte Mail an info@evcc.io).

andig, 2025-03-07:
Falls sich nichts ändert wird auch nix geschickt.

andig, 2025-03-07:
Ok, falsch gelesen. Die letzten 4 Byte sind der Wert: `a3 d7 41 30`. Das ist 11 als IEEE475 mit vertauschter Wortreihenfolge.

Das sieht für mich richtig aus!

motze92, 2025-03-07:
Kann es sein, dass keine 0 Werte geschickt werden? Also wenn kein Überschuss zur Verfügung ist? Meine WP z.b. "denkt" gerade sie hat 4.1 kW zur Verfügung, obwohl mein Auto gerade lädt.

[lp-2 ] DEBUG 2025/03/07 11:53:59 charge power: 0W
[WP ] TRACE 2025/03/07 11:53:59 modbus: send 13 9d 00 00 00 06 01 03 10 20 00 02
[WP ] TRACE 2025/03/07 11:53:59 modbus: recv 13 9d 00 00 00 07 01 03 04 ff bd 47 19
[WP ] TRACE 2025/03/07 11:53:59 modbus: send 13 9e 00 00 00 06 01 03 10 20 00 02
[WP ] TRACE 2025/03/07 11:54:00 modbus: recv 13 9e 00 00 00 07 01 03 04 ff bd 47 19
[lp-2 ] DEBUG 2025/03/07 11:54:00 charge total import: 39423.738kWh
[lp-2 ] DEBUG 2025/03/07 11:54:00 charger status: C
[WP ] TRACE 2025/03/07 11:54:00 modbus: send 13 9f 00 00 00 06 01 03 03 f0 00 02
[WP ] TRACE 2025/03/07 11:54:00 modbus: recv 13 9f 00 00 00 07 01 03 04 3d 35 42 41
[lp-2 ] DEBUG 2025/03/07 11:54:00 pv charge current: 6.38A = 6A + 0.381A (-263W @ 3p)
[WP ] TRACE 2025/03/07 11:54:04 modbus: send 13 a0 00 00 00 06 01 03 10 1a 00 02
[WP ] TRACE 2025/03/07 11:54:04 modbus: recv 13 a0 00 00 00 07 01 03 04 00 00 00 00
[lp-2 ] DEBUG 2025/03/07 11:54:04 charge power: 0W
[WP ] TRACE 2025/03/07 11:54:24 modbus: send 13 a1 00 00 00 06 01 03 10 1a 00 02
[WP ] TRACE 2025/03/07 11:54:24 modbus: close connection and retry, because of EOF
[WP ] TRACE 2025/03/07 11:54:39 modbus: send 13 a1 00 00 00 06 01 03 10 1a 00 02
[WP ] TRACE 2025/03/07 11:54:39 modbus: recv 13 a1 00 00 00 07 01 03 04 00 00 00 00
[lp-2 ] DEBUG 2025/03/07 11:54:39 charge power: 0W
[WP ] TRACE 2025/03/07 11:54:39 modbus: send 13 a2 00 00 00 06 01 03 10 20 00 02
[WP ] TRACE 2025/03/07 11:54:39 modbus: recv 13 a2 00 00 00 07 01 03 04 ff bd 47 19
[WP ] TRACE 2025/03/07 11:54:39 modbus: send 13 a3 00 00 00 06 01 03 10 20 00 02
[WP ] TRACE 2025/03/07 11:54:39 modbus: recv 13 a3 00 00 00 07 01 03 04 ff bd 47 19
[lp-2 ] DEBUG 2025/03/07 11:54:39 charge total import: 39423.738kWh
[lp-2 ] DEBUG 2025/03/07 11:54:39 charger status: C
[WP ] TRACE 2025/03/07 11:54:39 modbus: send 13 a4 00 00 00 06 01 03 03 f0 00 02
[WP ] TRACE 2025/03/07 11:54:39 modbus: recv 13 a4 00 00 00 07 01 03 04 00 4f 42 41
[lp-2 ] DEBUG 2025/03/07 11:54:39 pv charge current: 6.31A = 6A + 0.31A (-214W @ 3p)
[WP ] TRACE 2025/03/07 11:54:44 modbus: send 13 a5 00 00 00 06 01 03 10 1a 00 02
[WP ] TRACE 2025/03/07 11:54:44 modbus: recv 13 a5 00 00 00 07 01 03 04 00 00 00 00
[lp-2 ] DEBUG 2025/03/07 11:54:44 charge power: 0W
[WP ] TRACE 2025/03/07 11:55:04 modbus: send 13 a6 00 00 00 06 01 03 10 1a 00 02
[WP ] TRACE 2025/03/07 11:55:04 modbus: close connection and retry, because of EOF
[WP ] TRACE 2025/03/07 11:55:19 modbus: send 13 a6 00 00 00 06 01 03 10 1a 00 02
[WP ] TRACE 2025/03/07 11:55:19 modbus: recv 13 a6 00 00 00 07 01 03 04 00 00 00 00
[lp-2 ] DEBUG 2025/03/07 11:55:19 charge power: 0W
[WP ] TRACE 2025/03/07 11:55:19 modbus: send 13 a7 00 00 00 06 01 03 10 20 00 02
[WP ] TRACE 2025/03/07 11:55:19 modbus: recv 13 a7 00 00 00 07 01 03 04 ff bd 47 19
[WP ] TRACE 2025/03/07 11:55:19 modbus: send 13 a8 00 00 00 06 01 03 10 20 00 02
[WP ] TRACE 2025/03/07 11:55:19 modbus: recv 13 a8 00 00 00 07 01 03 04 ff bd 47 19
[lp-2 ] DEBUG 2025/03/07 11:55:19 charge total import: 39423.738kWh
[lp-2 ] DEBUG 2025/03/07 11:55:19 charger status: C
[WP ] TRACE 2025/03/07 11:55:19 modbus: send 13 a9 00 00 00 06 01 03 03 f0 00 02
[WP ] TRACE 2025/03/07 11:55:19 modbus: recv 13 a9 00 00 00 07 01 03 04 c0 7d 42 40
[lp-2 ] DEBUG 2025/03/07 11:55:19 pv charge current: 6.39A = 6A + 0.387A (-267W @ 3p)

[image]

mathiasmora, 2025-03-11:
@motze92: Ich vermute das Problem ist hier in der Berechnung

`[lp-2 ] DEBUG 2025/03/07 11:55:19 pv charge current: 6.39A = 6A + 0.387A (-267W @ 3p)`

Zu einem früheren Zeitpunkt wurde der WP ein Wert von 6A als Überschuss geschickt. EVCC nimmt jetzt an, dass die WP weiterhin mit diesen 6A schon aktiv ist. Die reale Leistung der WP ist aber 0W

`[lp-2 ] DEBUG 2025/03/07 11:55:19 charge power: 0W`

Aus dem Grund wird als pv charge current für die WP weiterhin die 6A geschickt da laut Berechnung sich nicht ändert und weiterhin die 6A zu Verfügung stehen. 

Vermutlich müsste es eigentlich es eigentlich so aussehen: 
`[lp-2 ] DEBUG 2025/03/07 11:55:19 pv charge current: 0.39A = 0A + 0.387A (-267W @ 3p)`
-> Es werden 0A geschickt. 

Ich hab das ganze mal in einem neuen Ticket zusammengefasst. #19669

SolarPowerEV, 2025-03-13:
Weiß jemand, was der Unterschied zwischen Register 4122 und 4124 ist? 
In der Register-Beschreibung oben im allerersten Post steht für beide derselbe Text drin:

4122 | elektrische Gesamtleistung (falls kein Wert vom Inverter vorhanden ist, wird jener vom Polynom verwendet)  
4124 | elektrische Gesamtleistung (falls kein Wert vom Inverter vorhanden ist, wird jener vom Polynom verwendet) 
4126 | thermische Leistung (basierend auf Durchflusssensor)
4128 | Wärmemenge gesamt

Aktuell wird im Template für energy noch Register 4128 (Wärmemenge gesamt) verwendet.
Dort müsste aber eigentlich die elektrische Gesamtleistung angezeigt werden.
Nur welches Register ist jetzt das passende? 4122 oder 4124? Da ich keine IDM besitze, kann ich das leider nicht austesten.

mathiasmora, 2025-03-15:
@SolarPowerEV: 
Aktuell gibt es keine Register für die elektrische Gesamtmenge in kwh für die IDM. Aus dem Grund wurde die Wärmemenge gesamt verwendet. 
4122 und 4124 sind jeweils Werte in kW und beziehen sich auf die Leistung und nicht die Energy. Können hier also nicht verwendet werden da falsche Einheit ;)

SolarPowerEV, 2025-03-15:
> [@SolarPowerEV](https://github.com/SolarPowerEV): Aktuell gibt es keine Register für die elektrische Gesamtmenge in kwh für die IDM. Aus dem Grund wurde die Wärmemenge gesamt verwendet. 4122 und [...]

Ok. Verstehe. Wenn die elektrische Gesamtleistung nur in kW berechnet wird, würde die Nutzung in der Anzeige mit kWh in der Tat die User nur in die Irre führen. Das tut die Wärmeleistung an der Stelle natürlich auch ein bisschen. Ich würde mich jedenfalls ordentlich erschrecken, wenn ich da mal draufschaue und dann denke, dass das der Stromverbrauch meiner WP war (ohne jeden Hinweis, dass ich den Wert eigentlich noch durch die Jahresarbeitszahl teilen müsste).

Seltsam finde ich allerdings, dass IDM zwei identische Register für scheinbar dasselbe hat, dann aber keinen "Stromzähler", der einfach nur stumpf die verbrauchten kWh hochzählt.

Mindestens intern muss die IDM doch aber auch die verbrauchten elektrischen kWh irgendwo hochzählen, um daraus dann die Jahresarbeitszahl berechnen und den Nutzer in einer Gesamtstatistik anzeigen zu können. Angesichts der gefühlt 10x größeren Anzahl an Registern im Vergleich zu Lambda finde ich es unerwartet, dass so ein fundamentaler Wert bei der IDM nicht über ein Register zur Verfügung gestellt wird (in der Lambda mit nur sehr wenigen Registern aber schon).

Aber gut, ich habe es versucht, hier noch das Template für die IDM konsistent zu machen. Da ich die Wärmepumpe nicht besitze und mit der Lambda aktuell auch noch ein paar Problemchen habe, schließe ich das Thema für mich hiermit ab.

Ich danke Dir für die Info und die Klarstellung, dass eine Änderung des vorhandenen IDM-Templates die Sache nur verschlimmbessern würde.

AndyNew2, 2025-03-17:
Hallo ich habe auch eine iDM Wärmepumpe iPump A3-11

Bezüglich der Modbus Werte:
4122 | elektrische Gesamtleistung (falls kein Wert vom Inverter vorhanden ist, wird jener vom Polynom verwendet)  
4124 | elektrische Gesamtleistung (falls kein Wert vom Inverter vorhanden ist, wird jener vom Polynom verwendet)
4126 | thermische Leistung (basierend auf Durchflusssensor)
4128 | Wärmemenge gesamt

iDM hat hier einiges hinzugefügt aber nicht wirklich dokumentiert. 
Der Wert bei 4122 ist dokumentiert, bei 4124 nicht, zeigt aber bei mir den gleichen Wert an.

AndyNew2, 2025-03-17:
Dann gibt es ja noch diese tollen Register:
4108 | Leistungsbegrenzung Wärmepumpe | RW |   | -1
4112 | Leistungsbegrenzung Kaskade | RW |   | -1

Hatte mich echt gefreut, als ich sie entdeckt habe. Nur leider bewirken sie bei mir gar nix. Wenn ich sie als float oder auch als int schreibe, wird in der iDM als geschriebener Wert immer 0 angezeigt (obwohl ich andere Werte geschrieben habe). Eine Aussage von iDM bezüglich der Werte war: Nicht für User gedacht.

AndyNew2, 2025-03-17:
Zur PV Leistung und Überschuss: Die Doku beschreibt nur die Register
74 | Überschuss
78 | PV Leistung 

Einige User hatten berichtet, dass die iDM Steuerung versuchen würde den Überschuss auf 0 zu regeln. Das kann ich für mich bei der iPump nicht bestätigen. Einzig und allein ein Komfort und ein unsinniger WW Boost kann eingestellt werden. Bei mir hat dies (und auch bei einem Nachbar) zu übermäßigen takten geführt, weil nunmal die PV Leistung und damit der Überschuss nicht konstant ist sondern auch an sonnigen Tagen dank ein paar Wolken zappelt.

Die iDM Doku sagt rein gar nichts aus über diese Werte.  Nur dass es sie gibt und dass sie als float zu schreiben sind.

Dann gibt es noch undokumentiert diese Register:
82 | Hausverbrauch | RW | 0 |  
84 | Batterieentladung | RW | 0 |  
86 | Batterie Füllstand | RW | -1

Die können auch als float geschrieben werden. Ausnahme ist der Batterie Füllstand, das ist ein int (Auflösung; LSB = 1%).
Wenn man diese Werte auch schreibt, wechselt die iDM Anzeige in ein detaillierteres PV Blockbild und macht eine genauere Auswertung der Energiewerte. Die iDM integriert intern die Leistungswerte und berechnet daraus Energiewerte.

Für den Komfort-Modus und den unsinnigen WW-Boost, steht dann eine Minimal SOC Einstellung für die Batterie bereit.

 Bis jetzt ist es mir leider noch nie gelungen, die iDM von außen direkt in der Leistungsabnahme zu regeln, um damit die PV-Leistung möglichst optimal ausnutzen zu können. Falls hierzu jemand andere Erfahrungen hat, bitte posten....

dschrempf, 2026-11-07:
Hat sich bei euch in der Zwischenzeit etwas getan? Ich versuche auch mittels Register 74 meine AERO zu regeln, aber ich schaffe nur "full speed" oder off nach 10min (minimale Laufzeit). Ich schreibe das Register manuell und hab schon ziemlich viel rum-probiert. Habt ihr Erfahrungen?

TRON4R, 2026-11-08:
Ich habe vor kurzem eine neue YAML-Definition für die Lambda Wärmepumpe erstellt und [in dieser Solution gepostet](https://github.com/evcc-io/evcc/discussions/35894). 
Diese YAML-Konfig meldet - anders als der Default-Zustand von evcc - wie von Lambda im Modus "Pos. E-Überschuss" vorgesehen, einfach nur im Minutentakt (konfigurierbar) den laufenden PV-Überschuss an das entsprechende ModBus-Register der Lambda. Das hat bei mir bisher super funktioniert. 

Wenn es Probleme mit den per Default von evcc gemeldeten Werten gibt (was mich nicht überraschen würde, denn dazu gab es ja schon eine lange Diskussion [in diesem Lambda-Thread](https://github.com/evcc-io/evcc/discussions/20179)), dann probier doch mal aus, meine YAML-Definition an die Register der IDM anzupassen. Nach meinem Verständnis aus den Diskussionen von damals verhalten sich IDM und Lambda vom Prinzip her absolut gleich was die Meldung von PV-Überschuss angeht, daher sollte die Logik eigentlich so 1:1 übernommen werden können.

dschrempf, 2026-11-08:
Danke, werde mich reinlesen. Ich habe jetzt auch noch geschafft, die Pumpe zu modulieren, indem ich das Register 4108 _vor_ them PV run auf den Cap (z.B. 1.5kW) setze. Aber eine ordentliche Steuerung wäre natürlich besser. Mich wundert es warum man nicht einfach die Power (oder den Modulationsfaktor) selbst setzen kann (natürlich im Rahmen des Möglichen).

dschrempf, 2026-11-08:
Danke! Meine Messungen an der AERO bisher:

- Die iDM rechnet ihren eigenen Verbrauch selbst zu Register 74 dazu. Die Standard-Vorlage von `evcc` zählt ihn damit doppelt (ich glaube, das ist das Problem von #19677). Den reinen Überschuss zu melden, wie in deiner YAML, passt also.
- Nur ein negativer Wert beendet den Lauf zuverlässig. Bei 0 abschneiden würde ich bei der iDM daher nicht.
- 4108 deckelt die Leistung, wenn man es vor dem Start setzt. Das benutze ich im Moment sehr zuverlässig.

andig, 2026-11-08:
Soweit ich mich erinnere war das Problem vor Jahren gelöst…

AndyNew2, 2026-11-08:
Ich habe eine iPump von iDM und keine der hier beschriebenen Register funktioniert.
Ich verstehe auch das Konzept hier nicht. iDM hat ca. 10 - 20 verschiedene Modelle von Wärmepumpen, die alle über eine ähnliche ModbusTCP Schnittstelle verfügen. Die werden aber je nach Modell verschieden interpretiert. Der Überschuss wird nicht einfach in Kompressorleistung umgewandelt, wenn über Register 74 geschrieben, sondern ist nur ein Hinweis auf Überschuss, der über anderweitige Konfiguration in der Wärmepumpe selbst verarbeitet wird. Dies ist mal besser mal schlechter.
Das tolle Register 4108 kann ich z.B. überhaupt nicht schreiben und hat natürlich auch überhaupt keine Wirkung.
Wie soll das vor Jahren gelöst gewesen sein?
Mir ist überhaupt nicht klar wie die Steuerung hier funktionieren soll, und iDM selbst sieht so eine Steuerung von außen leider überhaupt nicht vor. Lambda anscheindend schon, aber iDM nicht. Die Doku ist mehr als dünn, und die ModbusTCP Register sind mit einem Wort beschrieben. Wie hier irgendwas mit der Leistungssteuerung zusammenhängen soll oder nicht, ist von iDM nirgends beschrieben, es gibt Userberichte. Bei dem einen Modell macht es dies bei einem anderes was ganz anderes.... 
Das ganze ist hier auf sehr dünnen Eis....
Also wie gesagt, bei der iPump geht nur das Register 74 als Hinweis für Überschuss, der über weitere Optionen genutzt werden kann oder nicht. Dabei wird der Kompressor nicht an den Überschuss heranmoduliert, sondern es ist nur eine Frage der Zieltemperaturen....

dschrempf, 2026-11-09:
Ja, ich habe dieselbe Erfahrung gemacht. Mit Register 74 alleine die Pumpe zu steuern ist fast unmöglich. Aber 4108 funktioniert nun ganz gut. Einziger Nachteil, ich kann die Leistung nicht wirklich während des Laufs anpassen. Ich muss mich quasi davor schon entscheiden, wie viel kW der Kompressor ziehen soll. Sonst funktioniert die Pumpe super, aber dieser Aspekt ist sehr störend.

andig, 2026-11-09:
> Ich verstehe auch das Konzept hier nicht. iDM hat ca. 10 - 20 verschiedene Modelle von Wärmepumpen, die alle über eine ähnliche ModbusTCP Schnittstelle verfügen. Die werden aber je nach Modell verschieden interpretiert

Dann wäre es doch ganz wunderbar, wenn der Hersteller mal für Klarheit sorgen würde.

AndyNew2, 2026-11-09:
Da stimme ich zu. Ist aber leider eher Standard der vielen Wärmepumpen Hersteller. Wir sind bei iDM ja schon sehr glücklich eine offizielle ModbusTCP Anbindung zu haben. Bei vielen anderen Herstellern sucht man sich die Werte über WebGUIs etc. zusammen. Da gibt es oft offiziell gar nichts.
Wie gesagt die "Beschreibung", dieser ModbusTCP Anbindung ist ein Wort pro Register. Wir versuchen seit geraumer Zeit hier für mehr Klarheit zu sorgen, ich fürchte aber, dass hier iDM selbst den Überblick verloren hat...

dschrempf, 2026-11-09:
Bei Interesse, schau mal auf meinem Repo vorbei: https://github.com/dschrempf/idm-modbus. Ich habe da mit Hilfe von LLMs ziemlich viel akribisch gescannt und getested, auch mit WebSocket auf die Navigator 10 Oberfläche (ist aber AERO spezifisch). Ich habe auch zwei PDFs mit der Modbus TCP Konfiguration (Stand 2025).