evcc-io/evcc #19006: Unable to connect to local devices on macOS

rjpease, 2025-02-20:
### Describe the bug

EVCC unable to connect to local devices after macOS Sequoia upgrade.  While I understand that macOS is blocking the access EVCC it not requesting the access needed to connect to local devices.  

### Steps to reproduce

1. Configured evcc in homebrew
started evcc
not communicating with charger or meter
2. Start evcc service in homebrow
3. Error Occurs - I get "connect: no route to host" error messages, see log details below.

### Configuration details

```yaml
[main  ] INFO 2025/02/20 09:03:51 evcc 0.200.1
[main  ] INFO 2025/02/20 09:03:51 using config file: /Users/rjpease/evcc.yaml
[db    ] INFO 2025/02/20 09:03:51 using sqlite database: /Users/rjpease/.evcc/evcc.db
[circuit-main] DEBUG 2025/02/20 09:03:52 validation of max power disabled

Konfiguration (/Users/rjpease/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 30s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: enphase 
  usage: grid  
  host: 192.168.7.208  
  token: *****
  cache: 1s  
  name: grid1
- type: template
  template: enphase 
  usage: pv  
  host: 192.168.7.208  
  token: *****
  cache: 1s  
  name: pv2

chargers:
- type: template
  template: twc3 
  host: 192.168.7.187  
  name: wallbox4

loadpoints:
- title: Garage
  charger: wallbox4
  mode: off

site:
  title: Pease OC Home
  meters:
    grid: grid1
    pv:
    - pv2

Version: `0.200.1`
```

### Log details

```text
[site ] INFO 2025/02/20 09:54:18 site config:
[site ] INFO 2025/02/20 09:54:18 meters: grid ✓ pv ✓ battery ✗
[site ] INFO 2025/02/20 09:54:18 grid: power ✓ energy ✗ currents ✓
[site ] INFO 2025/02/20 09:54:18 pv 1: power ✓ energy ✓ currents ✓
[site ] INFO 2025/02/20 09:54:18 vehicles:
[site ] INFO 2025/02/20 09:54:18 vehicle 1: range ✓ finish ✗ status ✓ climate ✓ wakeup ✓
[lp-1 ] INFO 2025/02/20 09:54:18 loadpoint 1:
[lp-1 ] INFO 2025/02/20 09:54:18 mode: off
[lp-1 ] INFO 2025/02/20 09:54:18 charger: power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1 ] INFO 2025/02/20 09:54:18 meters: charge ✓
[lp-1 ] INFO 2025/02/20 09:54:18 charge: power ✓ energy ✗ currents ✓
[lp-1 ] DEBUG 2025/02/20 09:54:18 phase timer inactive
[lp-1 ] DEBUG 2025/02/20 09:54:18 pv timer inactive
[lp-1 ] ERROR 2025/02/20 09:54:18 charger enabled: Get "http://192.168.7.187/api/1/vitals": dial tcp 192.168.7.187:80: connect: no route to host
[site ] DEBUG 2025/02/20 09:54:18 ----
[lp-1 ] ERROR 2025/02/20 09:54:18 charge power: Get "http://192.168.7.187/api/1/vitals": dial tcp 192.168.7.187:80: connect: no route to host
[lp-1 ] ERROR 2025/02/20 09:54:19 charge currents: Get "http://192.168.7.187/api/1/vitals": dial tcp 192.168.7.187:80: connect: no route to host
[circuit-main] DEBUG 2025/02/20 09:54:19 power: 0W
[circuit-main] DEBUG 2025/02/20 09:54:19 current: 0A
[site ] ERROR 2025/02/20 09:54:19 pv 1 power: Get "http://192.168.7.208/production.json": dial tcp 192.168.7.208:80: connect: no route to host
[site ] ERROR 2025/02/20 09:54:20 grid power: Get "http://192.168.7.208/production.json": dial tcp 192.168.7.208:80: connect: no route to host
[site ] DEBUG 2025/02/20 09:54:50 ----
[lp-1 ] ERROR 2025/02/20 09:54:50 charge power: Get "http://192.168.7.187/api/1/vitals": dial tcp 192.168.7.187:80: connect: no route to host
[lp-1 ] ERROR 2025/02/20 09:54:51 charge currents: Get "http://192.168.7.187/api/1/vitals": dial tcp 192.168.7.187:80: connect: no route to host
[circuit-main] DEBUG 2025/02/20 09:54:51 power: 0W
[circuit-main] DEBUG 2025/02/20 09:54:51 current: 0A
[site ] ERROR 2025/02/20 09:54:52 pv 1 power: Get "http://192.168.7.208/production.json": dial tcp 192.168.7.208:80: connect: no route to host
[site ] ERROR 2025/02/20 09:54:52 grid power: Get "http://192.168.7.208/production.json": dial tcp 192.168.7.208:80: connect: no route to host
[site ] DEBUG 2025/02/20 09:55:20 ----
[lp-1 ] ERROR 2025/02/20 09:55:20 charge power: Get "http://192.168.7.187/api/1/vitals": dial tcp 192.168.7.187:80: connect: no route to host
[lp-1 ] ERROR 2025/02/20 09:55:21 charge currents: Get "http://192.168.7.187/api/1/vitals": dial tcp 192.168.7.187:80: connect: no route to host
[circuit-main] DEBUG 2025/02/20 09:55:21 power: 0W
[circuit-main] DEBUG 2025/02/20 09:55:21 current: 0A
[site ] ERROR 2025/02/20 09:55:21 pv 1 power: Get "http://192.168.7.208/production.json": dial tcp 192.168.7.208:80: connect: no route to host
[site ] ERROR 2025/02/20 09:55:22 grid power: Get "http://192.168.7.208/production.json": dial tcp 192.168.7.208:80: connect: no route to host
```

### What type of operating system or environment does evcc run on?

macOS

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.1

andig, 2025-02-20:
This sounds like https://github.com/golang/go/issues/68678 but we had resolved this already. As of Go 1.24 the linker should create the required UUID. I have absolutely no idea what might be causing this.

michaellocher, 2025-04-04:
Hey guys, i've managed it to get the permissions using: `codesign -s - --force --deep /usr/local/Cellar/evcc/0.202.1/bin/evcc` and then started the service…

andig, 2025-04-04:
I have no idea- this seems to be a different issue that the missing UUIDs from the linker.