evcc-io/evcc #19018: Fehler in bei Konfiguration des Fronius Symo GEN24 Plus

diestrohs, 2025-02-20:
### Describe the bug

Beim einbinden des Fronius **Symo** GEN24 Plus über die Konfiguations-GUI wird immer Fronius **Primo** GEN24 angezeigt (rot markiert)

[image]
Des Weiteren wird dafür keine Energie angezeigt (rot markiert). Wird dieser als SunSpec Hybridwechselrichter eingebunden, wird die Energie angezeigt (gelb markiert)
[image]
  

 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: --------------------------------------------

interval: 30s # control cycle interval

chargers:
- type: template
  template: ellipro 
  ski: xxxx xxxx xxxx xxxx xxxx xxxx xxxx xxxx xxxx xxxx  
  ip: 192.168.178.9  
  name: wallbox6

loadpoints:
- title: Garage
  charger: wallbox6
  mode: pv

eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----

          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
     
          -----END CERTIFICATE-----
```

### Log details

```text
[eebus ] DEBUG 2025/02/21 00:37:08 Send: d:_i:47859_Elli-Wallbox-2116A1B5IV notify 313167 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2025/02/21 00:37:10 Send: d:_i:47859_Elli-Wallbox-2116A1B5IV notify 313168 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2025/02/21 00:37:12 Send: d:_i:47859_Elli-Wallbox-2116A1B5IV notify 313169 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2025/02/21 00:37:14 Send: d:_i:47859_Elli-Wallbox-2116A1B5IV notify 313170 DeviceDiagnosisHeartbeatData
[site ] DEBUG 2025/02/21 00:37:15 ----
[lp-1 ] DEBUG 2025/02/21 00:37:15 charge power: 30W
[lp-1 ] DEBUG 2025/02/21 00:37:15 charge currents: [0.05 0.08 0]A
[site ] DEBUG 2025/02/21 00:37:15 grid power: 1566W
[site ] DEBUG 2025/02/21 00:37:15 pv 2 power: 0W
[site ] DEBUG 2025/02/21 00:37:15 grid powers: [300 1117 149]W
[site ] DEBUG 2025/02/21 00:37:15 battery 1 power: 0W
[site ] DEBUG 2025/02/21 00:37:15 grid currents: [2.1 5.46 1.35]A
[site ] DEBUG 2025/02/21 00:37:15 battery 1 soc: 4%
[site ] DEBUG 2025/02/21 00:37:16 pv 1 power: 0W
[eebus ] DEBUG 2025/02/21 00:37:16 Send: d:_i:47859_Elli-Wallbox-2116A1B5IV notify 313171 DeviceDiagnosisHeartbeatData
[site ] DEBUG 2025/02/21 00:37:16 pv power: 0W
[site ] DEBUG 2025/02/21 00:37:16 site power: 1666W
[lp-1 ] DEBUG 2025/02/21 00:37:16 charger status: B
[lp-1 ] DEBUG 2025/02/21 00:37:16 plan: charge 54m48s between 2025-02-21 10:35:12 +0100 CET until 2025-02-21 11:30:00 +0100 CET (power: 7360W, avg cost: 0.294)
[lp-1 ] DEBUG 2025/02/21 00:37:16 pv charge current: 0A = 0A + -3.62A (1666W @ 2p)
[eebus ] DEBUG 2025/02/21 00:37:18 Send: d:_i:47859_Elli-Wallbox-2116A1B5IV notify 313172 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2025/02/21 00:37:20 Send: d:_i:47859_Elli-Wallbox-2116A1B5IV notify 313173 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2025/02/21 00:37:20 mdns: remove - ski: 340fa31b02c43842abd6aff29a549c0f5c744ca9 name: Elli-Wallbox-2116A1B5IV brand: Elli model: Wallbox typ: Wallbox identifier: Elli-Wallbox-2116A1B5IV register: false host: wallbox-2116A1B5IV.local. port: 4712 addresses: []
[eebus ] DEBUG 2025/02/21 00:37:22 Send: d:_i:47859_Elli-Wallbox-2116A1B5IV notify 313174 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2025/02/21 00:37:24 Send: d:_i:47859_Elli-Wallbox-2116A1B5IV notify 313175 DeviceDiagnosisHeartbeatData
[site ] DEBUG 2025/02/21 00:37:25 ----
[lp-1 ] DEBUG 2025/02/21 00:37:25 charge power: 30W
[lp-1 ] DEBUG 2025/02/21 00:37:25 charge currents: [0.05 0.08 0]A
[site ] DEBUG 2025/02/21 00:37:25 grid power: 1559W
[site ] DEBUG 2025/02/21 00:37:25 pv 2 power: 0W
[site ] DEBUG 2025/02/21 00:37:25 grid powers: [300 1117 142]W
[site ] DEBUG 2025/02/21 00:37:25 battery 1 power: 0W
[site ] DEBUG 2025/02/21 00:37:25 grid currents: [2.1 5.47 1.33]A
[site ] DEBUG 2025/02/21 00:37:25 battery 1 soc: 4%
[site ] DEBUG 2025/02/21 00:37:26 pv 1 power: 0W
[eebus ] DEBUG 2025/02/21 00:37:26 Send: d:_i:47859_Elli-Wallb
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-02-21:
@naltatis der erste Teil- anderes Gerät angezeigt ist ja erwartet. Wie wäre es, wenn wir `type` nicht nur den Typ sondern `typ#ausgewähler Gerätename` speichern würden? Oder: wir erfinden ein neues `product` Feld.

@diestrohs die fehlende Energie ist vmtl. im Template einfach nicht umgesetzt. Gerne PR dafür.

naltatis, 2025-02-21:
Ja, stand heute ist das erwartungskonform, wenn auch noch nicht optimal.

@andig Ja, ich würd vorschlagen, dass wir `brand` und `description` des ausgewählten Products beim Anlegen mit absichern. Das `product` selbst hat heute keinen Identifier und über den Index würde ich nicht gehen wollen. Sollte sich, aus welchen Gründen auch immer Brand und Description in einem Template mal ändern würden wir (wie aktuell) wieder auf den ersten Eintrag zurückfallen. Das wäre aber mMn. durchaus verschmerzbar.

Einzige andere Alternative, die ich sehe wäre ein manuell gepflegtes Identifier Feld an jedem Product im Template inkl. Migrationslogik (analog `covers`) für Veränderungen. Die extra manuelle Pflege wäre mir das Feature heute allerdings nicht wert.

closed 2025-03-11 by andig