evcc-io/evcc #19019: Heatpump: Luxtronik2

jeriox, 2025-02-21:
**Describe the solution you'd like**
After reading the blogpost about the new heatpump support, I'd like to integrate my heatpump which is using a Luxtronik2 controller found in many devices of german brands. The controller offers a REST API, for which a lot of clients already exist (e.g. https://github.com/BenPru/luxtronik). However, it is not possible to set a SG Ready level via the API. The closest match (which is also what the manual specifies is happening when using the SG Ready contacts) is setting an offset in °C for the heating temperature (Heiz Ziel Korrektur), e.g. by increasing the value calculated by the heating curve by 5°C in boost mode. 

**Describe alternatives you've considered**
- The controller also supports different heating modes (Off/Automatic/Party), which can be set via the API and could be mapped to an appropriate SG Ready level
- The heatpump also supports a physical SG Ready contact via relais, however this would require buying a Shelly and getting the wiring inside the device right

**Additional context**
Behaviour of the controller via the physical contacts that could be replicated

[image]

michiproep, 2025-02-21:
Ich antworte mal auf Deutsch, weil das Thema richtig kompliziert ist. Ich weiß gar nicht wo ich anfangen soll.
Zunächst ist es ja so, dass die WP nicht nur heizt, sondern auch Warmwasserbereitung macht. Diese beiden Modi sind zu trennen, weil beim Heizen - und ich bezieh mich jetzt immer auf meine WP - je nach aktueller Vorlauftemperatur zwischen 2,4kw und 2,9kw gebraucht werden. Beim Warmwasser sind es 3kw bis 4,5kw.

Bei Luxtronik2 steuert der PV-Ready-Kontakt (nicht der SG-Ready, ja die haben 2 unterschiedliche) ausschließlich die Warmwasserbereitung. Ist der Kontakt an, setzt die WP 75° als WW-Soll, die aber nie erreicht werden.
Mit PV-Ready kannst du also ca. 30 Minuten lang 4kw in die WP leiten.

Heizen kann man, wie du schon gesagt hast, via heating_temperature, was den Offset deiner Standard-Keizkurve darstellt, erzwingen.
Damit bekomm ich es auch jetzt im Februar (gelegentlich) hin, dass falls das Auto nicht zu Hause ist und der Akku wegen der niedrigen Temperaturen nicht alles aufnehmen kann, ich den ganzen Tag sehr hoch heizen kann um damit fast die ganze Nacht zu überbrücken.

Ich war mir nie wirklich schlüssig, ob ich in evcc dafür zwei Loadpoints verwenden soll oder nicht. Hab mich dann aber für Einen entschieden, was natürlich andere Nachteile mit sich bringt.

Ich hab jetzt also einen Loadpoint /IntegratedDevice, das via mqtt den von evcc vorgeschlagenen Zustand an ein Skript übermittelt.
Der Zielzustand der WP errechnet sich nun eigentlich aus enabled und maxcurrent
Das eigentliche Problem dabei ist, dass ich mich nicht wirklich auf maxcurrent verlassen konnte, weil einerseits die WP natürlich viel zu träge reagiert und andererseits evcc häufig (und manchmal auch nur kurzzeitig) den maximalen maxcurrent meldet (und umgekehrt), obwohl keine 4kw zur Verfügung standen. Zusätzlich will man ja auch noch häufiges An/Ausschalten verhindern.

Das war jetzt erstmal ein kleiner Ausflug in meine Wärmepupenwelt. Da gibt's noch viel mehr zu beachten.

evcc macht das schon richtig gut, aber ich denke, dass es Ende nur richtig gut mit Wärmepumpen funktionieren kann, denen man einfach den aktuellen Überschuss mitteilen kann und die - ähnlich eines EAutos - die zur Verügung stehende Leistung selbstständig sinnvoll verwenden.

andig, 2025-02-21:
> Behaviour of the controller via the physical contacts that could be replicated

Es braucht eine vollständige Modbus Doku die zeigt wie sich die Modi setzen lassen.

michiproep, 2025-02-21:
@andig : Luxtronik unterstützt kein Modbus

andig, 2025-02-21:
Ne api Doku brauchts trotzdem 😬

michiproep, 2025-02-21:
@andig Da hast du natürlich Recht. Das beste was du findest ist wahrscheinlich
https://github.com/coolchip/luxtronik2 und https://github.com/UncleSamSwiss/ioBroker.luxtronik2

Da mein Skript im iobroker über die Zeit richtig kompliziert wurde und "1000" Situationen abdeckt, würde ich, falls ihr wirklich vor habt, Luxtronik nativ in evcc zu integrieren, Folgendes vorschlagen:
-Im Template müssen die beiden Leistungsbereiche für Heizung und WW hinterlegt werden, die jeweiligen Temperaturerhöhungen, ZB Heizung +4K, WW +15K und die zeitlichen Thresholds (siehe unten)
-evcc muss dann auf Basis von maxCurrent, also Zb bei >2kw die Heizung auf +2 stellen und bei >4kw zusätzlich das WW auf +15

Was aber wirklich wichtig ist, dass der jeweilige "Ausschaltvorgang" jetzt nicht ausschließlich auf Basis des Loadpoint Thresholds basisiert, sondern auch die Absenkung des maxcurrent von evcc aus einem Threshold unterliegt.
Also, falls WW auf +15 gestellt wird, kann man nicht bei der nächsten Wolke wieder auf 4A runterregeln. Einmal an sollte dann auch zumindest für 15 Min "an" bedeuten.

andig, 2025-02-22:
Das ist sehr spezifische Logik- die werden wir nicht umsetzen. Was geht ist entweder:
- der WP den aktuellen Überschuss mitteilen (falls sie damit umgehen kann)
- oder einen Betriebsmodus zu setzen (z.B: forciert statt eco oder normal)

Wenn es in dieses Raster nicht rein past dann bleibt noch eine "custom" Implementierung mit externer Logik.

jeriox, 2025-02-23:
>     * der WP den aktuellen Überschuss mitteilen (falls sie damit umgehen kann)

das scheidet für die Steuerung hier auf jeden Fall aus (nicht unterstützt).

>     * oder einen Betriebsmodus zu setzen (z.B: forciert statt eco oder normal)

ginge da tatsächlich nur ein fester modus oder kann es auch etwas komplexer sein? Also für meinen Usecase würde vollkommen ausreichen den Heiz-Offset in °C zu setzen, also von der Logik her "setzte diesen Wert auf 0 im Normalmodus, setzte diesen Wert auf +3 im Boost-Modus". Das sollte ja im Code vermutlich nicht viel anders aussehen als einen Modus zu setzen?

andig, 2025-02-23:
Ja, das können wir gerne machen- nur keine komplexe Logik mit erst das, dann das. Wenn man den Puffer (Speicher vs. WW) auswählen kann wären das schon 2 Möglichkeiten, ggf kann man das dann auch als 2 Charger konfigurieren. Willst Du es mal probieren? Siehe Stiebel bzgl. Umsetzung.

andig, 2025-02-23:
Die "erhöhte Betriebsweise" lässt sich ja vllt. auch direkt ansteuern?

michiproep, 2025-02-24:
> Das ist sehr spezifische Logik- die werden wir nicht umsetzen. 
 Hi  @andig , das versteh ich vollkommen.
Was das Leben aber wesentlich erleichtern würde ist, wenn man das Ganze auf zwei Loadpoints aufteilen könnte. Dazu müsste evcc allerdings sowas wie "mutual exclusive loadpoints" unterstützen, weil man mit einem Gerät in diesem Fall entweder WW macht oder heizt, nicht aber beides gleichzeitig.

andig, 2025-02-24:
> ginge da tatsächlich nur ein fester modus oder kann es auch etwas komplexer sein?

Bei vielen WP geht das über Programmierung des Modus

> Was das Leben aber wesentlich erleichtern würde ist, wenn man das Ganze auf zwei Loadpoints aufteilen könnte. 

Aktuell sehe ich da keine gute Lösung. Lass uns das wieder anschauen wenn sich die "Heating" Unterstützung etwas weiter entwickelt hat. Für's erste- mit Schmerzen- wontfix.

closed 2025-02-24 by andig

jeriox, 2025-02-24:
> Für's erste- mit Schmerzen- wontfix.

gilt das auch für die einfache Variante nur für heating?

> Willst Du es mal probieren? Siehe Stiebel bzgl. Umsetzung.

ich kann im märz mal schauen ob ich Zeit und Lust habe mich in eure Codebase einzuarbeiten

andig, 2025-02-24:
Wenn die einfache Variante über einen Modus erreichbar ist kein Problem. Beispiele hatte ich schon genannt.

andig, 2025-02-25:
Schau Dir das gerne in Ruhe an. Wenn Du eine konkrete Idee hast lass uns gerne wieder auf machen.

daniel309, 2025-05-23:
Die luxtronik kann jetzt modbus. Nennt sich SHI Funktion und muss manuell aktiviert werden.

Details und Register Doku: https://github.com/raibisch/mylibs/blob/main/LuxModbusSHI/LuxtronikSHI.md

jeriox, 2025-05-25:
@daniel309 danke für das Update, das ist ja mal mega cool! Dadurch dürfte eine Einbindung ja noch deutlich einfacher sein als über die oben beschriebene Krücke.

Hast du da vor was zu bauen? Bei mir ist es zeitlich leider immer noch etwas schwierig

daniel309, 2025-05-25:
> Hast du da vor was zu bauen? Bei mir ist es zeitlich leider immer noch etwas schwierig

Im Moment nicht, nein.