evcc-io/evcc #19023: Panasonic Wärmepumpen Anbindung über Heishamon MQTT

sloodown, 2025-02-21:
Hallo zusammen,

**Is your feature request related to a problem? Please describe.**

Panasonic Wärmepumpen Anbindung über Heishamon
**Describe the solution you'd like**

Anbindung über MQTT

Link zum Github Projekt:

https://github.com/Egyras/HeishaMon

Link zu Dokumentation

https://github.com/Egyras/HeishaMon/blob/master/ProtocolByteDecrypt.md#protocol-byte-decrypt-info

Beste Grüße!

sloodown, 2025-02-21:
Deswegen explizit der Hinweis auf die Heishamon Platine.

Es gibt Leistungs- sowie Energiemessung intern. Ich habe auch externe Zähler (Strom und Wärmemenge) aber eine Abweichung über Jahr gesehen von >2% also zu vernachlässigen. Die Dimmung geht über die Heishamon Platine problemlos per MQTT. Die Dip Switches sind mir nicht bekannt ob in Generation J,K,L oder M. Die SG Ready Kontakte können beliebig parametriert werden per MQTT.

Ich würde es gerne ohne Umwege direkt in EVCC einbinden, aktuell sind meine Panasonic WPs alle ne Wallbox klappt zwar auch ist aber nicht sauber umgesetzt.

premultiply, 2025-02-21:
Bitte konkrete Anwendungsfälle.

Für die interne Leistungsschätzung der Panasonic WPs in groben abgerundeten 0,2 kW Stufen lohnt eher keine aufwändige Anbindung?
Das geht mit einem externen Zähler einfacher, schneller und viel genauer.

andig, 2025-02-21:
Ansteuerung der SG Ready Eingänge/Modi wäre ein case falls das ginge.

andig, 2025-02-22:
Alternativ: Ansteuerung über verfügbaren Überschuss, siehe https://github.com/evcc-io/evcc/pull/19008

andig, 2025-02-24:
Ping @sloodown

sloodown, 2025-02-25:
Sorry wegen der späten Antwort..

Ansteuerung über verfügbaren Überschuss klingt gut, hatte ich auch schon in Betracht gezogen.

premultiply, 2025-02-25:
Die Pana bietet keine Eingriffsmöglichkeit/Register o.ä. dafür.

Die 0-10V-Schnittstelle macht das Gegenteil davon was hier benötigt würde:
Leistungslimitierung im oberen Bereich

Einzig nutzbares Interface ist SG-Ready die man intern auf zwei prozentuale Vorgabewerte vorkonfigurieren kann. Insgesamt somit 4 Kombinationen
0% (veraltete RSE-Sperre)
100% (Normaler Betrieb)
User1 %
User2 %

andig, 2025-02-25:
Dann lasst uns hier zu machen. Wenn es eine gute Idee gibt, _was_ man sinnvoll machen könnte wäre das dann auch schnell umgesetzt.

closed 2025-02-25 by andig

mk98b30, 2025-03-08:
Ich halte die Anbindung von Heishamon an EVCC zur Steuerung der Panasonic Wärmepumpen für wichtig, wenn man digitale Schnittstellen nutzen möchte.

Eine direkte digitale Kommunikation ist flexibler und zukunftssicherer, als die Ansteuerung analoger Kontakte (für die bei den Panasonic Wärmepumpen eine Zusatzplatine benötigt wird, die durch Heishamon eigentlich überflüssig ist).

andig, 2025-03-08:
Die Frage ist doch trotzdem: wie _konkret_?

mk98b30, 2025-03-08:
Es gibt mehrere Möglichkeiten. Heishamon bietet alle Werte als JSON Output über http an und auch als MQTT Sensor Topic. Zum Steuern gibt es Control Topics über MQTT oder eine REST API.

Beispiel:
MQTT: send mqtt message to base_topic/commands/SetTopic (e.g.: panasonic_heat_pump/commands/SetHeatpump)

HTTP REST API: http://x.x.x.x/command?[topic]=[value]&[topic]=[value] (e.g.: http://x.x.x.x/command?SetQuietMode=3&SetZ1HeatRequestTemperature=21_

Quelle: https://github.com/Egyras/HeishaMon/blob/master/MQTT-Topics.md#command-topics

andig, 2025-03-08:
Die Frage ist doch nicht was geht, sondern was ein sinnvoller Ansatz wäre: wie konkret?!

mk98b30, 2025-03-08:
Konkret wäre aus meiner Sicht, das EVCC auch die Panasonic Wärmepumpen via Heishamon unterstützt. (Da Panasonic ab Werk nicht steuerbar ist und selbst mit Zusatzplatine nur analog.)

"Sollte deine Wärmepumpe eine Schnittstelle haben, die wir noch nicht unterstützen, erstelle gerne ein GitHub Issue mit allen notwendigen Informationen"

SG Ready setzen für PV Überschuss und §14a Dimmung wäre aus meiner Sicht der erste Ansatz. Den Ansatz mit Shellys und der Zusatzplatine von Panasonic zu arbeiten finde ich unsauber. Besonders weil die Anbindung relativ einfach wäre, da gut dokumentierte Schnittstellen vorhanden sind und Heishamon viele Panasonic Wärmepumpen unterstützt.

https://docs.evcc.io/docs/devices/heating
https://docs.evcc.io/blog/2024/08/17/highlights-14a-enwg-ocpp-loadmanagement-elli

andig, 2025-03-11:
@mk98b30 siehe https://github.com/evcc-io/evcc/issues/19023#issuecomment-2708338595. Wir haben uns jetzt wunderbar im Kreis gedreht. Was heisst das denn nun konkret mit Bezug auf Heishamon?

andig, 2025-03-11:
> Ansteuerung über verfügbaren Überschuss klingt gut, hatte ich auch schon in Betracht gezogen.

@sloodown Dito. Was heisst das konkret?

basvinken, 2025-05-20:
> Was heisst das konkret?

Heizen wenn die Sonne scheint

Die einfachste Anwendung ist das Erwärmen von Wasser. 
Wenn die Sonne scheint, möchte man warmes Wasser erzeugen. Grundsätzlich möchte man alle 24 Stunden während der Sonnenstunden das Wasser erwärmen. 
Es hängt alles stark von der Nutzung von warmem Wasser ab, aber wenn man das Wasser wärmer machst als eingestellt, kann man eine längere Zeitspanne überbrücken. 
Mit der API von heisamon kann man das alles steuern.

andig, 2025-05-20:
Das geht mit `type: sgready` oder `type: heatpump`, siehe die diversen Beispiele dazu.

RenatusRo, 2025-05-20:
Das muestte jemand mal mit  heisamon ausprobieren und dann auf die Liste schreiben.

andig, 2025-06-24:
> Wenn das funktioniert, müsste heisamon dann nicht auf dieser Liste stehen? https://docs.evcc.io/docs/devices/heating

Dafür müsste es erstmal für das Gerät ausgeprägt werden.

premultiply, 2025-06-24:
> SG Ready ist aus meiner Sicht viel zu unflexibel um den heutigen Anforderungen gerecht zu werden. Deshalb unterstützen die innovativsten Wärmepumpen aktuell EEBus.

Sehr steile These, fehlender use case.

Ob man jetzt die WP über die vorgesehenen Kontakte ansteuert oder das mit einem Heishamon-Zusatzboard simuliert macht im Verhalten keinen Unterschied.

Allerdings braucht es dazu jemanden der die entsprechenden Schnittstellen bei Heishamon raussucht, implementiert und testet.

Wer schon etwas länger eine WP betreibt wird mit der Zeit nüchterner was die realen Möglichkeiten und vor allem tatsächlichen Notwendigkeiten einer externen Steuerung des Heizsystems im Jahresverlauf betrifft. 😉

sloodown, 2025-06-24:
> Dann fehlt mir leider immer noch die Anbindung ans EMS und damit auch §14a Dimmung. Es ist auch nicht sinnvoll, das EMS über Homeassistant mit der Wärmepumpe / Heishamon kommunzieren zu lassen. Dann brauche ich EVCC als EMS nicht, wenn ich sowieso alle Regeln, Schnittstellen, etc. selber definieren muss.
> 
> [@premultiply](https://github.com/premultiply) SG Ready ist aus meiner Sicht viel zu unflexibel um den heutigen Anforderungen gerecht zu werden. Deshalb unterstützen die innovativsten Wärmepumpen aktuell EEBus.
> 
> Konkrete Use Cases müsste ich erst definieren, hier aber etwas Inspiration: https://chatgpt.com/share/685a78ff-64b0-800c-b8fe-234973fd0153
> 
> https://kiwigrid.com/de/artikel/connectivity-partnerschaft-mit-vaillant-wie-kiwigrid-waermepumpen-ins-energiemanagement-einbindet

Es gibt sowieso noch keine zertifizierten Gateways die womit man über EEBus dimmen könnte, vielleicht landen mal Endes Jahres mal welche wirklich am Markt. Und das dimmen an sich geht mit der Panasonic sowie nicht so wie es sein müsste für 14a. Selbst wenn man eine moderne Wärmepumpe hat mit EEBus bringt sie einem aktuell nichts, ich halte es sogar für möglich das manche Netzbetreiber auch einfach die Netze ausbauen. Die wenigstens wollen ernsthaft eine Steuerung etablieren.

mk98b30, 2025-06-24:
> Das wird hier ziemlich OT und mir sind da viel zu viele realitätsferne Annahmen über WPs und Heizsysteme im allgemeinen drin.
> 
> Um nochmal auf den Kern zurückzukommen: Eine Panasonic WP bietet herstellerseitig alle üblichen Schnittstellen um sowohl eine temporäre Verbrauchserhöhung als auch die Dimmung für §14a EnWG umzusetzen.
> 
> Wenn diese Hardware-Schnittstellen nun durch Heishamon ersetzt werden sollen braucht es dafür jemanden der explizit herausfindet wie genau dies softwareseitig via Heishamon erfolgt und dies dann auch Implementiert und vor allem testet.
> 
> Ansonsten bleibt es wie es ist: Panasonic ist eine SG-Ready kompatible WP wie alle anderen modernen WPs auch.

Ich würde das sehr gerne (in ein paar Monaten, wenn ich auch dafür Zeit habe) übernehmen und würde mich dann hier nochmal melden. Ich wollte nur den Kommentar von @hangingloose ergänzen um aufzuzeigen, dass es aus meiner Sicht absolut sinnvoll ist Wärmepumpen auch über SG-Ready hinaus in ein HEMS zu integrieren.