evcc-io/evcc #19024: Beim Anlegen neuer Ladepunkte über UI werden die alten alle zu "Ladepunkt"

StevieC121176, 2025-02-21:
Ich habe vor kurzem meine beiden Ladepunkte in der UI neu angelegt und sie genau wie die alten in der yaml benannt. 
In der Übersicht der Ladevorgänge werden die alten Ladepunkte jetzt nur noch als "Ladepunkt" angezeigt anstatt mit Ihren alten Namen. Das verfälscht die ganze grafische Auswertung

[image]

VolkerK62, 2025-02-21:
Hast du von 0.200.0  schon die Nightly Versionen benutzt?

StefanSchoof, 2025-02-21:
Wichtig dazu auch

https://github.com/evcc-io/evcc/pull/18868

> Note: In 0.200 loadpoint title was not stored in the database. We have no way to auto-recover. The only way to fix this data right now is manual db-edit.

StevieC121176, 2025-02-21:
Oh, ja ich hab von der 0.200.0 die nightly benutzt. 

Dann werde ich mich wohl mal am ändern der db versuchen.

closed 2025-02-21 by StevieC121176

StefanSchoof, 2025-02-21:
Oder du wartest bis https://github.com/evcc-io/evcc/pull/19016 in einem Release ist.