evcc-io/evcc #19026: EM2GO Charger via OCPP cause generic error - {"status":"NotImplemented"}

TheAgentHH, 2025-02-21:
### Describe the bug

My EM2GO (Model EM022AS0OCPP) charger seems to be limited in regards to OCPP.
In evcc aprox 1 year ago it was working fine, but now we seems to have a problem with the OCPP support of this charger.

### Steps to reproduce

1. just start EVCC, this error is part of the 30sec communication cycle between charger and evcc
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 30s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: kostal-plenticore
  id: 71
  host: 192.168.220.16
  port: 1502
  usage: pv
  modbus: tcpip
  endianness: little
  minsoc: 5
  maxsoc: 100
  watchdog: 60s
  name: pv1
- type: template
  template: kostal-plenticore
  id: 71
  host: 192.168.220.16
  port: 1502
  usage: battery
  modbus: tcpip
  endianness: little
  minsoc: 5
  maxsoc: 100
  watchdog: 60s
  name: battery2
- type: template
  template: kostal-ksem-inverter
  id: 71
  host: 192.168.220.16
  port: 1502
  usage: grid
  modbus: tcpip
  name: grid3

chargers:
- type: template
  template: ocpp
  stationid: 84702764194358
  connector: 1
  idtag: *****
  connecttimeout: 5m
  timeout: 2m
  autostart: false
  nostop: false
  getconfiguration: true
  bootnotification: true
  meterinterval: 10s
  chargingrateunit: W
  name: wallbox2

loadpoints:
- title: Carport-rechts
  charger: wallbox2
  mode: off

site:
  title: Mein Zuhause
  meters:
    grid: grid3
    pv:
    - pv1
    battery:
    - battery2

Version: `0.200.1 (acbcf546)`
```

### Log details

```text
evcc --database ~/.evcc/evcc.db
[main  ] INFO 2025/02/21 09:15:10 evcc 0.200.1 (acbcf546)
[main  ] INFO 2025/02/21 09:15:10 using config file: /home/markus/evcc.yaml
[db    ] INFO 2025/02/21 09:15:10 using sqlite database: /home/markus/.evcc/evcc.db
[main  ] INFO 2025/02/21 09:15:10 listening at :7070
[84702764194358-1] DEBUG 2025/02/21 09:15:11 !! registering 84702764194358:1
[84702764194358-1] DEBUG 2025/02/21 09:15:11 waiting for chargepoint: 5m0s
[ocpp  ] DEBUG 2025/02/21 09:16:00 charge point connected: 84702764194358
[84702764194358-1] DEBUG 2025/02/21 09:16:30 failed configuring availability: timeout
[84702764194358-1] DEBUG 2025/02/21 09:16:34 failed configuring WebSocketPingInterval: NotSupported
[84702764194358-1] DEBUG 2025/02/21 09:16:34 !! connected 84702764194358:1
[site  ] INFO 2025/02/21 09:16:34 site config:
[site  ] INFO 2025/02/21 09:16:34   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2025/02/21 09:16:34     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/02/21 09:16:34     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/02/21 09:16:34     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[lp-1  ] INFO 2025/02/21 09:16:34 loadpoint 1:
[lp-1  ] INFO 2025/02/21 09:16:34   mode:        off
[lp-1  ] INFO 2025/02/21 09:16:34   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2025/02/21 09:16:34   meters:      charge ✗
[site  ] DEBUG 2025/02/21 09:16:34 set buffer soc: 50
[site  ] DEBUG 2025/02/21 09:16:34 set battery discharge control: true
[lp-1  ] DEBUG 2025/02/21 09:16:34 phase timer inactive
[lp-1  ] DEBUG 2025/02/21 09:16:34 pv timer inactive
[ocpp  ] ERROR 2025/02/21 09:16:34 ocpp message (221780226): GenericError - Key: 'CallResult.Payload.Status' Error:Field validation for 'Status' failed on the 'compositeScheduleStatus' tag ([3,"221780226",{"status":"NotImplemented"}])
[site  ] DEBUG 2025/02/21 09:17:04 ----
[lp-1  ] DEBUG 2025/02/21 09:17:04 charge power: 0W
[site  ] DEBUG 2025/02/21 09:17:04 grid power: 594W
[site  ] DEBUG 2025/02/21 09:17:04 battery 1 power: 3W
[site  ] DEBUG 2025/02/21 09:17:04 battery 1 soc: 5%
[site  ] DEBUG 2025/02/21 09:17:04 pv 1 power: 211W
[site  ] DEBUG 2025/02/21 09:17:04 site power: 597W
[lp-1  ] DEBUG 2025/02/21 09:17:04 charger status: A
[lp-1  ] INFO 2025/02/21 09:17:04 car disconnected
[lp-1  ] DEBUG 2025/02/21 09:17:04 set battery boost: false
[lp-1  ] DEBUG 2025/02/21 09:17:04 set session soc limit: 0
[lp-1  ] DEBUG 2025/02/21 09:17:04 set session energy limit: 0
[ocpp  ] ERROR 2025/02/21 09:17:05 ocpp message (1633819893): GenericError - Key: 'CallResult.Payload.Status' Error:Field validation for 'Status' failed on the 'compositeScheduleStatus' tag ([3,"1633819893",{"status":"NotImplemented"}])
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.1 (acbcf546)

mfuchs1984, 2025-02-21:
The charger responds with an invalid response. According to the ocpp standard, only accepted and rejected are valid for GetCompositeSchedule. Does is work apart from this issue? If so, you can ignore the error.

andig, 2025-02-21:
/cc @dniakamal could this be fixed?

dniakamal, 2025-02-21:
> /cc [@dniakamal](https://github.com/dniakamal) could this be fixed?

Thanks for reporting. We already informed our developers to fix this issue. Sorry for the inconvenience!

Will inform you about the release date of the new firmware. Updates will be done automatically if EVSE is connected to the internet.

TheAgentHH, 2025-02-21:
/cc @dniakamal 
I‘m monitoring the outbound connections and I never have seen my charger to make any outbound requests.
My charger is running on AC_SCE_3.12EG
I‘m more that happy to get some firmware updates if available.

andig, 2025-02-21:
Great support, thank you @dniakamal !

TheAgentHH, 2025-02-22:
@mfuchs1984 
charging works, so I can ignore it for the moment.
Looks like there is another issue with the OCPP within the EVSE charger, as it breaks the telemetry to work

[site ] TRACE 2025/02/22 16:54:48 telemetry: charge: Δ140/7361Wh @ 22080W
[site ] TRACE 2025/02/22 16:54:48 POST https://api.evcc.io/v1/charge
[site ] TRACE 2025/02/22 16:54:48 {"instanceId":"810ded74e2517f26e1a281ecb4f94a8c72647b3328a1342ad9a4d8863a76b27a","chargePower":22080,"greenPower":367,"chargeEnergy":7.360922171169334,"greenEnergy":0.13992784783570614} -- {"error":"implausible charge power: 22080"}
[site ] ERROR 2025/02/22 16:54:48 telemetry: upload failed: unexpected status: 400 (Bad Request)
[site ] DEBUG 2025/02/22 16:55:17 ----

I'm going to open another bug for that.

TheAgentHH, 2025-02-22:
@andig Okay, ich habe heute festgestellt das mit grid meter nicht an der Stelle sitzt, wo er hingehört. Charger und grid meter sind parallel zum Netzanschluss und daher sieht er nicht wo der Strom herkommt Vielleicht ist das schon des Rätzels Lösung ?

TheAgentHH, 2025-02-22:
by the way.. ich bin von EVCC und der Community so begeistern das ich auf eine große Sponsor Token wechseln werde

closed 2025-03-03 by andig

mfuchs1984, 2025-03-03:
Bzgl. der geladenen Energie gibts in https://github.com/evcc-io/evcc/discussions/18943 interessante Beobachtungem bzgl. der Session Energy bei Ladeunterbrechungen und Phasenwechsel.