evcc-io/evcc #19036: Victron Batterie nicht gesperrt

Mikka20, 2025-02-21:
### Describe the bug

Hallo,

obwohl die Option "Verhindere Entladung im Schnell-Modus und bei geplantem Laden." aktiviert ist, wird die Batterie während des Schnelladens entladen.

Bisher getestet mit den Versionen und von einem Konfigurationsfehler ausgegangen.
0.132.0
0.132.1
0.133.0
0.200.0
0.200.1
0.200.1 (18247b3f)

### Steps to reproduce

Auto an Wallbox anschließen und laden lassen :-)

### Configuration details

```yaml
network:
  schema: http
  host: evcc.localdomain
  port: 7070

interval: 30s 

sponsortoken: ****

telemetry: false

log: info

meters:
   - name: eastron-sdm72
     type: template
     template: eastron
     usage: charge
     modbus: tcpip
     id: 2
     host: IP-Adresse
     port: 8899
   - name: victron_battery
     type: template
     template: victron-energy
     usage: battery
     host: IP-Adresse
     port: 502 
     capacity: 381
     minsoc: 50
     maxsoc: 
   - name: victron_pv
     type: template
     template: victron-energy
     usage: pv
     host: IP-Adresse
     port: 502
     minsoc: 20
     maxsoc: 
   - name: victron_grid
     type: template
     template: victron-energy
     usage: grid
     host: IP-Adresse
     port: 502

chargers:
  - name: HeidelbergEnergieControl
    type: template
[... 247 lines cut ...]
[site  ] DEBUG 2025/02/19 21:29:08 battery 1 soc: 95%
[site  ] DEBUG 2025/02/19 21:29:08 grid power: 31W
[site  ] DEBUG 2025/02/19 21:29:08 pv 1 power: 0W
[site  ] DEBUG 2025/02/19 21:29:08 site power: 607W
[lp-1  ] DEBUG 2025/02/19 21:29:09 charge voltages: [233 232 232]V
[lp-1  ] DEBUG 2025/02/19 21:29:09 detected connected phases: 3p
[lp-1  ] DEBUG 2025/02/19 21:29:09 charge total import: 2737.627kWh
[lp-1  ] DEBUG 2025/02/19 21:29:09 charger status: B
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.1 (18247b3f)

tugsi, 2025-02-21:
Moin, 
hab auch ein Victron-System und kann den "Fehler" so nicht bestätigen.
Bei mir sperrt er die Batterie.
Was hast Du in den Batterieeinstellungen im evcc bei dem Punkt: Batterieunterstütztes Fahrzeugladen eingestellt? 
Vielleicht einen niedrigeren Prozentwert, als Du in der Batterie hattest? 
Wenn ich das richtig im Log lese, hatte Dein Hausakku einen SOC von 97% beim starten. 
Eventuell hast Du beim Batterieunterstütztes Fahrzeugladen so etwas wie 95% oder niedriger drinstehen?

Mikka20, 2025-02-22:
Hi tugsi,

ja genau das war es, super vielen Dank für den Tipp! Ich bin davon ausgegangen, dass wenn man diese Option einschaltet, die anderen Einstellungen aus erkraft gesetzt werden.

Eine Frage habe ich trotzdem noch, ist das bei dir auch so, dass beim Laden (also beim Startvorgang) für ein paar Sekunden die Akkus trotzdem mit 11kW belastet werden?

VG

tugsi, 2025-02-22:
Moin,
schön das ich helfen konnte. 
Es gibt auch noch unter der Einstellung zum laden, da wo man 1-oder3-phasig auch einstellen kann, einen experiementellen Punkt: Schnell aus Hausbatterie laden.  Auch da kannst die Hausbatterie mit "belasten". 
ist aber Standard Aus.

Zu Deinem anderen Punkt, da hatte ich bisher nie drauf geachtet, habe aber gerade einfach mal auf Schnell-Laden eingestellt und beobachtet. 
Das Auto fängt an zu laden, alles kommt aus dem Netz, Batterie wurde vorher durch PV geladen.
Allerdings so rund 10 Sekunden später, lädt er aus der Batterie mit für so rund 30s und schaltet erst dann wieder ab.
Ist mir bisher noch nicht aufgefallen, muss aber auch zugeben, habe ich noch nie drauf geachtet. 
Ich vermute, dass es eine Regelgeschichte ist, da kommt sich kurzfristig etwas in die Quere, so als ob evcc erst sperrt, Victron dann aufmacht aufgrund einer Trägheit und dann evcc dies wieder sperrt. 
Um das genau zuanalysieren, müsste man sich die Regelparameter im Victron anschauen, was gibt er als Soll für die Batterie und wann kommt die von evcc. 
Andersherum, wenn es immer nur so kurz ist, dann könnte ich damit leben. Richtig wäre sicherlich anders. 
In meinem Fall zieht er 5kW aus dem Akku, da nur ein MP2-5000er dranhängt.
Gruß

tugsi, 2025-02-23:
Da @andig dies ja gettaged hat, ich denke zur Überprüfung, sollte noch irgendwas gebraucht werden, bitte einfach melden.
Ich habe gerade auch gesehen, in der Konfiguration steht als Standard das Regelverhalten 20 Sekunden.
Daher vermute ich eher, dass meine gefühlten 30s bis die Batterie wieder geblockt wurde, die 20s Regelverhalten waren.

Bei mir war es ja so: 

- PV speist Hausakku
- EV wird im Lademodus schnell gestartet
- PV und Grid speisen das EV
- Victron merkt, mit Verzögerung, Grid wird bezogen und steuert den Hausakku dazu
- nach max 20s merkt evcc, Hausakku wird mit entladen und sperrt ihn 

Jetzt kenn ich die Programmierung im Background nicht, aber sperrt evcc generell den Hausakku oder erst wenn daraus bezogen wird?
Bei ersterem ist es ein "Bug", dannn sollte es nicht passieren (ausser man ist noch im Regelverhalten, also max 20s).
Wird der Hausakku nicht pauschal geblockt, dann kann es kurzfristig passieren. 

Laut @Mikka20 dauert es bei ihm ja auch nur ein paar Sekunden und das könnte das Regelverhalten sein.

endurance1968, 2025-02-23:
Also ich denke das ist Regelträgheit. Ich habe 2 Go-E ladestationen, Victron Speicher + Eingebundene WP.
Vor kurzen war Strom mal wieder etwas billiger, da sind Ladestationen und Speicher gleichzeitig angesprungen. Trotz Limit auf 32.5A bin ich dann für ~10+s? auf 35.xA hoch am Netzübergabepunkt. Ich habe 10s Regelintervall. Kurz danach wurde die WP abgeschaltet und Speicher reduziert (der reduziert sich da Peakshaving aber von selbst). Zum Glück sind SLS ja mit absicht träge sonst wäre das Haus vom Netz getrennt worden.
Wenn das Hochfahren der WB etwas langsamer in Schritten erfolgen würde - könnte die Regelung evtl. besser greifen. Wobei ich noch nie mitgelogt habe ob und wie die Stromstärke hochgefahren wird (oder ob das schlagartig auf max geht).

tugsi, 2025-02-23:
@endurance1968 Bin da ganz bei Dir. Bin beruflich im Bereich Mess- und Regelsteuerung und Automatisierung, man kann sich manchmal auch dumm und dämlich regeln, es gibt immer die aufgrund der Trägheit etwas nachlaufen können oder sogar auch müssen. 
SLS sind absichtlich extrem träge, auch Deine NH-Sicherungen im Übergabekasten müssten schon um das x-fache überschritten sein und das über Minuten. 
Von meiner Seiet sehe ich auch keinen direkten Handlungsbedarf, da es für einige Sekunden vernachlässigter ist.
Wenn @Mikka20 dies genauso sieht, kann er das Issue ja schliessen.

closed 2025-02-23 by andig