evcc-io/evcc #19041: Tasmota SML IR Head outputs 0

jeriox, 2025-02-21:
### Describe the bug

evcc reports a power of 0W for the  Tasmota IR head altough the script has been set according to the docs and evcc can correctly query the API (see trace output of `evcc meter` in the log section below)

### Steps to reproduce

1. Set script from docs in Tasmota head
2. Include Tasmota head in evcc.yaml according to docs
3. Observe empty value

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 10s # control cycle interval

sponsortoken: *****

meters:
  - type: template
    template: sma-energy-meter
    usage: grid
    name: grid1
    host: ***
  - name: pv_neu
    type: template
    template: sma-hybrid
    usage: pv

    # Modbus TCP
    modbus: tcpip
    id: 3
    host: ***
    port: 502
  - name: battery
    type: template
    template: sma-hybrid
    usage: battery

    # Modbus TCP
    modbus: tcpip
    id: 3
    host: ***
    port: 502
  - name: pv_alt
    type: template
    template: tasmota-sml
    usage: pv
    host: ***

site:
  title: Erlenweg
  meters:
    grid: grid1
    pv:
      - pv_neu
      - pv_alt
    battery: battery

chargers:
  - name: wallbox1
    type: template
    template: heidelberg
    id: 1
    device: /dev/ttyAMA0
    baudrate: 19200
    comset: 8E1
    modbus: rs485serial

loadpoints:
  - title: Garage
    charger: wallbox1
    vehicle: Kona
    disable:
      delay: 10m

tariffs:
  grid:
    type: tibber
    token: *****
  feedin:
    type: fixed
    price: 0.14
  solar:
    type: template
    template: solcast
    site: ***
    token: *****

vehicles:
  - name: Kona
    type: template
    template: hyundai
    capacity: 65.4
    user: *****
    password: *****
```

### Log details

```text
pv_alt
------
[tasmota] TRACE 2025/02/21 15:11:58 GET http://***/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2025/02/21 15:11:58 {"StatusSNS":{"Time":"2025-02-21T16:11:57","SML":{"Total_in":0.0002,"Total_out":746.5222,"Power_curr":-1187,"Meter_Id":"0a0149534b00057dd61d"}}}
Power:  0W
Energy: 0.0kWh
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.1

VolkerK62, 2025-02-21:
`tasmota-sml` is only for grid usage.
The value is negative, that is the problem.
try `tasmota` https://docs.evcc.io/docs/devices/meters#tasmota-1-phase-meter

andig, 2025-02-21:
@VolkerK62 why should a grid meter not display negative values?

VolkerK62, 2025-02-21:
???
THe tasmota meter (name: pv_alt) is configured as pv-meter!
So negative values are displayed with `0`

andig, 2025-02-22:
@VolkerK62 misunderstood you in "tasmota-sml is only for grid usage.", sorry. Imho we still shouldn't abs the readings, similar to https://github.com/evcc-io/evcc/pull/18841 (/cc @thierolm). 

Closing this as config issue.

closed 2025-02-22 by andig

thierolm, 2025-02-22:
In der Tasmota Umsetzung gebe ich für den SML Lesekopf nur bei der usage grid etwas aus:
https://github.com/evcc-io/evcc/blob/8e521ca6530cedd0c798d6cff31bac1b6df97c2e/meter/tasmota.go#L69-L86

thierolm, 2025-02-22:
@jeriox kannst du bitte testen, ob dein Usecase mit dem PR oben funktioniert?
Bitte aber auf das standard `tasmota` template wechseln:

```yaml
  - name: pv_alt
    type: template
    template: tasmota
    usage: pv
    host: ***
```

jeriox, 2025-02-23:
> [@jeriox](https://github.com/jeriox) kannst du bitte testen, ob dein Usecase mit dem PR oben funktioniert? Bitte aber auf das standard `tasmota` Template wechseln:

Es kommt ein Wert raus, aber evcc mag das Vorzeichen nicht

```
[site  ] WARN 2025/02/23 12:30:36 pv_alt power: -4499W is negative - check configuration if sign is correct
```

thierolm, 2025-02-23:
@premultiply kannst du hier helfen? Ich gebe jetzt den Power-Werte des Tasmota SML Lesekopfs 1:1 aus. @jeriox hat den daran angeschlossenen Zähler offensichtlich an seiner alten PV Anlage. Muss ich für die usage `pv` und `battery` dann das Vorzeichen drehen?

thierolm, 2025-02-23:
@jeriox kannst du bitte mit der Version die Ausgabe von `evcc -l trace meter` teilen?

jeriox, 2025-02-23:
> [@jeriox](https://github.com/jeriox) kannst du bitte mit der Version die Ausgabe von `evcc -l trace meter` teilen?

``` 
pv_alt
------
[tasmota] TRACE 2025/02/23 13:16:23 GET http://***/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2025/02/23 13:16:23 {"StatusSNS":{"Time":"2025-02-23T13:16:23","SML":{"Total_in":0.0002,"Total_out":779.0518,"Power_curr":-4727,"Meter_Id":"0a0149534b00057dd61d"}}}
Power:  -4727W
Energy: 779.1kWh

```

premultiply, 2025-02-23:
@thierolm Genau. Da die Zähler mit SML-Schnittstelle quasi immer "offizielle" Zähler sind, zeigen diese immer mit der Import-Seite (Netzbezug, Energiezählwerk 1.8.0) vom Netzanschluss weg. Einspeisung ist daher dort immer negativ (A-) und die zugehörige Energiemenge unter 2.8.0 zu finden.

thierolm, 2025-02-23:
> Genau

OK, dann drehe ich das SML Power-Vorzeichen für die usages `pv` und `battery`. 
Für die usages `grid` und `charge` lasse ich es, so wie es kommt.

Richtig?

premultiply, 2025-02-23:
Exakt.
Aber irgendwer wird es trotzdem andersrum haben... 🤦🏻‍♂️😅

thierolm, 2025-02-23:
@jeriox kannst du [#19072](https://github.com/evcc-io/evcc/pull/19072) noch einmal testen? Es müsste jetzt wie gewollt funktionieren.
Wenn du Vorschläge zur Optimierung der Dokumentation hast, schick mir die entsprechenden Links zur Doku und deinen Vorschlagstext, dann können wir sie einbauen.
@premultiply eigentlich könnten wir die templates auch zu einem zusammenführen, ich finde es aber besser, das SML template separat zu haben, weil da die Hinweise für das Tasmota Lesekopf Script drin sind, was meinst du?

jeriox, 2025-02-24:
> [@jeriox](https://github.com/jeriox) kannst du [#19072](https://github.com/evcc-io/evcc/pull/19072) noch einmal testen? Es müsste jetzt wie gewollt funktionieren. 

Ich bin jetzt leider ne Weile unterwegs und hab keinen direkten Zugriff auf den Lesekopf :/

> Wenn du Vorschläge zur Optimierung der Dokumentation hast, schick mir die entsprechenden Links zur Doku und deinen Vorschlagstext, dann können wir sie einbauen. [...]

Ich glaube die verschiedenen Templates haben genau meine Verwirrung ausgelöst. Ich bin über https://docs.evcc.io/docs/devices/meters#tasmota-sml-ir-lesekopf-f%C3%BCr-smarte-stromz%C3%A4hler-bspw-hichi gestolpert und habe dann meine Config entsprechend eingerichtet, weil ich ja nunmal genau einen solchen Lesekopf habe. Es gibt von dort ja keinen Verweis auf das andere Template, und da das Gerät ja genau gepasst hat bin ich gar nicht auf die Idee gekommen dass ich was anderes nehmen muss, der Verweis kam dann erst in diesem Issue auf. Einen konkreten Vorschlag kann ich gerade nicht machen, da ich ehrlicherweise immer noch nicht genau verstanden habe was die Templates unterscheidet und welches man für welchen Fall benötigt.

jeriox, 2025-03-31:
> [@jeriox](https://github.com/jeriox) kannst du [#19072](https://github.com/evcc-io/evcc/pull/19072) noch einmal testen? Es müsste jetzt wie gewollt funktionieren. 

So ich bin endlich mal dazu gekommen auf das aktuelle Release zu updaten. Da funktioniert es für mich leider wieder nicht...

```
pv_alt
------
[tasmota] TRACE 2025/03/28 17:26:57 GET http://192.168.178.77/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2025/03/28 17:26:57 {"StatusSNS":{"Time":"2025-03-28T18:26:56","SML":{"Total_in":0.0002,"Total_out":1637.9433,"Power_curr":-64,"Meter_Id":"0a0149534b00057dd61d"}}}
Power:  0W
Energy: 0.0kWh
```

thierolm, 2025-03-31:
Deine Tags im JSON fangen mit Großbuchstaben an. Du müsstest das Script im Lesekopf so ändern das es zu folgender Definition passt:

https://github.com/thierolm/evcc/blob/9fa1991a0f319855e11811e49c1089e245632828/meter/tasmota/types.go#L86

Also so muss das Ergebnis aussehen:
```json
{"StatusSNS":{"Time":"2025-03-28T18:26:56","SML":{"total_in":0.0002,"total_out":1637.9433,"power_curr":-64,"Meter_Id":"0a0149534b00057dd61d"}}}
```

thierolm, 2025-03-31:
Oh, sehe gerade, dass die Doku da nicht so ganz passt ...
https://docs.evcc.io/docs/devices/meters#tasmota-sml-ir-lesekopf-f%C3%BCr-smarte-stromz%C3%A4hler-bspw-hichi
Passe ich bei Gelegenheit an, oder jeder, der Zeit vorher dazu hat ;-)

thierolm, 2025-03-31:
ja, weil es mein PR durch die vergangene Zeit nicht ins Release geschafft hat ... :-(

thierolm, 2025-03-31:
@jeriox da musst du mal @andig fragen, ob er https://github.com/evcc-io/evcc/pull/19072 nochmal aufmacht, oder ich einen neuen eröffnen soll.
Ich muss den negativen Power Wert deines Drehstromzählers den dann deiner PV hast dann bei Nutzung als pv oder battery Meter in einen positiven umwandeln.

thierolm, 2025-04-01:
@jeriox bitte teste jetzt nochmal den #19072