evcc-io/evcc #19050: charging plan visualisation in UI - timezone bug

tomaszrondio, 2025-02-22:
### Describe the bug

UI: 

[image]

log:
`plan: charge 2h50m46s between 2025-02-22 02:00:00 +0100 CET until 2025-02-22 09:00:00 +0100 CET `

it knows it should start charging at 1:00 (2:00 + 0100) but shows 2:00

### Steps to reproduce

1. be in different than CET timezone :)
2. Plan charging
3. compare visualization with logs

### Configuration details

```yaml
does not matter
```

### Log details

```text
no issue in logs
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.1

StefanSchoof, 2025-02-22:
Which time zone setting habe you on your Linux?

What is the output of `timedatectl`?

andig, 2025-02-22:
@tomaszrondio so the plan is correct (and correctly set) but the visualisation is wrong? Is that what you're saying?

andig, 2025-02-22:
>  Please show the output of /api/state, too.

Still need this for when the issue appears. Thank you.

tomaszrondio, 2025-02-22:
I will try to write some simple tool that calls this endpoint in order to gather logs. My „problem” is that I use dynamic tariffs and usually the cheapest electricity is in the middle of night.

andig, 2025-02-22:
You don't need a tool. Just open the url in your browser, please. At the time there is a problem.

tomaszrondio, 2025-02-22:
I usually sleep then :D will handle that and get back to You asap

StefanSchoof, 2025-02-22:
>  It knows in logs that I am CET + 1:00. 

Are you sure? The log says `02:00:00 +0100 CET`.  My interpretation would be plus hour from UTC with the name CET.

tomaszrondio, 2025-02-22:
i am runnig evcc as hassio addon. Ha cli says: 

[image]

evcc log says: 
`[lp-1 ] DEBUG 2025/02/22 21:06:55 charger status: A` which is correct

tomaszrondio, 2025-02-22:
current api/state response 
```{
  "result": {
    "auth": {

    },
    "aux": [],
    "battery": [],
    "batteryDischargeControl": false,
    "batteryGridChargeActive": false,
    "batteryMode": "unknown",
    "bufferSoc": 0,
    "bufferStartSoc": 0,
    "circuits": {
      "main": {
        "power": 238.057,
        "current": 1.302,
        "maxPower": 4600,
        "maxCurrent": 20
      }
    },
    "currency": "PLN",
    "eebus": false,
    "ext": [],
    "fatal": null,
    "forecast": {
      "grid": [
        {
          "start": "2025-02-22T00:00:00+01:00",
          "end": "2025-02-22T01:00:00+01:00",
          "price": 1.02171
        },
        {
          "start": "2025-02-22T01:00:00+01:00",
          "end": "2025-02-22T02:00:00+01:00",
          "price": 0.61483
        },
        {
          "start": "2025-02-22T02:00:00+01:00",
          "end": "2025-02-22T03:00:00+01:00",
          "price": 0.59513
        },
        {
          "start": "2025-02-22T03:00:00+01:00",
          "end": "2025-02-22T04:00:00+01:00",
          "price": 0.60253
        },
        {
          "start": "2025-02-22T04:00:00+01:00",
          "end": "2025-02-22T05:00:00+01:00",
          "price": 0.60201
        },
        {
          "start": "2025-02-22T05:00:00+01:00",
          "end": "2025-02-22T06:00:00+01:00",
          "price": 1.02048
        },
        {
          "start": "2025-02-22T06:00:00+01:00",
          "end": "2025-02-22T07:00:00+01:00",
          "price": 1.11219
        },
        {
          "start": "2025-02-22T07:00:00+01:00",
          "end": "2025-02-22T08:00:00+01:00",
          "price": 1.12159
        },
        {
          "start": "2025-02-22T08:00:00+01:00",
          "end": "2025-02-22T09:00:00+01:00",
[... 215 lines cut ...]
    "tariffGrid": 1.117,
    "tariffPriceHome": 1.117,
    "tariffPriceLoadpoints": 1.117,
    "vehicles": {
      "tesla": {
        "title": "Tereska",
        "icon": "car",
        "capacity": 75,
        "phases": 1,
        "minSoc": 20,
        "limitSoc": 80,
        "repeatingPlans": [
          {
            "weekdays": [1, 2, 3, 4, 5, 6, 0],
            "time": "09:00",
            "tz": "Europe/Warsaw",
            "soc": 65,
            "active": true
          }
        ]
      }
    },
    "version": "0.200.1"
  }
}```

StefanSchoof, 2025-02-22:
> i am runnig evcc as hassio addon.

Then your are in the docker container and the cet comes from the Dockerfile

https://github.com/evcc-io/evcc/blob/32fc13a916f342862c737d9acb34f57fc8355736/Dockerfile#L68

tomaszrondio, 2025-02-22:
>  ENV TZ=Europe/Berlin

Berlin is fine. Close enough to Warsaw :)

andig, 2025-02-22:
> "planActive": false,

@tomaszrondio I don't see plan data anywhere

tomaszrondio, 2025-02-22:
Sorry. There is an issue with price Forecast provider.  Will post api response again once everything is fine

tomaszrondio, 2025-02-24:
it looks like the issue is solved in 0.200.2

[image]

[image]

`[lp-1 ] DEBUG 2025/02/24 16:05:18 plan: charge 40m45s between 2025-02-25 01:00:00 +0100 CET until 2025-02-25 09:00:00 +0100 CET (power: 3680W, avg cost: 0.618)`

```{
  "result": {
    "auth": {

    },
    "aux": [],
    "battery": [],
    "batteryDischargeControl": false,
    "batteryGridChargeActive": false,
    "batteryMode": "unknown",
    "bufferSoc": 0,
    "bufferStartSoc": 0,
    "circuits": {
      "main": {
        "power": 153.658,
        "current": 1.215,
        "maxPower": 4600,
        "maxCurrent": 20
      }
    },
    "currency": "PLN",
    "eebus": false,
    "ext": [],
    "fatal": null,
    "forecast": {
      "grid": [
        {
          "start": "2025-02-24T00:00:00+01:00",
          "end": "2025-02-24T01:00:00+01:00",
          "price": 1.12282
        },
        {
          "start": "2025-02-24T01:00:00+01:00",
          "end": "2025-02-24T02:00:00+01:00",
          "price": 0.71164
        },
        {
          "start": "2025-02-24T02:00:00+01:00",
          "end": "2025-02-24T03:00:00+01:00",
          "price": 0.67879
        },
        {
          "start": "2025-02-24T03:00:00+01:00",
          "end": "2025-02-24T04:00:00+01:00",
          "price": 0.67625
        },
        {
          "start": "2025-02-24T04:00:00+01:00",
          "end": "2025-02-24T05:00:00+01:00",
          "price": 0.67633
        },
        {
          "start": "2025-02-24T05:00:00+01:00",
          "end": "2025-02-24T06:00:00+01:00",
          "price": 1.13512
        },
        {
          "start": "2025-02-24T06:00:00+01:00",
          "end": "2025-02-24T07:00:00+01:00",
          "price": 1.41802
        },
        {
[... 344 lines cut ...]
    "tariffGrid": 1.381,
    "tariffPriceHome": 1.381,
    "tariffPriceLoadpoints": 1.381,
    "vehicles": {
      "tesla": {
        "title": "Tereska",
        "icon": "car",
        "capacity": 75,
        "phases": 1,
        "minSoc": 20,
        "limitSoc": 80,
        "repeatingPlans": [
          {
            "weekdays": [1, 2, 3, 4, 5, 6, 0],
            "time": "09:00",
            "tz": "Europe/Warsaw",
            "soc": 65,
            "active": true
          }
        ]
      }
    },
    "version": "0.200.2"
  }
}

closed 2025-02-24 by tomaszrondio

andig, 2025-02-24:
There were no changes. Whatever it was is still there.

tomaszrondio, 2025-02-24:
Now I know what kind of logs are helpful. Will reopen when I notice something odd.

tomaszrondio, 2025-02-25:
Still all good. No idea what happened this one time