evcc-io/evcc #19057: TWO Huawei Luna2000 in cascaded setup --> false charging power

Kugelfang666, 2025-02-22:
### Describe the bug

Hi all,

I have a Huawei SunKTL10M1 connected to TWO Luna2000S0 with each 15kWh capacity in a cascaded setup. e.g. Inverter connected to Luna1 connected to Luna2

Home Assistant queries the Inverter via its near field WiFi while EVCC is connected to the sDongle. 

Apparently EVCC is only reading the charging power of one battery (I'm assuming the primary) and not the secondary battery. This results in energy which in fact is going into charging the battery appearing as consumption of the house. 

Any way to fix this? 

[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
- name: pv_inverter
    type: template
    template: huawei-dongle-powersensor
    usage: pv
    modbus: tcpip
    id: 1
    host: 192.168.178.144
    port: 502
    timeout: 15s

  - name: battery_inverter
    type: template
    template: huawei-dongle-powersensor
    usage: battery
    modbus: tcpip
    id: 1
    host: 192.168.178.144
    port: 502
    timeout: 15s
    capacity: 30
```

### Log details

```text
[site ] DEBUG 2025/02/22 10:03:01 ----
[lp-1 ] DEBUG 2025/02/22 10:03:01 charge power: 0W
[site ] DEBUG 2025/02/22 10:03:01 grid power: 110W
[site ] DEBUG 2025/02/22 10:03:01 grid currents: [122 -215 203]A
[site ] DEBUG 2025/02/22 10:03:01 pv 2 power: 3094W
[site ] DEBUG 2025/02/22 10:03:01 battery 1 power: -2824W
[site ] DEBUG 2025/02/22 10:03:02 pv 1 power: 2714W
[site ] DEBUG 2025/02/22 10:03:03 battery 1 soc: 35%
[site ] DEBUG 2025/02/22 10:03:04 pv power: 5808W
[site ] DEBUG 2025/02/22 10:03:04 battery has priority at soc 35% (< 40%)
[site ] DEBUG 2025/02/22 10:03:04 site power: 260W
[lp-1 ] DEBUG 2025/02/22 10:03:04 charge total import: 19.444kWh
[lp-1 ] DEBUG 2025/02/22 10:03:05 charger status: B
[lp-1 ] DEBUG 2025/02/22 10:03:05 vehicle api refresh
[lp-1 ] DEBUG 2025/02/22 10:03:05 pv charge current: 0A = 0A + -1.13A (260W @ 1p)
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-02-22:
Please show

    evcc meter

Kugelfang666, 2025-02-22:
```
[db ] TRACE 2025/02/22 10:56:05 SELECT COALESCE(SUM(charged_kwh * solar_percentage) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2025-01-23 10:56:05.406" AND charged_kwh > 0 AND solar_percentage IS NOT NULL 1 <nil>
[db ] TRACE 2025/02/22 10:56:05 SELECT COALESCE(SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2025-01-23 10:56:05.406" AND charged_kwh > 0 AND solar_percentage IS NOT NULL 1 <nil>
[db ] TRACE 2025/02/22 10:56:05 SELECT COALESCE(SUM(charged_kwh * price_per_kwh) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2025-01-23 10:56:05.406" AND charged_kwh > 0 AND price_per_kwh IS NOT NULL 1 <nil>
[db ] TRACE 2025/02/22 10:56:05 SELECT COALESCE(SUM(charged_kwh * co2_per_kwh) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2025-01-23 10:56:05.406" AND charged_kwh > 0 AND co2_per_kwh IS NOT NULL 1 <nil>
[db ] TRACE 2025/02/22 10:56:05 SELECT COALESCE(SUM(charged_kwh * solar_percentage) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-02-23 10:56:05.408" AND charged_kwh > 0 AND solar_percentage IS NOT NULL 1 <nil>
[db ] TRACE 2025/02/22 10:56:05 SELECT COALESCE(SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-02-23 10:56:05.408" AND charged_kwh > 0 AND solar_percentage IS NOT NULL 1 <nil>
[db ] TRACE 2025/02/22 10:56:05 SELECT COALESCE(SUM(charged_kwh * price_per_kwh) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-02-23 10:56:05.408" AND charged_kwh > 0 AND price_per_kwh IS NOT NULL 1 <nil>
[db ] TRACE 2025/02/22 10:56:05 SELECT COALESCE(SUM(charged_kwh * co2_per_kwh) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-02-23 10:56:05.408" AND charged_kwh > 0 AND co2_per_kwh IS NOT NULL 1 <nil>
[db ] TRACE 2025/02/22 10:56:05 SELECT COALESCE(SUM(charged_kwh * solar_percentage) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2025-01-01 00:00:00" AND charged_kwh > 0 AND solar_percentage IS NOT NULL 1 <nil>
[db ] TRACE 2025/02/22 10:56:05 SELECT COALESCE(SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2025-01-01 00:00:00" AND charged_kwh > 0 AND solar_percentage IS NOT NULL 1 <nil>
[db ] TRACE 2025/02/22 10:56:05 SELECT COALESCE(SUM(charged_kwh * price_per_kwh) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2025-01-01 00:00:00" AND charged_kwh > 0 AND price_per_kwh IS NOT NULL 1 <nil>
[db ] TRACE 2025/02/22 10:56:05 SELECT COALESCE(SUM(charged_kwh * co2_per_kwh) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2025-01-01 00:00:00" AND charged_kwh > 0 AND co2_per_kwh IS NOT NULL 1 <nil>
[db ] TRACE 2025/02/22 10:56:05 SELECT COALESCE(SUM(charged_kwh * solar_percentage) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "0000-00-00 00:00:00" AND charged_kwh > 0 AND solar_percentage IS NOT NULL 1 <nil>
[db ] TRACE 2025/02/22 10:56:05 SELECT COALESCE(SUM(charged_kwh), 0) FROM sessions WHERE finished >= "0000-00-00 00:00:00" AND charged_kwh > 0 AND solar_percentage IS NOT NULL 1 <nil>
[db ] TRACE 2025/02/22 10:56:05 SELECT COALESCE(SUM(charged_kwh * price_per_kwh) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "0000-00-00 00:00:00" AND charged_kwh > 0 AND price_per_kwh IS NOT NULL 1 <nil>
[db ] TRACE 2025/02/22 10:56:05 SELECT COALESCE(SUM(charged_kwh * co2_per_kwh) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "0000-00-00 00:00:00" AND charged_kwh > 0 AND co2_per_kwh IS NOT NULL 1 <nil>
```

andig, 2025-02-22:
Ähhh…. Was ist das?

Kugelfang666, 2025-02-22:
apparently not what you were looking for :-)

[image]

How do I place the query for an EVCC instance running as HomeAssitant Addon. I don't know how to access the console for that

mucki12, 2025-02-22:
Nur mal so als Schnellschuss:
Es gibt zwei LUNA Speicher, aber in der evcc.yaml ist nur eine konfiguriert? Hat der zweite LUNA nicht eine eigene ID? Was passiert, wenn man in der evcc.yaml den LUNA noch einmal mit eigenem Namen und id 2 (angenommen es ist id 2) anlegt?

premultiply, 2025-02-22:
Looks like the second battery storage unit is missing in your configuration.
Adding it should fix the problem.
See `storageunit: 2` option in advanced template configuration for this battery.

Remember to also assign this second battery meter to the `site` configuration.

closed 2025-02-22 by andig

Kugelfang666, 2025-02-22:
OH OK! I was not aware that there is an option to define the storage unit! 

I was assuming that the both batteries are automatically detected. Thanks a lot for the clarification! 

I've got it working now.