evcc-io/evcc #19071: residual power initialised with wrong value

hybe2022, 2025-02-22:
### Describe the bug

In the config under site residualPower is set to 50 while in the logs I see it is initialised with 300
Is this a bug or am I doing something wrong? 

### Steps to reproduce

1. Set residualPower to 100  or 50
2. Start evvc
3. Logs show in both cases a value of 300 is used instead 
...

### Configuration details

```yaml
site:
  title: Home
  meters:
    grid: grid1
    pv:
    - pv1
    - pv2
    battery:
    - battery1
  residualPower: 50
```

### Log details

```text
[main ] INFO 2025/02/22 17:42:27 evcc 0.200.1
[main ] INFO 2025/02/22 17:42:27 using config file: /etc/evcc.yaml
[db ] INFO 2025/02/22 17:42:27 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2025/02/22 17:42:28 listening at :7070
[circuit-charger] DEBUG 2025/02/22 17:42:33 validation of max power disabled
[lp-1 ] DEBUG 2025/02/22 17:42:33 charge total import: 4975.983kWh
[site ] INFO 2025/02/22 17:42:33 site config:
[site ] INFO 2025/02/22 17:42:33 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2025/02/22 17:42:33 grid: power ✓ energy ✓ currents ✓
[site ] INFO 2025/02/22 17:42:33 pv 1: power ✓ energy ✓ currents ✗
[site ] INFO 2025/02/22 17:42:33 pv 2: power ✓ energy ✓ currents ✗
[site ] INFO 2025/02/22 17:42:33 battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✓
[site ] INFO 2025/02/22 17:42:33 vehicles:
[site ] INFO 2025/02/22 17:42:33 vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site ] INFO 2025/02/22 17:42:33 vehicle 2: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site ] INFO 2025/02/22 17:42:33 vehicle 3: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1 ] INFO 2025/02/22 17:42:33 loadpoint 1:
[lp-1 ] INFO 2025/02/22 17:42:33 mode: pv
[lp-1 ] INFO 2025/02/22 17:42:33 charger: power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1 ] INFO 2025/02/22 17:42:33 meters: charge ✓
[lp-1 ] INFO 2025/02/22 17:42:33 charge: power ✓ energy ✓ currents ✓
[site ] DEBUG 2025/02/22 17:42:33 set priority soc: 70
[site ] DEBUG 2025/02/22 17:42:33 set battery discharge control: true
[site ] DEBUG 2025/02/22 17:42:33 set residual power: 300
[lp-1 ] DEBUG 2025/02/22 17:42:33 phase timer inactive
[lp-1 ] DEBUG 2025/02/22 17:42:33 pv timer inactive
[site ] WARN 2025/02/22 17:42:33 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site ] DEBUG 2025/02/22 17:42:33 ----
[lp-1 ] DEBUG 2025/02/22 17:42:33 charge power: 0W
[lp-1 ] DEBUG 2025/02/22 17:42:33 charge currents: [0 0 0]A
[circuit-charger] DEBUG 2025/02/22 17:42:33 power: 0W
[circuit-charger] DEBUG 2025/02/22 17:42:33 current: 0A
[circuit-main] DEBUG 2025/02/22 17:42:33 power: 8W
[circuit-main] DEBUG 2025/02/22 17:42:33 current: 1.66A
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

StefanSchoof, 2025-02-22:
Do you change the config at `/etc/evcc.yaml` or in your home directory?

hybe2022, 2025-02-22:
I do not change the /etc/evcc.yaml nor I do have any evcc.yaml in /home/user/.
The snippet above is from /etc/evcc.yaml.

VolkerK62, 2025-02-22:
Might be from the UI-configuration.
Check "control behavior".

hybe2022, 2025-02-22:
There is no option for site configuration in UI, nor I did a any configuration in UI up to now.

VolkerK62, 2025-02-22:
go to "configuration"
enable "experimental"
open in Area "General" the item "control behavior"

hybe2022, 2025-02-22:
Experimental was enabled before. Didn't know there is another setting behind as the time interval which is shown in the overview. So there I found the 300W. Thanks

Means residualPower setting from yaml is obsolete now as the UI setting which is hidden overwrites the yaml setting while the time interval is taken from yaml.
That is not consistent and should be changed/fixed.

andig, 2025-02-22:
/cc @naltatis any special handling needed to avoid this?

closed 2025-02-22 by andig

hybe2022, 2025-02-22:
For me it is not ok that for one item the prio is on UI setting while it es hidden and without enabling experimential even not shown while another setting (interval) has the prio in the yaml and is taken from there. 
It siehe should be consistent from where settings are taken and if something is taken from UI it should be visible in foreground.