evcc-io/evcc #19078: Fronius Symo Hybrid über Modbus incl. Batteriesteuerung

Wisley7, 2025-02-22:
Da die SolarAPI-V1 von Fronius nicht mit aktiven Batteriesteuerung funktioniert, und das Gen24-Template für den alten SymoHybrid ungeeignet ist, habe ich mich an eine Variante für die EVCC.yaml gemacht.

Ich hoffe dieses kann als festes Template integriert werden. Allerdings fehlt mir dazu das nötige knowhow.

Hier der Auszug:

meters:
  - name: grid1 # Fronius Datamanager
    type: "custom"
    power:
      source: "sunspec"
      uri: 192.168.178.50:502  # IP-Adresse deines Fronius-Wechselrichters
      id: 240  # Modbus-ID
      value:
        - 213:W  # register für Leistung (Watt)
    energy:
      source: "sunspec"
      uri: 192.168.178.50:502
      id: 240
      value:
        - 213:TotWhImp
      scale: 0.001
    currents:
      - source: "sunspec"
        uri: 192.168.178.50:502
        id: 240
        value:
          - 213:AphA
      - source: "sunspec"
        uri: 192.168.178.50:502
        id: 240
        value:
          - 213:AphB
      - source: "sunspec"
        uri: 192.168.178.50:502
        id: 240
        value:
          - 213:AphC
  - name: symohybrid
    type: "custom"
    power:
      source: "sunspec"
      uri: 192.168.178.50:502  # IP-Adresse deines Fronius-Wechselrichters
      id: 1  # Modbus-ID
      value: 160:1:DCW  # register für PV-Leistung (Watt)
  - name: battery
    type: "custom"
    power: # Da nur ein Register für Be-/Entladen besteht, musste über ChaSt (3-Entladen/4-Laden) eine Umrechnung stattfinden
      source: "calc"
      mul:
        - source: "calc"
          sign:
            source: "calc"
            add:
              - source: "modbus"
                uri: 192.168.178.50:502
                id: 1
                register:
                  address: 40324  # Registeradresse für ChaSt
                  type: holding  # Typ des Registers
                  decode: int16  # Interpretation als 16-Bit-Ganzzahl
                scale: -1  # Skalierung, um 3 in -3 und 4 in -4 umzuwandeln
              - source: "const"
                value: 3.5  # Offset, um -3 in 0.5 und -4 in -0.5 umzuwandeln
        - source: "sunspec"
          uri: 192.168.178.50:502
          id: 1
          value: 160:2:DCW  # Register für Batterieleistung
          scale: 1
    soc:
      source: "sunspec"
      uri: 192.168.178.50:502
      id: 1
      value: 124:0:ChaState  # Beispielregister für Batterie-Ladezustand (SOC)
    batterymode:
      source: "switch"
      switch:
        - case: 1  # Normalmodus
          set:
            source: "sequence"
            set:
              - source: "const"
                value: 0
                set:
                  source: "sunspec"
                  uri: 192.168.178.50:502
                  id: 1
                  value: 124:0:StorCtl_Mod
              - source: "const"
                value: 100  # %
                set:
                  source: "sunspec"
                  uri: 192.168.178.50:502
                  id: 1
                  value: 124:0:OutWRte
        - case: 2  # Holdmodus
          set:
            source: "sequence"
            set:
              - source: "const"
                value: 2
                set:
                  source: "sunspec"
                  uri: 192.168.178.50:502
                  id: 1
                  value: 124:0:StorCtl_Mod
              - source: "const"
                value: 0  # %
                set:
                  source: "sunspec"
                  uri: 192.168.178.50:502
                  id: 1
                  value: 124:0:OutWRte
        - case: 3  # Ladmodus
          set:
            source: "sequence"
            set:
              - source: "const"
                value: 1  # Off
                set:
                  source: "sunspec"
                  uri: 192.168.178.50:502
                  id: 1
                  value: 124:0:ChaGriSet
              - source: "const"
                value: 2
                set:
                  source: "sunspec"
                  uri: 192.168.178.50:502
                  id: 1
                  value: 124:0:StorCtl_Mod
              - source: "const"
                value: -50  # Maximale Laderate in % (z. B. 50 %)
                set:
                  source: "sunspec"
                  uri: 192.168.178.50:502
                  id: 1
                  value: 124:0:OutWRte

andig, 2025-02-22:
Könntest du das bitte als code formatieren damit die Einrückungen erkennbar aind? Vielen Dank!

Wisley7, 2025-02-22:
Leider funktionieren die vorhandenen Sunspec Templates nicht, da Register angefragt werden, welche nicht vorhnden sind. 
Ein reines Auslesen der Wechselrichter hat funktioniert, allerdings nicht als Batterie incl. Batteriesteuerung.

```
meters:
  - name: grid1
    type: "custom"
    power:
      source: "sunspec"
      uri: 192.168.178.50:502  # IP-Adresse deines Fronius-Wechselrichters
      id: 240  # Modbus-ID
      value:
        - 213:W  # Beispielregister für Leistung (Watt)
    energy:
      source: "sunspec"
      uri: 192.168.178.50:502
      id: 240
      value:
        - 213:TotWhImp
      scale: 0.001
    currents:
      - source: "sunspec"
        uri: 192.168.178.50:502
        id: 240
        value:
          - 213:AphA
      - source: "sunspec"
        uri: 192.168.178.50:502
        id: 240
        value:
          - 213:AphB
      - source: "sunspec"
        uri: 192.168.178.50:502
        id: 240
        value:
          - 213:AphC
  - name: symohybrid
    type: "custom"
    power:
      source: "sunspec"
      uri: 192.168.178.50:502  # IP-Adresse deines Fronius-Wechselrichters
      id: 1  # Modbus-ID
      value: 160:1:DCW  # Beispielregister für Leistung (Watt)
  - name: battery
    type: "custom"
    power:
      source: "calc"
      mul:
        - source: "calc"
          sign:
            source: "calc"
            add:
              - source: "modbus"
                uri: 192.168.178.50:502
                id: 1
                register:
                  address: 40324  # Registeradresse für ChaSt
                  type: holding  # Typ des Registers
                  decode: int16  # Interpretation als 16-Bit-Ganzzahl
                scale: -1  # Skalierung, um 3 in -3 und 4 in -4 umzuwandeln
              - source: "const"
                value: 3.5  # Offset, um -3 in 0.5 und -4 in -0.5 umzuwandeln
        - source: "sunspec"
          uri: 192.168.178.50:502
          id: 1
          value: 160:2:DCW  # Beispielregister für Batterieleistung
          scale: 1
    soc:
      source: "sunspec"
      uri: 192.168.178.50:502
      id: 1
      value: 124:0:ChaState  # Beispielregister für Batterie-Ladezustand (SOC)
    batterymode:
      source: "switch"
      switch:
        - case: 1  # Normalmodus
          set:
            source: "sequence"
            set:
              - source: "const"
                value: 0
                set:
                  source: "sunspec"
                  uri: 192.168.178.50:502
                  id: 1
                  value: 124:0:StorCtl_Mod
              - source: "const"
                value: 100  # %
                set:
                  source: "sunspec"
                  uri: 192.168.178.50:502
                  id: 1
                  value: 124:0:OutWRte
        - case: 2  # Holdmodus
          set:
            source: "sequence"
            set:
              - source: "const"
                value: 2
                set:
                  source: "sunspec"
                  uri: 192.168.178.50:502
                  id: 1
                  value: 124:0:StorCtl_Mod
              - source: "const"
                value: 0  # %
                set:
                  source: "sunspec"
                  uri: 192.168.178.50:502
                  id: 1
                  value: 124:0:OutWRte
        - case: 3  # Ladmodus
          set:
            source: "sequence"
            set:
              - source: "const"
                value: 1  # Off
                set:
                  source: "sunspec"
                  uri: 192.168.178.50:502
                  id: 1
                  value: 124:0:ChaGriSet
              - source: "const"
                value: 2
                set:
                  source: "sunspec"
                  uri: 192.168.178.50:502
                  id: 1
                  value: 124:0:StorCtl_Mod
              - source: "const"
                value: -50  # Maximale Laderate in % (z. B. 50 %)
                set:
                  source: "sunspec"
                  uri: 192.168.178.50:502
                  id: 1
                  value: 124:0:OutWRte
```

andig, 2025-02-22:
> Leider funktionieren die vorhandenen Sunspec Templates nicht, da Register angefragt werden, welche nicht vorhnden sind.

Ich kann die nicht folgen- du speichst doch ausschliesslich Standardregister an? Mit welchem Sunspec Template hattest du das verglichen und was ist da konkret falsch? Es wäre schön wenn wir nicht doppelt pflegen müssten sondern im Zweifel das sunspec Template korrigieren könnten.

Wisley7, 2025-02-22:
sunspec-battery-control124 spricht den Register [{160 3 DCW}] und [{160 4 DCW}] an, welcher bei SymoHybrid nicht vorhanden sind.
der SymoHybrid hat nur 2 mppt kanäle, einer für PV einer für Batterie, dieser gibt auch leider keine negativen Werte Raus, deshalb der umweg über ChaSt / 40324

sunspec-hybrid hat das selbe Problem
sunspec-wechselrichter funktioniert zwar, aber rein als Auslesen keine Batteriekontrolle integriert.

Beleuchte mich, falls ich in der vergangenheit zu doof war, und das mit vorhandenen Mitteln geht.

premultiply, 2025-02-22:
Was sagt denn `evcc sunspec` zu dem WR?

Wisley7, 2025-02-22:
[main  ] INFO 2025/02/22 23:18:36 evcc 0.200.2
--------- Model 1 common ---------
Mn            Fronius    string
Md   Symo Hybrid 5.0-    string
Opt          1.31.1-5    string
Vr           0.3.30.0    string
SN   3020100108110004    string
DA               1.00    uint16
Pad            0x0071    pad
--------- Model 113 inverter ---------
A                    1.26    float32
AphA                 0.42    float32
AphB                 0.42    float32
AphC                 0.42    float32
PPVphAB             405.2    float32
PPVphBC             407.2    float32
PPVphCA             403.8    float32
PhVphA              233.4    float32
PhVphB                235    float32
PhVphC              233.4    float32
W                     297    float32
Hz                  49.97    float32
VA                    297    float32
VAr                     0    float32
PF                    100    float32
WH          4.0061108e+07    float32
DCA                   n/a    float32
DCV                   n/a    float32
DCW                304.46    float32
TmpCab                n/a    float32
TmpSnk                n/a    float32
TmpTrns               n/a    float32
TmpOt                 n/a    float32
St                      4    enum16
StVnd                   4    enum16
Evt1           0x00000000    bitfield32
Evt2           0x00000000    bitfield32
EvtVnd1        0x00000000    bitfield32
EvtVnd2        0x00000000    bitfield32
EvtVnd3        0x00000000    bitfield32
EvtVnd4        0x00000000    bitfield32
--------- Model 120 nameplate ---------
DERTyp                          4    enum16
WRtg                      5000.00    uint16
WRtg_SF                         1    sunssf
VARtg                     5000.00    uint16
VARtg_SF                        1    sunssf
VArRtgQ1                  5000.00    int16
VArRtgQ2                      n/a    int16
VArRtgQ3                      n/a    int16
VArRtgQ4                 -5000.00    int16
VArRtg_SF                       1    sunssf
ARtg                         7.20    uint16
ARtg_SF                        -2    sunssf
PFRtgQ1                     -0.80    int16
PFRtgQ2                       n/a    int16
PFRtgQ3                       n/a    int16
PFRtgQ4                      0.80    int16
PFRtg_SF                       -3    sunssf
WHRtg                    11520.00    uint16
WHRtg_SF                        0    sunssf
AhrRtg                        n/a    uint16
AhrRtg_SF                     n/a    sunssf
MaxChaRte                11520.00    uint16
MaxChaRte_SF                    0    sunssf
MaxDisChaRte             11520.00    uint16
MaxDisChaRte_SF                 0    sunssf
Pad                        0x8000    pad
--------- Model 121 settings ---------
WMax                   5000.00    uint16
[... 106 lines cut ...]
WChaMax                    11520.00    uint16
WChaGra                      100.00    uint16
WDisChaGra                   100.00    uint16
StorCtl_Mod                  0x0000    bitfield16
VAChaMax                        n/a    uint16
MinRsvPct                      0.00    uint16
ChaState                      31.70    uint16
StorAval                        n/a    uint16
InBatV                          n/a    uint16
ChaSt                             3    enum16
OutWRte                      100.00    int16
InWRte                       100.00    int16
InOutWRte_WinTms                n/a    uint16
InOutWRte_RvrtTms               n/a    uint16
InOutWRte_RmpTms                n/a    uint16
ChaGriSet                         1    enum16
WChaMax_SF                        0    sunssf
WChaDisChaGra_SF                  0    sunssf
VAChaMax_SF                     n/a    sunssf
MinRsvPct_SF                     -2    sunssf
ChaState_SF                      -2    sunssf
StorAval_SF                     n/a    sunssf
InBatV_SF                       n/a    sunssf
InOutWRte_SF                     -2    sunssf
root@EVCC-TEST:~#

Wisley7, 2025-03-03:
Kann ich hier noch was zuarbeiten, oder wird hier nix mehr in Betracht gezogen?

andig, 2025-03-03:
> Da die SolarAPI-V1 von Fronius nicht mit aktiven Batteriesteuerung funktioniert, und das Gen24-Template für den alten SymoHybrid ungeeignet ist, habe ich mich an eine Variante für die EVCC.yaml gemacht.

Es gibt ja auch ein `sunspec-battery-control` Template. Warum nicht das nehmen? Ich hab leider immer noch nicht verstanden, was hier konkret neu oder anders ist.

MoJo2600, 2025-03-03:
> Es gibt ja auch ein `sunspec-battery-control` Template. Warum nicht das nehmen? Ich hab leider immer noch nicht verstanden, was hier konkret neu oder anders ist.

Das `sunspec-battery-control` funktioniert in dieser bestimmten Kombination aus Wechselrichter und Batterie nicht. Bei mir sind z.B. die Werte für Grid und Batterie die selben und charge/discharging falsch herum #18330. Mit anderen Templates bekomme ich Fehlermeldungen. Ich habe das irgendwo einmal komplett zusammengeschrieben, aber ich finde es gerade nicht. Aber meine Debug infos sind in ticket #18330.

Wisley7, 2025-03-03:
Wie schon mehrfach beschrieben,
Das Template „  sunspec-battery-control“ spricht nicht die richtigen Register an, welche der SymoHybrid hat. 160-3 und160-4 gibt es nicht. 
Und 160-2 ist für die Batterie zuständig. Gibt aber keinen negativen Wert raus.

andig, 2025-03-03:
Ahh, das hatte ich überlesen. Das ist ja für die Anwender nicht offensichtlich. Wäre es da nicht schick, die Templates aufzuräumen? Aus -hybrid und -battery-control müsste jeweils eine -2 und -4 Option werden je nach Anzahl der Strings. Wenn wir es elegant halten wollen könnte man dann gleich `ChaSt` nutzen.

Alternativ: wir fügen das hier als (weiteren) Workaround hinzu. Dafür muss es bitte aufgeräumt werden (Register Namen statt ids) und ein PR erstellt.

andig, 2025-03-03:
Freiwillige vor :)

Wisley7, 2025-03-04:
Mit dem Register ChaSt gibt es noch ein Problem im leeren Zustand. Da gibt der WR einen anderen Wert raus als beladen/entladen. Und hierdurch wird Funktion die Funktion nicht wie gedacht. Evcc denkt in dem Moment, dass die Leistung, welche in den Akku geht , aus dem Akku raus geht. 
Gibt es die Möglichleit eine Wenn dann Funktion zu nutzen. Sprich wenn ChaSt 2 dann multipliziere mit -1 ? Dann könnte ich das Problem nochmal angehen

andig, 2025-03-04:
Was wäre die komplette Logik die du da brauchst?

Wisley7, 2025-03-04:
Ich beschreibe mal das ganze Problem, denn vll gibt es eine einfachere Lösung als meine, denn ich bin kein Programierer und bin nur durch Logik und viel Probieren zu der jetzigen Lösung gekommen.
der MPPT Kanal (160:2:DCW) der Batterie gibt immer ein positiven Wert aus, egal ob charging oder discharging.
der Kanal ChaSt (124:ChaSt) gibt Folgendes raus: 
charging : 4
discharging: 3
empty: 2 (dieser Wert ist aber auch bei charging so lange gesetzt, bis der Akku so voll geladen ist, bis er theoretisch wieder discharged werden kann / ungefähr von 5 - 8%)

als idee:

wenn ChaSt ist 2 oder 4 dann lese (160:2:DCW) als negativen Wert sonst als positiven Wert

Wisley7, 2025-04-21:
> Hab eine bessere Möglichkeit gefunden, die Umrechnung der Batterieleistung zu ermöglichen. Hierbei wird die PV Leistung von der WechselrichterLeistung abgezogen, was übrig bleibt, ist die Leistung der Batterie. Bis jetzt zufriedenstellende Ergebnisse, die nächsten Tage teste ich noch den Modus Laden aus Netz, sowie Kalibrierladung und das Problem beim Aufladen bis 8%. Wenn das so funktioniert, könnte das komplett als allg. Template für Hybrid WR mit nur einem MPPT genutzt werden.
> 
> ```
> - name: battery
>   type: "custom"
>   power:
>     source: "calc"
>     add:
>     -  source: "sunspec"
>        uri: 192.168.178.50:502
>        id: 1
>        value: 113:DCW # Wechselrichterleistung
>        scale: 1
>     -  source: "sunspec"
>        uri: 192.168.178.50:502
>        id: 1
>        value: 160:1:DCW  # PV-Leistung
>        scale: -1
> ```

Hat alles funktioniert…

MoJo2600, 2025-04-23:
Ich habe das auch ausprobiert, aber bei mir es so, dass ich sobald ich etwas mit ModBus TCP mache, der Wechselrichter laufend neu startet. Hast Du das Problem auch? Ich habe jetzt umgestellt auf Web und da Problem ist verschwunden.

Wisley7, 2025-04-23:
Ein Neustart des WR kommt in den meisten Fällen dann zustande, wenn zu viele Anfragen/Befehle an den WR gehen.
Wie ist der Intervall von evcc? Gibt es andere Software die auch über modbus abfragt?

MoJo2600, 2025-04-23:
Ich habe jetzt einmal testweise alles auf Solar Net API umgestellt, also alles über das Webinterface. Sobald ich auch nur die Batterie über Modbus TCP mit dem WR sprechen lasse, bekomme ich die Abstürze... Das ist echt komisch. Wahrscheinlich stört der Abruf der Daten eine interne Kommunkation vom WR und der Batterie und oder dem Smartmeter?

andig, 2025-04-23:
Ticket bei Fronuis aufmachen? Absturz ist ein Mangel...

MoJo2600, 2025-04-23:
Habe ich schon, geht bei Fronius nur über Elektriker. Aussage ist "Etwas sört die Kommunikation" und Gerät ist ausserhalb der Garantie... Aber wenn nur ich das Problem habe, dann hat er an der Stelle wirklich eine Macke, leider.

closed 2025-05-05 by github-actions[bot]