evcc-io/evcc #19092: Sponsor token outdated

chrstmnn, 2025-02-23:
### Describe the bug

The given sponsor token is outdated and not valid from today on. Could you please update it as I am currently setting evcc up. Thank you :)

### Steps to reproduce

1. Visit https://docs.evcc.io/docs/sponsorship
2. Check trial token valid date
3.
...

### Configuration details

```yaml
n/a
```

### Log details

```text
n/a
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

naltatis, 2025-02-23:
Ist aktualisiert.

closed 2025-02-23 by naltatis

juwild83, 2025-02-23:
Ich bekomme trotzdem diese Meldung, auch mit den neuen Token:

Fehler beim Starten. Überprüfe deine Konfiguration und starte neu.
sponsortoken: rpc error: code = Unknown desc = token has invalid claims: token is expire

naltatis, 2025-02-23:
@juwild83 kann es sein, dass du das Token mal über config UI eingetragen hast und jetzt bei evcc.yaml probierst? In Config UI eingetragene Werte haben immer Vorrang.

juwild83, 2025-02-23:
Ja das wars...
Ich hatte zwar über das UI noch keinen Token eingegeben aber es mal gestartet. Jetzt habe ich den Token über die UI eingegeben und es funktioniert, danke.