evcc-io/evcc #19101: cannot create vehicle type 'template': cannot create vehicle type 'mercedes': token expired

comessoon, 2025-02-23:
### Describe the bug

Hallo Ich habe gestern Das Update auf Version 0.200.2 upgedatet seite bekommen ich Keine Verbindung mehr zu Unserem EQC Mit Version 0.200.1 hat ich das Problem noch nicht.
Verhalten:

[image]

[mercedes] DEBUG 2025/02/23 12:40:47 identity.NewIdentity - database token found
[mercedes] DEBUG 2025/02/23 12:40:47 identity.NewIdentity - refreshToken started
[mercedes] TRACE 2025/02/23 12:40:47 POST https://id.mercedes-benz.com/as/token.oauth2
[mercedes] TRACE 2025/02/23 12:40:47 grant_type=refresh_token&refresh_token=RdfGcX8bQeqvN8vjlvKg4cKYlivr28bmJlS.jW3VYu -- {"error_description":"unknown, invalid, or expired refresh token","error":"invalid_grant"}

### Steps to reproduce

Nach dem Update wurde mir Unser Mercedes als Offline Angezeigt. worauf ich ihn einmal entfernt und danach mit neu generierten Tokens wieder Hinzufügte habe. Jetzt scheint er aber nicht die neuen Tokens zu nutzen den in de Logs Steht noch das alte token.
 

### Configuration details

```yaml
Konfiguration in Datenbank
```

### Log details

```text
[mercedes] DEBUG 2025/02/23 12:40:47 identity.NewIdentity - database token found
[mercedes] DEBUG 2025/02/23 12:40:47 identity.NewIdentity - refreshToken started
[mercedes] TRACE 2025/02/23 12:40:47 POST https://id.mercedes-benz.com/as/token.oauth2
[mercedes] TRACE 2025/02/23 12:40:47 grant_type=refresh_token&refresh_token=RdfGcX8bQeqvN8vjlvKg4cKYlivr28bmJlS.jW3VYu -- {"error_description":"unknown, invalid, or expired refresh token","error":"invalid_grant"}
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.2

comessoon, 2025-02-23:
Auch ein wieder eintragen in die YAML sorgt dafür das Immer noch laut Log das Alte AccessToken genutzt wird.

- name: EQC
  type: template
  template: mercedes
  user: c*********@gmail.com # Mercedes Me Nutzer-Id (email)
  region: EMEA # MB me Region (EMEA, APAC, NORAM)
  vin: W1**************
  capacity: 80 # AkkukapaziÃ¤t in kWh (optional)    
  identifiers: 
    - 04F08EE28A6B84
  accessToken: 
  refreshToken: XS5zzlXKT47rZq9999999999999

Suchen

TRACE
mercedes
[mercedes] DEBUG 2025/02/23 12:53:45 identity.NewIdentity - database token found
[mercedes] DEBUG 2025/02/23 12:53:45 identity.NewIdentity - refreshToken started
[mercedes] TRACE 2025/02/23 12:53:45 POST https://id.mercedes-benz.com/as/token.oauth2
[mercedes] TRACE 2025/02/23 12:53:45 grant_type=refresh_token&refresh_token=RdfGcX8bQeqvN8vjlvKg4cKYlivr28bmJlS.jW3VYu -- {"error_description":"unknown, invalid, or expired refresh token","error":"invalid_grant"}

AccessToken Wurde Absichtlich entfernt und das RefreshToken wurde in den letzte stellen abgeändert.

andig, 2025-02-23:
/cc @ReneNulschDE ich dachte das Problem hätten wir gelöst?

ReneNulschDE, 2025-02-23:
Nein, es immer noch der alte token aus der DB gelöscht werden. Was wir gefixt haben: Die MB API liefert keinen neuen Refreshtoken mehr wenn ein refresh token erneuert wird. (Also kein rolling mehr)...

comessoon, 2025-02-23:
> Nein, es muss immer noch der alte token aus der DB gelöscht werden. Was wir gefixt haben: Die MB API liefert keinen neuen Refreshtoken mehr wenn ein refresh token erneuert wird. (Also kein rolling mehr)...

@ReneNulschDE  Wir lösche ich das Token raus?

ReneNulschDE, 2025-02-23:
@comessoon : Du postest das in mehreren issues/pr. Ich habe [hier](https://github.com/evcc-io/evcc/pull/18893#issuecomment-2676968203) geantwortet.