evcc-io/evcc #19106: smartfox-em2  read: connection reset by peer when using for usage: grid and pv concurrently

mdecker77, 2025-02-23:
### Describe the bug

I am testing evcc with smartfox pro (EM2) and found the same issue as described here: https://github.com/evcc-io/evcc/discussions/17085

[lp-1 ] DEBUG 2025/02/23 14:06:25 charge power: 0W
[lp-1 ] DEBUG 2025/02/23 14:06:25 charge currents: [0 0 0]A
[site ] DEBUG 2025/02/23 14:06:25 grid power: -3658W
[site ] DEBUG 2025/02/23 14:06:25 pv 1 power: 4180W
[site ] ERROR 2025/02/23 14:06:25 pv 1 energy: Get "http://192.168.251.4/values.xml": read tcp 192.168.251.3:59268->192.168.251.4:80: read: connection reset by peer
[site ] DEBUG 2025/02/23 14:06:25 grid powers: [-1244 -1276 -1138]W
[site ] DEBUG 2025/02/23 14:06:25 grid currents: [-5.24 -5.45 -4.89]A
[site ] DEBUG 2025/02/23 14:06:25 site power: -3658W
[lp-1 ] DEBUG 2025/02/23 14:06:25 charge total import: 7798.839kWh
[lp-1 ] DEBUG 2025/02/23 14:06:25 charger status: A 

In tcpdump, i can see that evcc is performing one HTTP request per metric resulting in 8 requests per second for grid metric alone. When smartfox-em template is also used for "pv" and "aux", then I can see these "connection reset by peer" messages because smartfox is sending RST (Reset) Flag in TCP packet.

Unfortunately, smartfox pro em2 can only provide <uri>/values.xml containing ALL metrics in one XML response. In my opinion, it would be smarter to extract all the required data from this XML string in one go, instead of requesting the same URL dozens of times and only extract one metric at a time. 

If that does not work, then evcc should provide a way for specifying a wait time before the next http request is sent.

### Steps to reproduce

1. enable smartfox-em2 template for grid and pv (and optionally for aux)
2. wait 2 minutes

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 30s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: smartfox-em2
  usage: grid
  host: 192.168.251.4
  name: grid1
- type: template
  template: smartfox-em2
  usage: pv
  host: 192.168.251.4
  name: pv1
#- type: template
#  template: smartfox-em2
#  usage: aux
#  host: 192.168.251.4
#  name: aux1

chargers:
#- type: template
#  template: keba-modbus
#  id: 255
#  host: 192.168.251.208
#  port: 502
#  modbus: tcpip
#  welcomecharge: false
#  name: wallbox3
- type: template
  template: go-e-v3
  host: 192.168.251.15
  name: go-e

loadpoints:
- title: Carport
  charger: go-e
  mode: off

site:
  title: My home
  meters:
    grid: grid1
    pv: pv1
#    aux: aux1

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.22 # [currency]/kWh
    zones:
      - days: Mo-So
        hours: 0-6
        price: 0.16 # EUR/kWh
      - days: Mo-So
        hours: 22-0
        price: 0.16 # EUR/kWh
      - days: Sa
        hours: 13-22
        price: 0.16 # EUR/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.10 # [currency]/kWh

Version: `0.200.2`
```

### Log details

```text
[main  ] INFO 2025/02/23 14:12:19 evcc 0.200.2
[main  ] INFO 2025/02/23 14:12:19 using config file: /root/evcc/evcc.yaml
[db    ] INFO 2025/02/23 14:12:19 using sqlite database: /root/.evcc/evcc.db
[main  ] INFO 2025/02/23 14:12:19 listening at :7070
[lp-1  ] DEBUG 2025/02/23 14:12:19 charge total import: 7798.839kWh
[site  ] INFO 2025/02/23 14:12:19 site config:
[site  ] INFO 2025/02/23 14:12:19   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2025/02/23 14:12:19     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2025/02/23 14:12:19     pv 1:      power ✓ energy ✓ currents ✗
[lp-1  ] INFO 2025/02/23 14:12:19 loadpoint 1:
[lp-1  ] INFO 2025/02/23 14:12:19   mode:        off
[lp-1  ] INFO 2025/02/23 14:12:19   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2025/02/23 14:12:19   meters:      charge ✓
[lp-1  ] INFO 2025/02/23 14:12:19     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2025/02/23 14:12:19 phase timer inactive
[lp-1  ] DEBUG 2025/02/23 14:12:19 pv timer inactive
[site  ] DEBUG 2025/02/23 14:12:19 ----
[lp-1  ] DEBUG 2025/02/23 14:12:19 charge power: 0W
[lp-1  ] DEBUG 2025/02/23 14:12:19 charge currents: [0 0 0]A
[site  ] DEBUG 2025/02/23 14:12:19 grid power: -3932W
[site  ] DEBUG 2025/02/23 14:12:19 grid powers: [-1360 -1318 -1254]W
[site  ] ERROR 2025/02/23 14:12:19 grid currents: Get "http://192.168.251.4/values.xml": read tcp 192.168.251.3:41740->192.168.251.4:80: read: connection reset by peer
[site  ] ERROR 2025/02/23 14:12:19 grid energy: Get "http://192.168.251.4/values.xml": read tcp 192.168.251.3:41752->192.168.251.4:80: read: connection reset by peer
[site  ] DEBUG 2025/02/23 14:12:19 pv 1 power: 4500W
[site  ] DEBUG 2025/02/23 14:12:19 aux 1 power: 0W
[site  ] DEBUG 2025/02/23 14:12:20 site power: -3932W
[lp-1  ] DEBUG 2025/02/23 14:12:20 charge total import: 7798.839kWh
[lp-1  ] DEBUG 2025/02/23 14:12:20 charger status: A
[lp-1  ] INFO 2025/02/23 14:12:20 car disconnected
[lp-1  ] DEBUG 2025/02/23 14:12:20 set battery boost: false
[lp-1  ] DEBUG 2025/02/23 14:12:20 set session soc limit: 0
[lp-1  ] DEBUG 2025/02/23 14:12:20 set session energy limit: 0
[site  ] DEBUG 2025/02/23 14:12:50 ----
[lp-1  ] DEBUG 2025/02/23 14:12:50 charge power: 0W
[lp-1  ] DEBUG 2025/02/23 14:12:50 charge currents: [0 0 0]A
[site  ] DEBUG 2025/02/23 14:12:50 grid power: -3959W
[site  ] DEBUG 2025/02/23 14:12:50 grid powers: [-1368 -1327 -1264]W
[site  ] DEBUG 2025/02/23 14:12:50 grid currents: [-5.74 -5.67 -5.38]A
[site  ] DEBUG 2025/02/23 14:12:51 pv 1 power: 4530W
[site  ] DEBUG 2025/02/23 14:12:51 aux 1 power: 0W
[site  ] DEBUG 2025/02/23 14:12:51 site power: -3959W
[lp-1  ] DEBUG 2025/02/23 14:12:51 charge total import: 7798.839kWh
[lp-1  ] DEBUG 2025/02/23 14:12:51 charger status: A
[site  ] DEBUG 2025/02/23 14:13:20 ----
[lp-1  ] DEBUG 2025/02/23 14:13:20 charge power: 0W
[lp-1  ] DEBUG 2025/02/23 14:13:20 charge currents: [0 0 0]A
[site  ] DEBUG 2025/02/23 14:13:20 grid power: -4003W
[site  ] DEBUG 2025/02/23 14:13:20 grid powers: [-1390 -1348 -1265]W
[site  ] ERROR 2025/02/23 14:13:20 grid currents: Get "http://192.168.251.4/values.xml": read tcp 192.168.251.3:52214->192.168.251.4:80: read: connection reset by peer
[site  ] DEBUG 2025/02/23 14:13:20 aux 1 power: 0W
[site  ] DEBUG 2025/02/23 14:13:21 pv 1 power: 4540W
[site  ] DEBUG 2025/02/23 14:13:21 site power: -4003W
[lp-1  ] DEBUG 2025/02/23 14:13:21 charge total import: 7798.839kWh
[lp-1  ] DEBUG 2025/02/23 14:13:21 charger status: A
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.2

mdecker77, 2025-02-23:
Screenshots from wireshark.

[image]
[image]

andig, 2025-02-23:
> Screenshots from wireshark.

What do they  tell us?

mdecker77, 2025-02-23:
It is showing that SMARTFOX device s sending reset TCP packet to terminate
the connection. My interpretation is that the device can not handle the
high number of requests per seconds.

-- 

*DI (FH) Martin Decker*
Oracle 10g/11g/12c Certified Master
Red Hat Certified Engineer (RHCE)

*ora-solutions GmbH*
Fälle 20
A-6822 Satteins
AUSTRIA
T: +43 (0) 699 1131 6463
***@***.***
http://www.ora-solutions.net
LG Feldkirch / FN 612112 s

closed 2025-02-23 by andig