evcc-io/evcc #19107: Support for Belgian Dynamic Energy Tariffs (Frank Energie)

geysent, 2025-02-23:
Hello,

I am currently using EVCC to charge my EV with my solar energy and am now considering a dynamic tariff. This would help me charge my car at the cheapest times during winter.

However, I noticed that no Belgian energy providers are currently supported in EVCC. Would it be possible to add support for Frank Energie (https://www.frankenergie.be/)? I can already read the tariffs via Home Assistant, so the data is available.

Thank you for your great work, and I hope this addition can be considered!

Thijs

andig, 2025-02-23:
Does it have an api?

RenatusRo, 2025-02-23:
They don't.

@geysent: you need to take the tarifkaart from 
https://www.frankenergie.be/nl/voorwaarden
Calculate all the heffingen and some estimation for the vaste vergoedingen en capaciteitstarief and use their formula.

And then configure it in evcc following this example for trevion.be
(get yourself an entsoe token):
(the feedin looks somewhat awkward: you recover the market price from grid: and apply the formule terugleveringsvergoeding)

```
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    type: entsoe # Entso-E european market data
    domain: BZN|BE # https://transparency.entsoe.eu/content/static_content/Static%20content/web%20api/Guide.html#_areas
    securitytoken: 95e3dfaa-.....
    formula: ((price * 1.07 + 0.0013) + 0.15858) * 1.21
  feedin:
    # see https://github.com/evcc-io/evcc/discussions/16465
    type: custom
    price:
      source: http
      uri: http://localhost:7070/api/state
      jq: (((.result.tariffGrid/1.21 - 0.15858-0.0013)/1.07) * 0.86) - 0.005

```