evcc-io/evcc #19128: evcc/loadpoints/<id>/phases: enabled phases (writable)

aichingerk, 2025-02-23:
### Describe the bug

Hallo,
Wann wurde dieser MQTT Eintrag evcc/loadpoints/<id>/phases: enabled phases (writable) raus genommen ?
Oder wie kann man nun über mqtt diesen wert wieder auslesen ?
Möchte manuell umschalten können, damit bei min+pv bei Automatic Stellung von 1p3 auch eine Unterscheidung gemacht werden kann. Denn wenn der letzte Wert auf 3P war nimmt er für min+pv 3 Phasig und das möchte ich nicht und/oder manuell umschalten auf 1 Phasig bei min+pv

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
nicht relevant
```

### Log details

```text
nicht relevant
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-02-24:
Siehe Doku- da hat sich nichts geändert.

closed 2025-02-24 by andig

aichingerk, 2025-02-24:
Ja, ich habe die Doku gelesen, früher konnte man hier auch den Wert lesen der für die Phasen verantwortlich war wie in der Doku:
evcc/loadpoints/<id>/phases: enabled phases (writable)
enabled phases eben, das schreiben über diesen Parameter funktioniert noch. (0,1,3)
Man sieht auch laut MQTT Explorer diesen Wert nicht mehr:

[image]

Das einzige was den alten Parameter anbelangt ist phasesActive nun, darum wollte ich nachfragen ob ihr die Struktur geändert habt.

andig, 2025-02-24:
/cc @Maschga wärst Du evtl. so nett und würdest die Doku nachziehen?

Maschga, 2025-02-24:
Ja, gerne.
Bin derzeit dabei die gesamte mqtt-Dokumentation vollständig zu dokumentieren: https://github.com/evcc-io/docs/pull/746

Das dauert aber noch an.