evcc-io/evcc #19130: Charging Loss / Efficiency

djfanatix, 2025-02-24:
**Is your feature request related to a problem? Please describe.**

Feature request (long term): Maybe we could implement the calculation of charging losses/ efficienty during charging.
We have the charged energy (kWh) from the loadpoint and from the Vehicle (API), so we can calculate.
We could make this visual in the UI or charging overview.

andig, 2025-02-24:
Maybe, but not a priority. For time being we're focussing on charging and energy management. Noted, but let's please close for time being.

naltatis, 2025-02-24:
A prerequisite would be to also have SoC added to sessions. We're planning to do that that but it's not trivial to get this right due to the asynchronous and not always table nature of getting SoC from the vehicle.

This is not a real duplicate, more of an extension. But I'll link it to the existing issue.

closed 2025-02-24 by naltatis