evcc-io/evcc #19136: Forecast screen diagram hover Information

puntipunti, 2025-02-24:
### Describe the bug

Nice diagram. dthe mouseover-hover information is missalligned 

[image]

### Steps to reproduce

1. configure solar forecast 
2. open forecast diagram
3. move mouse over the diagram
4. the green point showin the value 2at mouse-position is not alligned. maybe since the X-Axis has for today and tomorrow different scales?
3. the time marks at the x-axis is not shown correct for today
...

### Configuration details

```yaml
currency: EUR
grid:
  type: template
  template: smartenergy
  #type: awattar
  #region: at # optional, choose at for Austria
  charges: 0.11 # optional, additional charges per kWh
  tax: 0.2 # optional, additional tax (0.1 for 10%)
  
feedin:
  type: fixed
  price: 0.08 # 0,12 [currency]/kWh

solar:
  type: template
  template: forecast-solar
  lat: 47.02063707
  lon: 15.78853445
  dec: 30 # 0 = horizontal, 90 = vertikal
  az: 0 # -180 = Norden, -90 = Osten, 0 = Süden, 90 = Westen, 180 = Norden
  kwp: 30

co2:
  type: template
  template: electricitymaps-free
  token:
  zone: AT # siehe https://api.electricitymap.org/v3/zones
```

### Log details

```text
no log entry for ui
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.2

An-Wi, 2025-02-24:
Bei mir gibt es keine Probleme bei der Mouse-over-Anzeige, allerdings fehlt mir auch die (dynamische) Preisanzeige, weil ich einen Festpreis hinterlegt habe:

[image]

Jedoch habe ich hier in der Vorhersage inhaltliche Unterschiede je nach Darstellung Hell oder Dunkel.
Im Dunkel-Modus wird die CO2-Linie ausgegraut mit angezeigt, was ich sehr gut finde. Umgedreht bei CO2-Anzeige ist die Solar-Prognose grau im Hintergrund. 
Bei "Hell" (siehe 1. Screenshot) fehlt sie bei in beiden Ansichten (Solar und CO2).

[image]

Ansonsten bin ich von diesem neue Feature absolut begeistert - es ist einfach genial! 
Auch die Integration auf der Hauptseite unter "IN" mit den heute noch zu erwartender PV-Produktion und den Schnell-Link dahinter in diese Vorhersage hier finde ich extrem gut gelungen. 👍

puntipunti, 2025-02-24:
es liegt am Dynamischen Strompreis!
mein Anbieter smartenergy (smartcontrol) liefert je 1/4 Stunde einen wert (4 werte je stunde) -- dann entsteht obiges Problem.
wenn ich den Tarif auf awattar umstelle funktioniert die Anzeige!

[image]
grid:
 # type: template
 # template: smartenergy
  type: awattar
  region: at # optional, choose at for Austria
  charges: 0.11 # optional, additional charges per kWh
  tax: 0.2 # optional, additional tax (0.1 for 10%)

wieder umgestellt auf smartcontrol:

[image]
grid:
  type: template
  template: smartenergy
  # type: awattar
  # region: at # optional, choose at for Austria
  charges: 0.11 # optional, additional charges per kWh
  tax: 0.2 # optional, additional tax (0.1 for 10%)

smartenergy] TRACE 2025/02/24 15:35:51 GET https://apis.smartenergy.at/market/v1/price

[image]

evtl kann man die Preise auch im stundentakt bekommen -- ansonsten muss man die nich "00" Min verwerfen.

puntipunti, 2025-02-24:
da hat das reinposten der config ein Formatierungswirrwar erzeugt -- aber ich glaube die essenz ist zu entnehmen.

puntipunti, 2025-02-24:
Beim einstellen der Preisgrenze werden die 1/4 h werte ignoriert und nur die ganzen Stunden herangezogen:

[image]

vs 
forecast wo auch die 1/4h werte angezeigt werden obwohl die X-Achse einen Raster von 1h hat:

[image]

naltatis, 2025-02-24:
@andig Ursache sind hier die Viertelstündchen Preisslots bei **smartenergy**. Passt du den Tarif an?

naltatis, 2025-02-24:
@puntipunti ja, das Thema kürzere Zeitslots haben wir auf dem Zettel und werden wir auch adressieren. Aktuell unterstützen wir aber nur stündliche Slots.

andig, 2025-02-24:
Es hat sich seit https://github.com/evcc-io/evcc/pull/12971#issue-2188417488 nichts geändert:

> Not tested. I'm honestly unsure how the planner would handle 15m prices. See https://github.com/evcc-io/evcc/issues/12915.

Bleibt damit zu leben oder den Tarif rauszuwerfen.

naltatis, 2025-02-24:
@andig Das ist ja ein stündlicher Tarif. Hier müssen wir nur das Datenformat anpassen.

https://private-user-images.githubusercontent.com/122234619/416237873-a4f47a12-3ae3-4462-94bb-9e76587df04d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDA0MTYxOTMsIm5iZiI6MTc0MDQxNTg5MywicGF0aCI6Ii8xMjIyMzQ2MTkvNDE2MjM3ODczLWE0ZjQ3YTEyLTNhZTMtNDQ2Mi05NGJiLTllNzY1ODdkZjA0ZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMjI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDIyNFQxNjUxMzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mOTM3ZDFmMjY1NGQzZDQ5ZjBhZWYzYmU0MGM1ZDdhMWM3MGVhZDliNjBjYWIxY2VjNWNkYzhmNzYwNmUyNDY1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.ZhEjVxLP99Q0sB80rd14UcYofLB1xBJYmptZ8LDw_k4

puntipunti, 2025-02-24:
Ja genau!
Beim Import des json nur die Werte mit Minute 00 übernehmen -- die restlichen 1/4h werte verwerfen, da evcc derzeit nicht damit umgehen kann und ohnehin immer derselbe wert drin steht (stundentarif).

puntipunti, 2025-02-24:
aber auch das Frontend (das Diagramm der Vorhersage) könnte unerwartete Daten des Backend verwerfen.
irgendwo sollte diese WHERE Klausel eingebaut werden.
so sieht es lustig aus ;-)
[image]
LG

andig, 2025-02-25:
Closed in 268cb9c9740c7d0b40b3853798fcdaea48053e95. Tatsächlich würde hier yaml statt go ausreichen wenn man nicht noch die Slots filtern müsste.