evcc-io/evcc #19140: Seit Umstellung auf DB-config ist "In" oft abweichend

nicx, 2025-02-24:
### Describe the bug

Ich habe meine komplette yaml-config weggeschmissen und EVCC neu per UI eingerichtet. Seither habe ich oft den Fall, dass wohl die werte für "In" nicht zu den gemessenen Werten des Meters passen. Daher sieht die Anzeige in der UI so aus: 

[image]

Mit älteren "Vor-UI-KOnfig-Versionen" und der yaml-konfig hast ich das Problem nie.

Rein von den Abfrageintervallen habe ich keine Veränderung zur yaml Konfiguration gemacht. Woran kann diese Diskrepanz liegen? Wie kann ich nach der Ursache herausfinden?

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
keine yaml Konfiguration mehr vorhanden
```

### Log details

```text
keine auffälligen Logmeldung erkennbar:

site ] DEBUG 2025/02/24 14:19:09 ----
[lp-2 ] DEBUG 2025/02/24 14:19:09 charge power: 0W
[lp-2 ] DEBUG 2025/02/24 14:19:09 charge currents: [0 0 0]A
[lp-1 ] DEBUG 2025/02/24 14:19:09 charge power: 0W
[lp-1 ] DEBUG 2025/02/24 14:19:09 charge currents: [0.012 0.01 0.013]A
[circuit-wallboxen] DEBUG 2025/02/24 14:19:09 power: 0W
[circuit-wallboxen] DEBUG 2025/02/24 14:19:09 current: 0.013A
[circuit-main] DEBUG 2025/02/24 14:19:09 power: -329W
[circuit-main] DEBUG 2025/02/24 14:19:09 current: 1.85A
[lpc ] DEBUG 2025/02/24 14:19:09 power: -329W
[lpc ] DEBUG 2025/02/24 14:19:09 current: 1.85A
[site ] DEBUG 2025/02/24 14:19:09 grid power: -329W
[site ] DEBUG 2025/02/24 14:19:09 battery 1 power: -626W
[site ] DEBUG 2025/02/24 14:19:09 pv 1 power: 1316W
[site ] DEBUG 2025/02/24 14:19:09 grid currents: [-1.54 1.85 -1.31]A
[site ] DEBUG 2025/02/24 14:19:09 battery 1 soc: 47%
[site ] DEBUG 2025/02/24 14:19:09 battery has priority at soc 47% (< 100%)
[site ] DEBUG 2025/02/24 14:19:09 site power: -229W
[lp-1 ] DEBUG 2025/02/24 14:19:09 charge total import: 7106.242kWh
[lp-1 ] DEBUG 2025/02/24 14:19:09 charger status: B
[lp-1 ] DEBUG 2025/02/24 14:19:09 !! pvScalePhases DOWN activePhases: 1, available power: 229W, scalable: false
[lp-1 ] DEBUG 2025/02/24 14:19:09 !! pvScalePhases UP maxPhases: 3, available power: 229W, scalable: false
[lp-1 ] DEBUG 2025/02/24 14:19:09 !! pvScalePhases EXIT
[lp-1 ] DEBUG 2025/02/24 14:19:09 pv charge current: 0.996A = 0A + 0.996A (-229W @ 1p)
[site ] DEBUG 2025/02/24 14:19:39 ----
[lp-2 ] DEBUG 2025/02/24 14:19:39 charge power: 0W
[lp-1 ] DEBUG 2025/02/24 14:19:39 charge power: 0W
[lp-2 ] DEBUG 2025/02/24 14:19:39 charge currents: [0 0 0]A
[lp-1 ] DEBUG 2025/02/24 14:19:39 charge currents: [0.012 0.01 0.013]A
[circuit-wallboxen] DEBUG 2025/02/24 14:19:39 power: 0W
[circuit-wallboxen] DEBUG 2025/02/24 14:19:39 current: 0.013A
[circuit-main] DEBUG 2025/02/24 14:19:39 power: 63W
[circuit-main] DEBUG 2025/02/24 14:19:39 current: 1.81A
[lpc ] DEBUG 2025/02/24 14:19:39 power: 63W
[lpc ] DEBUG 2025/02/24 14:19:39 current: 1.81A
[site ] DEBUG 2025/02/24 14:19:39 grid power: 63W
[site ] DEBUG 2025/02/24 14:19:39 battery 1 power: 88W
[site ] DEBUG 2025/02/24 14:19:39 pv 1 power: 689W
[site ] DEBUG 2025/02/24 14:19:39 grid currents: [-1 1.81 -0.99]A
[site ] DEBUG 2025/02/24 14:19:39 battery 1 soc: 47%
[site ] DEBUG 2025/02/24 14:19:39 site power: 251W
[lp-2 ] DEBUG 2025/02/24 14:19:39 charge total import: 215.598kWh
[lp-2 ] DEBUG 2025/02/24 14:19:39 charger status: A
[site ] DEBUG 2025/02/24 14:20:09 ----
[lp-2 ] DEBUG 2025/02/24 14:20:09 charge power: 0W
[lp-2 ] DEBUG 2025/02/24 14:20:09 charge currents: [0 0 0]A
[lp-1 ] DEBUG 2025/02/24 14:20:09 charge power: 0W
[lp-1 ] DEBUG 2025/02/24 14:20:09 charge currents: [0.012 0.01 0.013]A
[circuit-wallboxen] DEBUG 2025/02/24 14:20:09 power: 0W
[circuit-wallboxen] DEBUG 2025/02/24 14:20:09 current: 0.013A
[circuit-main] DEBUG 2025/02/24 14:20:09 power: 39W
[circuit-main] DEBUG 2025/02/24 14:20:09 current: 1.77A
[lpc ] DEBUG 2025/02/24 14:20:09 power: 39W
[lpc ] DEBUG 2025/02/24 14:20:09 current: 1.77A
[site ] DEBUG 2025/02/24 14:20:09 grid power: 39W
[site ] DEBUG 2025/02/24 14:20:09 battery 1 power: -1151W
[site ] DEBUG 2025/02/24 14:20:09 pv 1 power: 808W
[site ] DEBUG 2025/02/24 14:20:09 grid currents: [-0.99 1.77 -0.96]A
[site ] DEBUG 2025/02/24 14:20:09 battery 1 soc: 47%
[site ] DEBUG 2025/02/24 14:20:09 battery has priority at soc 47% (< 100%)
[site ] DEBUG 2025/02/24 14:20:09 site power: 139W
[lp-1 ] DEBUG 2025/02/24 14:20:09 charge total import: 7106.242kWh
[lp-1 ] DEBUG 2025/02/24 14:20:09 charger status: B
[lp-1 ] DEBUG 2025/02/24 14:20:09 !! pvScalePhases DOWN activePhases: 1, available power: -139W, scalable: false
[lp-1 ] DEBUG 2025/02/24 14:20:09 !! pvScalePhases UP maxPhases: 3, available power: -139W, scalable: false
[lp-1 ] DEBUG 2025/02/24 14:20:09 !! pvScalePhases EXIT
[lp-1 ] DEBUG 2025/02/24 14:20:09 pv charge current: 0A = 0A + -0.604A (139W @ 1p)
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.200.2

nicx, 2025-02-24:
ich habe selbst ein wenig recherchiert. hier mal meine alte yaml Konfiguration:

```
meters:
  - name: Huawei_DTSU666
    type: template
    template: huawei-dongle-powersensor
    usage: grid
    modbus: tcpip
    id: 1
    host: 192.168.3.4
    port: 502
  - name: Huawei_SUN2000
    type: template
    template: huawei-dongle-powersensor
    usage: pv
    modbus: tcpip
    id: 1
    host: 192.168.3.4
    port: 502
  - name: Huawei_LUNA2000
    type: template
    template: huawei-dongle-powersensor
    usage: battery
    modbus: tcpip
    id: 1
    host: 192.168.3.4
    port: 502
```

wie man sieht habe ich als template für grid, pv und battery jeweils "huawei-dongle-powersensor" verwendet.

In der UI kann ich dies nicht mehr so auswählen: Für grid und battery gibt es nur "HUAWEI SUN2000", für PV hingegen sowohl "HUAWEI SUN2000" als auch "HUAWEI SUN2000 (sdongle)".

Aktuell habe ich für PV das sdongle template ausgewählt. kann das problem vielleicht damit zusammenhängen? wie kann ich per UI die alte yaml Konfiguration abbilden?

VolkerK62, 2025-02-24:
Das "?" kommt daher, dass die Eingangsleistung nicht der Ausgansleistung entspricht.
 
"huawei-dongel-powersensor" = "huawei-sun2000"

nicx, 2025-02-24:
Eigentlich habe ich ja das sdongel am WR dran, daher frage ich mich wozu das sdongle-Template notwendig ist?!

@VolkerK62 danke, ich versuche mal PV ebenfalls auf "HUAWEI SUN2000" umzustellen und beobachte das Verhalten.

naltatis, 2025-02-24:
Ja **HUAWEI SUN2000** entspricht dem `huawei-sun2000` Template.
`huawei-dongle-powersensor` ist ein Alias für `huawei-sun2000` (Rückwärtskompatiblität)
Wenn du alle darauf Umstellst solltest du das exakt gleiche Verhalten wie aus der yaml bekommen. Wenn nicht kommentiere gerne noch. Ich mach hier erst einmal zu.

closed 2025-02-24 by naltatis