evcc-io/evcc #19143: Ladepunkt EEbus kompatible Wallbox per Config UI anlegen scheitert.

CKlapper, 2025-02-24:
### Describe the bug

https://github.com/evcc-io/evcc/discussions/18924

Das Problem scheint mehrere Nutzer zu treffen.

### Steps to reproduce

1. Umzug der evcc.yaml in die Config UI. Daten Loadpoint und Charger aus der yaml entfernen.
2. evcc neu starten
3. Ladepunkt anlegen (EEbus kompatible Wallbox)
4. Daten eingeben (die gleiche wie vorher in der yaml)
5. Prüfen -> funktioniert
6. Speichern. -> Das Rädchen dreht sich und dreht sich.

### Configuration details

```yaml
chargers:
- name: my_wallbox
  type: template
  template: eebus 
  ski: 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3
  host: 192.168.0.71 

loadpoints:
- title: Garage
  charger: my_wallbox
```

### Log details

```text
Wiess nicht ob das hilfreich ist:

[eebus ] DEBUG 2025/02/20 17:23:06 incoming connection request from 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3
[eebus ] DEBUG 2025/02/20 17:23:06 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 websocket read error: websocket: close 4400: STATE_WAIT_CLOSE_CONNECTION
[eebus ] DEBUG 2025/02/20 17:23:06 ski 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 disconnected
[eebus ] DEBUG 2025/02/20 17:23:18 incoming connection request from 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3
[eebus ] DEBUG 2025/02/20 17:23:18 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 websocket read error: websocket: close 4400: STATE_WAIT_CLOSE_CONNECTION
[eebus ] DEBUG 2025/02/20 17:23:18 ski 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 disconnected
[eebus ] DEBUG 2025/02/20 17:23:36 incoming connection request from 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3
[eebus ] DEBUG 2025/02/20 17:23:36 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 websocket read error: websocket: close 4400: STATE_WAIT_CLOSE_CONNECTION
[eebus ] DEBUG 2025/02/20 17:23:36 ski 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 disconnected
[eebus ] DEBUG 2025/02/20 17:23:45 trying to connect to 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 at cp-2.local
[eebus ] DEBUG 2025/02/20 17:23:45 initiating connection to 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 at cp-2.local:4711/ship/
[eebus ] DEBUG 2025/02/20 17:23:45 connection to 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 failed: dial tcp: lookup cp-2.local on 192.168.0.2:53: no such host
[eebus ] DEBUG 2025/02/20 17:23:45 trying to connect to 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 at 192.168.0.71
[eebus ] DEBUG 2025/02/20 17:23:45 initiating connection to 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 at 192.168.0.71:4711/ship/
[eebus ] DEBUG 2025/02/20 17:23:46 ski 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 connected
[eebus ] DEBUG 2025/02/20 17:23:46 Send: read 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2025/02/20 17:23:46 Recv: d:_i:4711_CC612_2S0R_CC:NodeManagement to NodeManagement read 5438987 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2025/02/20 17:23:46 Send: d:_i:4711_CC612_2S0R_CC reply 2 5438987 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2025/02/20 17:23:46 Recv: d:_i:4711_CC612_2S0R_CC:NodeManagement to NodeManagement reply 5438991 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2025/02/20 17:23:46 Send: d:_i:4711_CC612_2S0R_CC call 3 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2025/02/20 17:23:46 Send: d:_i:4711_CC612_2S0R_CC read 4 NodeManagementUseCaseData
[eebus ] DEBUG 2025/02/20 17:23:46 cs-lpc: 0 DeviceDiagnosis Server found
[eebus ] DEBUG 2025/02/20 17:23:46 cs-lpp: 0 DeviceDiagnosis Server found
[eebus ] DEBUG 2025/02/20 17:23:46 Send: d:_i:4711_CC612_2S0R_CC read 5 DeviceClassificationManufacturerData
[eebus ] DEBUG 2025/02/20 17:23:46 Send: d:_i:4711_CC612_2S0R_CC read 6 DeviceDiagnosisStateData
[eebus ] DEBUG 2025/02/20 17:23:46 ski 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 event cem-evsecc-EvseConnected
[eebus ] DEBUG 2025/02/20 17:23:47 Recv: d:_i:4711_CC612_2S0R_CC:NodeManagement to NodeManagement read 5438995 NodeManagementDestinationListData
[eebus ] DEBUG 2025/02/20 17:23:47 Send: d:_i:4711_CC612_2S0R_CC reply 7 5438995 NodeManagementDestinationListData
[eebus ] DEBUG 2025/02/20 17:23:47 Recv: d:_i:4711_CC612_2S0R_CC:NodeManagement to NodeManagement result 5439001 3 ResultData 0
[eebus ] DEBUG 2025/02/20 17:23:47 Recv: d:_i:4711_CC612_2S0R_CC:NodeManagement to NodeManagement reply 5439006 4 NodeManagementUseCaseData
[eebus ] DEBUG 2025/02/20 17:23:47 ski 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 event cem-evsecc-UseCaseSupportUpdate
[eebus ] DEBUG 2025/02/20 17:23:47 Recv: d:_i:4711_CC612_2S0R_CC:NodeManagement to NodeManagement read 5438999 NodeManagementUseCaseData
[eebus ] DEBUG 2025/02/20 17:23:47 Send: d:_i:4711_CC612_2S0R_CC reply 8 5438999 NodeManagementUseCaseData
[eebus ] DEBUG 2025/02/20 17:23:47 Recv: d:_i:4711_CC612_2S0R_CC:DeviceClassification to DeviceClassification reply 5439010 5 DeviceClassificationManufacturerData
[eebus ] DEBUG 2025/02/20 17:23:47 ski 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 event cem-evsecc-DataUpdateManufacturerData
[eebus ] DEBUG 2025/02/20 17:23:47 Recv: d:_i:4711_CC612_2S0R_CC:NodeManagement to NodeManagement call 5439013 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2025/02/20 17:23:47 Send: d:_i:4711_CC612_2S0R_CC result 9 5439013 ResultData 0
[eebus ] DEBUG 2025/02/20 17:23:47 Recv: d:_i:4711_CC612_2S0R_CC:DeviceDiagnosis to DeviceDiagnosis reply 5439016 6 DeviceDiagnosisStateData
[eebus ] DEBUG 2025/02/20 17:23:47 ski 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 event cem-evsecc-DataUpdateOperatingState
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.200.1

andig, 2025-02-24:
> failed: dial tcp: lookup cp-2.local on 192.168.0.2:53: no such host

Nochmal probieren?

andig, 2025-02-24:
@naltatis eine Lösung wäre (mehrfach angesprochen) auch defekte Geräte speichern zu können. Dann gehts wenigstens nach dem Restart.

loeffelpan, 2025-03-09:
Wurde leider automatisch geschlossen.
Bitte wieder öffnen.

CKlapper, 2025-03-09:
Zur Info. mit v.0.200.8 konnte ich den Ladepunkt jetzt in der UI speichern. Mit den Versionen davor habe ich es nicht probiert.

loeffelpan, 2025-03-09:
Ging bei mir jetzt auch.
Aber auch erst im zweiten Versuch.

CarG70, 2025-03-15:
ich versuche auch gerade erfolglos meinen ID.Charger Pro 1 aus der Yaml umzuziehen. Alle anderen Geräte sind da schon raus. Egal ob über eebus oder über das VW Profil, die Box wird erfolgreich erkannt, läßt ich aber nicht speichern. Endloses drehen des Rädchen, wie oben beschrieben. EVCC Version 0.200.9 auf Raspi ohne HA oder sonstiges. Wie habt Ihr das geschafft?