evcc-io/evcc #19145: Remove spaces from forecast-solar horizon

schenlap, 2025-02-24:
### Describe the bug

Wenn man bei forcast-solar Leerzeichen bei horizon angibt, dann schlägt die Abfragt fehl.

Die Anfrage dazu war: `[solar-template-1] TRACE 2025/02/24 19:38:09 GET https://api.forecast.solar/estimate/xx.xxx/yy.yyy/20/-20/8.4?time=utc&full=1&resolution=60&horizon=5, 60, 5, 20`

Entfernt man die Leerzeichen aus der Config, dann geht es. evcc sollte die Leerzeichen selbstständig entfernen.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
- type: template
      template: forecast-solar
      lat: xx.xxx
      lon: yy.yyy
      dec: 20
      az: -20 #SSO
      kwp: 8.4
      horizon: '5, 60, 5, 20'
```

### Log details

```text
[solar-template-1] TRACE 2025/02/24 19:38:09 <html>
<head><title>400 Bad Request</title></head>
<body>
<center><h1>400 Bad Request</h1></center>
<hr><center>nginx</center>
</body>
</html>
[tariff] ERROR 2025/02/24 19:38:09 unexpected status: 400 (Bad Request)
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.3

andig, 2025-02-24:
Ist ein Konfigurationsfehler- einfach keine iengeben?

closed 2025-02-24 by andig

schenlap, 2025-02-24:
Kann man so sehen. So habe ich es auch gelöst. Für den Benutzer solls aber so einfach wie möglich sein und so wenig Fehlermöglichkeit bieten wie möglich. Ich wette da fällt bald der nächste darüber. Ließe sich ja leicht lösen in Software.

andig, 2025-02-24:
Im template leider nixht. Cc @premultiply für besseren Kommentar.

premultiply, 2025-02-24:
Mehr als "unnötige Leerzeichen bei der Eingabe weglassen" fällt mir dazu leider als Antwort auch nicht ein.