evcc-io/evcc #19153: EVCC 0.200.3 Sofarsolar HYD site power  & battery power  um eine Größenordnung falsch

ThomasGeb, 2025-02-24:
### Describe the bug

Nach dem Update auf 0.200.3 von 0.200.1 scheint es, als wenn die Leistung der Batterie und des "Netzzählers" um eine Größenordnung zu klein sind. Ein Downgrade auf 0.200.1 behebt den Fehler. 

statt 300W Batterypower werden 30W ausgegeben und in den Logs angezeigt

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
-
```

### Log details

```text
[site ] DEBUG 2025/02/24 22:17:56 site power: 130W
[lp-1 ] DEBUG 2025/02/24 22:17:56 charger status: A
[site ] DEBUG 2025/02/24 22:18:03 ----
[lp-1 ] DEBUG 2025/02/24 22:18:03 charge power: 0W
[lp-1 ] DEBUG 2025/02/24 22:18:03 charge currents: [0 0 0]A
[db:8 ] TRACE 2025/02/24 22:18:03 modbus: send 01 d5 00 00 00 06 01 03 04 88 00 01
[site ] DEBUG 2025/02/24 22:18:03 pv 2 power: 0W
[db:8 ] TRACE 2025/02/24 22:18:03 modbus: recv 01 d5 00 00 00 05 01 03 02 00 00
[site ] DEBUG 2025/02/24 22:18:03 grid power: -0W
[db:5 ] TRACE 2025/02/24 22:18:03 modbus: send 01 d6 00 00 00 06 01 03 05 c4 00 01
[db:5 ] TRACE 2025/02/24 22:18:03 modbus: recv 01 d6 00 00 00 05 01 03 02 00 00
[site ] DEBUG 2025/02/24 22:18:03 pv 1 power: 0W
[db:6 ] TRACE 2025/02/24 22:18:03 modbus: send 01 d7 00 00 00 06 01 03 06 67 00 01
[db:6 ] TRACE 2025/02/24 22:18:03 modbus: recv 01 d7 00 00 00 05 01 03 02 ff fe
[site ] DEBUG 2025/02/24 22:18:03 battery 1 power: 20W
[db:6 ] TRACE 2025/02/24 22:18:03 modbus: send 01 d8 00 00 00 06 01 03 06 9a 00 02
[db:6 ] TRACE 2025/02/24 22:18:04 modbus: recv 01 d8 00 00 00 07 01 03 04 00 00 46 5f
[db:8 ] TRACE 2025/02/24 22:18:04 modbus: send 01 d9 00 00 00 06 01 03 04 93 00 01
[db:8 ] TRACE 2025/02/24 22:18:04 modbus: recv 01 d9 00 00 00 05 01 03 02 00 00
[db:5 ] TRACE 2025/02/24 22:18:04 modbus: send 01 da 00 00 00 06 01 03 06 86 00 02
[db:5 ] TRACE 2025/02/24 22:18:04 modbus: recv 01 da 00 00 00 07 01 03 04 00 03 5b b3
[site ] DEBUG 2025/02/24 22:18:04 pv power: 0W
[db:6 ] TRACE 2025/02/24 22:18:04 modbus: send 01 db 00 00 00 06 01 03 06 68 00 01
[db:6 ] TRACE 2025/02/24 22:18:04 modbus: recv 01 db 00 00 00 05 01 03 02 00 43
[site ] DEBUG 2025/02/24 22:18:04 battery 1 soc: 67%
[db:8 ] TRACE 2025/02/24 22:18:04 modbus: send 01 dc 00 00 00 06 01 03 04 9e 00 01
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.3

andig, 2025-02-24:
Missing config.

/cc @cschlipf

closed 2025-02-24 by andig

ThomasGeb, 2025-02-24:
> Missing config.
> 
> /cc [@cschlipf](https://github.com/cschlipf)

Hab umgestellt auf die Konfiguration über die ConfigUI. Sorry dass ich nicht weiß wie ich die Config exportiere so dass ihr sie brauchen könnet :(

andig, 2025-02-25:
Du kannst

    evcc config

nutzen.

@cschlipf die neuen Register scheinen eine andere Skalierung zu haben falls es ein g3 ist. Fix oder revert?

cschlipf, 2025-02-25:
OK, ich sehe Home Power ist nur auf Batteriebetrieb falsch. Sonst passt es. 

[image]

Ich denke der neue Scalefactor müsste -100 sein. Ich ändere das.

Ihr wart leider mit dem Release etwas zu schnell. Ich konnte es ja nicht mehr testen :-(

cschlipf, 2025-02-25:
@andig, Warum ist der Defect hier geschlossen worden?

cschlipf, 2025-02-25:
PR ist da.

Man sollte einfach unter Fieber keinen PR aufmachen :-(

PR #19158 (2025-02-25): Sofar: fix battery power scale
After switching to Power_Bat_Total I missed to update the scale. This fixes it.

Fixes #19153